esp8266/Arduino #1582: Compiler error: section `.text' will not fit in region `iram1_0_seg'

mvdbro, 2016-02-05:
Using latest commit (b5684b4). Getting this error:

../hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
../ESPEasy.cpp.elf section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status

(It was still ok with the commit from jan 30st and the sketch is unchanged)

What could be wrong?

igrr, 2016-02-05:
Which sketch produces this error?
Text section grew a bit when umm_malloc was added, but all the example
sketches can still be built successfully.

hallard, 2016-02-05:
hey guys, just bought my MAC 2 days ago, installed all from scratch and got the same error with my big sketches

On windows ouput is 
```
Sketch uses 367,586 bytes (35%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 47,284 bytes (57%) of dynamic memory, leaving 34,636 bytes for local variables. Maximum is 81,920 bytes.
```
on MAC same error
```
section text' will not fit in regioniram1_0_seg'  collect2exe: error: ld returned 1 exit status
```

sticilface, 2016-02-05:
jup.  I get it as well... interestingly same esp8266 folder used in platformio doesn't seem to generate the error

igrr, 2016-02-05:
I changed linker script when umm_malloc was added. Platformio stores linker
scripts separately from the core, which means you are probably using the
old linker script when building with platformio.
These changes place all functions from umm_malloc.c into IRAM. If you don't
have these changes, umm_malloc goes to flash, and IRAM has more room.

sticilface, 2016-02-05:
Ah, that might explain that. I'm trying to fine the culprit lib, as the examples build, but my sketch does not

hallard, 2016-02-05:
Sorry, even on Windows with latest git (I was not up to date on this PC)

Is there any quick & dirty fix we can do to be able get it working until fixed ?

igrr, 2016-02-05:
Revert the change to the linker script, I guess.

hallard, 2016-02-05:
humm, that's what I thought, no incidence with new libc_replacements.c and heap.c ?

igrr, 2016-02-05:
Could you guys drop me an .elf of one of your "large" sketches compiled
before these changes? I want to figure out how much space we need to free
up and what contributes to IRAM usage. Thanks.

sticilface, 2016-02-05:
The IDE seems to delete it.. it appears in the tmp folder, then vanishes with the error

sticilface, 2016-02-05:
ah sorry i didn't see 'before the error' you happy with one compiled with the linker changes reverted?

igrr, 2016-02-05:
Yes, that one will do.

sticilface, 2016-02-05:
https://www.dropbox.com/s/zavncym4guwu1xh/Melvanimate-example.cpp.elf?dl=0

hallard, 2016-02-05:
Sure, this one compiled with revert changes
http://hallard.me/download/sensor.ino.elf.gz

hallard, 2016-02-05:
As @igrr said, I confirm that reverting back only `eagle.app.v6.common.ld` solved this problem

modabasi, 2016-02-06:
I have the very same problem right now and it started after I've switched from the boards manager installation to git installation. I wanted to play with https://github.com/Links2004/arduinoWebSockets and it needed the git version to be installed. Now my older scripts which worked with the boards manager installation are giving "section `.text' will not fit in region `iram1_0_seg'" error.

supersjimmie, 2016-02-09:
Confirmed too, reverted ```agle.app.v6.common.ld```.

d3alek, 2016-02-10:
I encountered an unfortunate boot loop when running 04c6609 with reverted `eagle.app.v6.common.ld` on one of my ESP-12:

```
...
Fatal exception (28):

epc1=0x40202548, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000001, depc=0x00000000
...
```

Two ways to fix it: choose a smaller flash size (originally at `4M (1M SPIFFS)`), or roll back to 2e1be32 just before umm_malloc was added. Note, the code was running fine for a while until it suddenly started crashing on boot. 

I disassembled the .elf and found that 0x40202548 is somewhere in umm_info, which gave me a clue that the crash is umm_malloc related. Not sure why umm_malloc was called if my `eagle.app.v6.common.ld` was reverted.

PsY4, 2016-02-11:
Same here : Can't compile big sketches anymore, and reverting `eagle.app.v6.common.ld` allows compilation, but gives `Fatal exception (28):` :'(

igrr, 2016-02-18:
The change in 707c87f freed up approximately 520 bytes of IRAM, please give it a try.

PsY4, 2016-02-18:
Yes, that made the trick ! My big sketch now compiles again ! Thx a lot.

mvdbro, 2016-02-18:
Compiles fine again. Thanks!

closed 2016-02-18 by mvdbro

Silidrone, 2017-10-10:
Im sorry, I dont understand anything you talk about, what is `eagle.app.v6.common.ld`??