esp8266/Arduino #1584: String concatenation and ostringstream undefined reference

hubertokf, 2016-02-05:
So this is my code. This is just a test, the real problem happen on my other project. Im trying to generate some xml by concatenating strings. 

`#include "Arduino.h"
#include <stdio.h>
#include <string>
#include <sstream>

using namespace std;

template class basic_string<char>;

void setup() { Serial.begin(115200); }

void loop() {
  ostringstream stream;
  stream << "<icon>";

  string s1("toto");
  string s2("313");
  string s;

  s = "www." + s2;
}
`

closed 2016-02-05 by hubertokf

neu-rah, 2016-11-07:
confirmed using headers:
```c++
#include <iostream>     // std::cout, std::ios
#include <sstream>
```

neu-rah, 2016-11-07:
from similar problems it seems to be in uClibc, but uClibc would have to be re-linked with some extra options and then the platform linked with the new version...