esp8266/Arduino #1585: Strange issue with HTTP client payload

modabasi, 2016-02-05:
Hi all,

I'm having this strange issue... I'm making a GET request to my PHP script which only echoes "Done" if everything's fine... And in my Arduino code, I have this...

```
if (httpCode == 200) {
   String payload = http.getString();       
   Serial.println(payload);
}

```

I call the PHP script from my browser, verify that it only reads Done, same for the source code, but the payload printed in the serial monitor is this;

```
4
Done
0

```

If I change the echoed word to "Allright", the payload is this;

```
8
Done
0

```

As you can see I always get the number of characters of the echoed word, then the word, then a zero, then an empty line as the payload. Any ideas why this is happening to me?

Thanks in advance...

Links2004, 2016-02-06:
it looks like chunked encoding to me, supported in Staging version.
which version of the core do you use?

modabasi, 2016-02-06:
I should be using the stable version 2.0 as I'm loading the libraries through the boards manager using the following URL;

http://arduino.esp8266.com/stable/package_esp8266com_index.json

Links2004, 2016-02-06:
yes its stable, when you update to Staging it will work.

asetyde, 2016-02-06:
Hi Markus , when new 2.1.0 stable ll come ? any idea

igrr, 2016-02-06:
We've got a few pull requests up for review, and at least one blocker issue
(.text segment overflow). I guess we are going to do rc3 first, and if all
goes well for a few days, then release 2.1.0.

asetyde, 2016-02-06:
OK BIG THANKS

closed 2016-02-27 by igrr