esp8266/Arduino #1589: parametric  WiFi.softAP function (staging ver 2.1.0-rc2 ) still have problem

mkeyno, 2016-02-06:
I've check other issue and run debugging test and find, send the String or const char* parameter to the WiFi.softAP() function still not working , I find in https://github.com/esp8266/Arduino/blob/5f80ad5a78062882ff2bf509bae7d69996a53ca7/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp#L124  wifi config still persistent and  wifi_softap_set_config(&conf); return false , I hope @igrr  or @Links2004  or any guy help me how to revise the these header files to solve the problem , I'm so tried waiting for new version for minor issue and re download all this huge library again

Links2004, 2016-02-06:
can you provide a example for your problem?
enable debug for WiFi may also help to track it down.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

mkeyno, 2016-02-06:
this is the captive portal example of the library , which is only assign the new AP ssid when strictly put the string in  WiFi.softAP("with any string ","with any string "); // working

`#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(192, 168, 4, 1);
DNSServer dnsServer;
ESP8266WebServer webServer(80);
 const char* AP_SSID="My Smart IOT";
 const char* AP_Password="12345678";

String responseHTML = ""
  "<!DOCTYPE html><html><head><title>CaptivePortal</title></head><body>"
  "<h1>Hello World!</h1><p>This is a captive portal example. All requests will "
  "be redirected here.</p></body></html>";

void setup() {
  Serial.begin(115200);
  Serial.println("\n\n\n\t\t Booting Sketech...");
  WiFi.disconnect(true); delay(1000);
  //Serial.setDebugOutput(true);
  WiFi.mode(WIFI_AP);
                  // local_ip,   gateway,   subnet
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));

  WiFi.softAP(AP_SSID,AP_Password); // not working
 WiFi.softAP("with any string ","with any string "); // working

  // if DNSServer is started with "*" for domain name, it will reply with provided IP to all DNS request
  //         (port,  domain Name, resolved IP)
  dnsServer.start(DNS_PORT, "*", apIP);

  // replay to all requests with same HTMLb ([](){});
  webServer.onNotFound([]() {
								webServer.send(200, "text/html", responseHTML);
							  });
  webServer.begin();
}

void loop() {
  dnsServer.processNextRequest();
  webServer.handleClient();
}
`

marvinroger, 2016-02-07:
Well, it works fine for me, so that's probably an edge case.

mkeyno, 2016-02-07:
hi @marvinroger , you mean  this parametric  code ` WiFi.softAP(AP_SSID,AP_Password); `  work fine in the last staging version ?

mkeyno, 2016-02-07:
@igrr @Links2004 
hi guys , why this following code working and mine is not working?!!
`#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <FS.h>
#include <ArduinoOTA.h>

 

const char* ap_default_ssid = "My AP"; ///< Default SSID.
const char* ap_default_psk = "12345678"; ///< Default PSK.
String station_ssid = "keyno";
String station_psk = "----";

bool loadConfig(String *ssid, String *pass)
{
  *ssid = station_ssid;
  *pass = station_psk ;
 return true;
}
void setup() 
{
 Serial.begin(115200);
  
  delay(100);

  Serial.println("\r\n");
  Serial.print("Chip ID: 0x");
  Serial.println(ESP.getChipId(), HEX);
 loadConfig(&station_ssid, &station_psk); 
  WiFi.mode(WIFI_STA);
 delay(10);
 WiFi.begin(station_ssid.c_str(), station_psk.c_str());
 Serial.print("WiFi.SSID()=");Serial.println(WiFi.SSID());
 unsigned long startTime = millis();
  while (WiFi.status() != WL_CONNECTED && millis() - startTime < 10000)	  {
                                                                    Serial.write('.');
                                                                     Serial.print(WiFi.status());
                                                                    delay(500);
                                                                    }
  Serial.println();
  
  if(WiFi.status() == WL_CONNECTED)
  									  {
  										// ... print IP Address
  										Serial.print("IP address: ");
  										Serial.println(WiFi.localIP());
  									  }
 }

void loop() 
{
  // put your main code here, to run repeatedly:

}`

mkeyno, 2016-02-07:
My plan is to update my wifi setting through request from client and store it in EEPROM , I created  utill.h  header file along with main.ino file as per following , but I cant set WiFi.begin(a,b) & WiFi.softAP(a, b); parametric 
utill.h
`
#ifndef utill_h
#define utill_h

String Restor(byte);
//void Up_AP(void);
//void Up_Station(void);
void Add_count_reset(void);
//////////////////////////global parameters////////////////////
static char WIFImode=0;
  String AP_SSID="";
  String AP_Password="";
static String SSID="";
static String Password="";
static String MyHost="";
static char error_code='N';
static byte reset_num=0;

IPAddress   apIP(192, 168, 4, 1);  
IPAddress netMsk(255, 255, 255, 0);
///////////////////////////////////////////////////////////////
//////////////////////////////////////////////////////////////////////////////////////////////
//void Restor_All(void);
void Restor_All(void)
{
 WIFImode=(char)EEPROM.read(0);	Serial.print(" WIFImode =" );Serial.println(WIFImode);
  AP_SSID=Restor(1);           Serial.println(" AP_SSID =" + AP_SSID);//Serial.println(AP_SSID);
  AP_Password=Restor(21);      Serial.print(" AP_Password =");Serial.println(AP_Password);
 SSID=Restor(31);              Serial.print(" SSID =");Serial.println(SSID);
 Password=Restor(41);         Serial.print(" Password =");Serial.println(Password);
 MyHost=Restor(51);          Serial.print(" MyHost =");Serial.println(MyHost);
 reset_num=(byte)EEPROM.read(70); Serial.print(" reset_num =");Serial.println(reset_num);	
}
//////////////////////////////////////////////////////////////////////////////////////////////

String Restor(byte location)
{
  char str[20]={0};
  byte i=0;
  byte lenght=byte(EEPROM.read(location));
  if(lenght >0 && lenght< 21) for( i=0;i<lenght;i++) {
                                                      str[i]=(char)EEPROM.read(location + 1 + i);  str[i+1]='\0';
                                                       // Serial.println(c[i]);
                                                       // str[i]=(char)EEPROM.read(location + 1 + i);Serial.println(str[i]);
                                                    }
 //Serial.print(" str :");Serial.print(str); 
  return String(str);
        
 // Serial.print("lenght=");Serial.print(lenght); Serial.print(" String=");Serial.println(str);

}
//////////////////////////////////////////////////////////////////////////////////////////////

bool Store(int location, String str)
{
  char*p =&str[0];
  byte lenght=0;
  while(*p){
            EEPROM.write(location+1, p[0]);
             location++; p++; lenght++;   
              //Serial.print("lenght=");Serial.print(lenght); Serial.print(" location=");Serial.print(location);Serial.print(" char=");Serial.println(p);
            }
 
  EEPROM.write(location-lenght, lenght);    
 EEPROM.commit();  
  // Serial.print("lenght stored=");Serial.println(EEPROM.read(location-lenght));
  return true;
}
[... 76 lines cut ...]
   
  EEPROM.write(0, 'A');
  Store( 1, "Keyno_IOT+");   AP_SSID=Restor(1);
  Store(31, "keyno");       SSID=Restor(31);
  Store(41, "09127163464"); Password=Restor(41);
  Restor_All();
  
  
  if(WIFImode=='S'&& SSID.length()>2 && reset_num<5 ) Up_Station();
  else                                                Up_AP();
   
  
  server.onNotFound([]() {
			 server.send(200, "text/html", "fffffffffffffffffffff");
			  });
  server.begin();

}

void loop(void)
{
  server.handleClient();
}

`

marvinroger, 2016-02-07:
Yes, using latest staging version it works for me. Also, you might want to take a look at how I store config to EEPROM: https://github.com/marvinroger/homie-esp8266/tree/90a33f48e4993bd90ece969d3382b511d0ba8b0d 

In particular, `Config.cpp` and `Datatypes/ConfigStruct.hpp`

mkeyno, 2016-02-07:
hi @marvinroger  , are you sure about compatibility with last staging version , because I have had the following error

_ D:\arduino-1.6.4-windows\arduino-1.6.4\libraries\homie_ESP8266\src/Homie/Boot/Boot.hpp:4:25: fatal error: ESP8266WiFi.h: No such file or directory
 #include <ESP8266WiFi.h> _

marvinroger, 2016-02-07:
Yes I am. Have you selected the correct board in the Arduino IDE?

mkeyno, 2016-02-07:
which broad should I select ? such as nodemcu 1.0?

marvinroger, 2016-02-07:
Yes, for example. `ESP8266WiFi.h` (the error you're getting) is bundled into the esp8266 package, so the issue is on your side.

drmpf, 2016-02-08:
I have a library I use that does just that.  It might help you to compare the code with yours
http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html  has links to the library and an example sketch

hallard, 2016-02-10:
Guys,
while testing code for #1615 I saw strange things also (IDE 1.6.7 + latest GIT)

the line `WiFi.softAP("ESP-SOFTAP","allowme");` does show me SSID ESP_xxyyzz in Wifi List of my computer (mainly an old test), but I can connect to it, strange !!

igrr, 2016-02-10:
WPA passwords have to be at least 8 characters long, so if you are using "allowme" literally, it won't work.

hallard, 2016-02-10:
@igrr thanks for the tip, works better now is more char as PSK !!

closed 2016-03-12 by igrr