esp8266/Arduino #1594: Random crash ESP286 WEMOS D1    DHT 22 server

hymby, 2016-02-06:
Random crash ESP286 WEMOS D1     

a couple of hours Tested result: crash Wemos D1
what causes this?

///////////////////////////////////////////////////////////////////////////
Arduino side

#include <Ethernet.h>
#include <SPI.h>
byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xE1};///////////!!
IPAddress ip(192, 168, 1, 175);
IPAddress gateway(192, 168, 1, 254);
IPAddress subnet(255, 255, 255, 0);
EthernetClient client;

IPAddress server(192, 168, 1, 11); ////////////////// WEMOS D1 Side

long mintavetelIdeje = 0;
const long mintavetelSzunet = 5000;

void setup() {
  Serial.begin(9600);
  Serial.println("start");
  Ethernet.begin(mac, ip);
  delay(1000);
}

String request(char what) {
  String value = "";
  if (client.connect(server, 85)) {
    client.println(what);
    delay(500);
    while (client.available()) {
      char c = client.read();
      value.concat(c);
    delay(200);
  }
    client.stop();
  }
  return value;
}

void loop() {
  if (millis() - mintavetelIdeje > mintavetelSzunet) {
    mintavetelIdeje = millis();
    String temp = request('h');
    String hum = request('l');
    String pinState = request('p');
    Serial.println(temp + "\n" + hum + "\n" + pinState + "\n");
   
 }
}

//////////////////////////////////////////////////////////////////////////////////////////////

WEMOS D1 R2 side

#include <ESP8266WiFi.h>

WiFiServer server(85);////////////////////////////Arduino  Side!!

#include "DHT.h"   // dht  1.1.1   !!!!!
#define DHTPIN D2
#define DHTTYPE DHT22
DHT dht(DHTPIN, DHTTYPE);

const char* ssid = ".......";
const char* password = "........";

String temp;
float humidity_h;
float temp_c;
String value ;
unsigned long dht_read_time = 0;
unsigned long dht_read_time_when = 5000;
int ledState = LOW;
unsigned long previousMillis = 0;
unsigned long interval = 100;
const int led = D6;
long kul;
String data;
String pinState(int pin) {
  pinMode(pin, INPUT);
  pinMode(pin, HIGH);
  switch (digitalRead(pin)) {
    case 0: return "LOW";
    case 1: return "HIGH";
  }
}

void Strobe() {
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
    if (ledState == LOW)
      ledState = HIGH;
    else
      ledState = LOW;
    digitalWrite(led, ledState);
  }
}

void setup(void) {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  dht.begin();
  Serial.println("");
  pinMode ( led, OUTPUT );
  digitalWrite ( led, 0 );
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  server.begin();
}

void dht_read() {
  if (millis() - dht_read_time > dht_read_time_when) {
    dht_read_time = millis();
    temp_c = dht.readTemperature();
    humidity_h = dht.readHumidity();
  }
}

void accept_request() {
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  Serial.println("new client");
  while (!client.available()) {
    delay(1);
  }
  char c = client.read();
  switch (c) {
    case 'h':
      value = String(temp_c, 1);
      delay(100);
      break;
    case 'l':
      value = String(humidity_h, 1);
      delay(100);
      break;
    case 'p':
      value = pinState(D7);
      delay(100);
      break;
    default:
      value = "";
      delay(100);
      break;
  }
  if (!value.equals("")) {
    client.print(value);
    delay(1);
    Serial.print(c);
    Serial.println(": " + value);
  }
  delay(1);
  Serial.println("Client disonnected");
}

void loop(void) {
  dht_read();
  accept_request();
  Strobe();
 

  if (millis() - kul > 1000) {
    kul = millis();
    Serial.print("Free heap:");
    Serial.println(ESP.getFreeHeap(), DEC);
  }
}

hallard, 2016-02-07:
@hymby 
I didn't checked your code, but I also had random crashes on 75% of my nodes last week. At first I thought it was in my code but it wasn't, here what I did and now it works fine for 5 days without any crash and reset. I don't know wich one did the trick but doing all of this will not hurt. 
- check power quality and/or USB cable (by changing them to test), triple check this one my major problems came from this point.
- Put a big caps between 3V3 and GND (1000uF 10V) to increase resistance on power fluctuations and FCC
- Set flash speed to 40MHz (it's okay if you selected Wemos boards on IDE)
- Switch flash mode on QIO/DIO (not sure with which one I've done on this)
- Avoid multiple boards and Wifi modules setup too close, if it's the case, try to set them far away from each other more than 75cm/1m is fine (in my case)

I suggest making these changes 1 by 1 to know which one solve your issue and this may help others if you succeed ;-)

As you can see below, I'm really happy, 5 days uptime with no reset at all on 5 nodes, works great ;-)
[image]

hallard, 2016-02-07:
Oh, by the way just to say that DHT22 sensors are far from my favorites because despite the fact a measure is slow, they bit bang protocol with a IO/pin and are touching timing and IRQ and with WifI enabled and working I like letting these background task doing their stuff. 
I'm using some classic I2C one without doing any fancy stuff on IRQ in my sketch ;-)
Less consumption, less "fancy IRQ", faster, smaller, cheaper, excellent precision, well love them ;-)
check this [one](http://www.ebay.com/itm/Si7021-Industrial-High-Precision-Humidity-Sensor-I2C-Interface-for-Arduino-/201371694878?hash=item2ee2b0371e:g:rroAAOSw2VJVfqFt)

hymby, 2016-02-07:
Thank you for checking

Instead of trying so DHT22

float temp () {
   return random (-200.400) /10.0;
  }
I try to Proposal
Thanks

hymby, 2016-02-07:
The proposed've changed.
Unfortunately crash is still random.

I tried EXAMPLE ESP8266WIFI >> >>>> WifiWebServer code here within 1 hour crash. else if that is so then you can suspect a library error?

Arduino sie

#include <Ethernet.h>
#include <SPI.h>
byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xE1};
IPAddress ip(192, 168, 1, 175);
IPAddress gateway(192, 168, 1, 254);
IPAddress subnet(255, 255, 255, 0);
EthernetClient client1;
String led_off;
String led_on;
String value;
EthernetServer server(100);

IPAddress server1(192, 168, 1, 11);/////////////////////////// esp8266 wemos D1

void setup() {
  Serial.begin(115200);
  Ethernet.begin(mac, ip);
  Serial.print("start ");
  delay(1000);
}

String req(String mit) {
  String value = "";
  if (client1.connect(server1, 80)) { ///////////////////////////////!!!!Wemos
    delay(1);
    client1.println(mit);
    delay(1);
    client1.stop();
  }
  delay(1);
  return value;
}
void will_switch() {
  delay(1000);
  led_off = req("/gpio/1");
  delay(1000);
  led_on = req("/gpio/0");
  delay(1000);
}

void loop() {
  will_switch();
}

/////////////////////////////////////////////////////////////////////

ESP8266 Wemos D1 R2

Arduino 1.6.7 
File>>Example>>> ESP8266WIFI >> >>>> WifiWebServer code 

/*
 *  This sketch demonstrates how to set up a simple HTTP-like server.
 *  The server will set a GPIO pin depending on the request
 *    http://server_ip/gpio/0 will set the GPIO2 low,
 *    http://server_ip/gpio/1 will set the GPIO2 high
 *  server_ip is the IP address of the ESP8266 module, will be 
 *  printed to Serial when the module is connected.
 */

#include <ESP8266WiFi.h>
const char* ssid = "........";
const char* password = "...........";

// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(80);

void setup() {
  Serial.begin(115200);
  delay(10);

  // prepare GPIO2
  pinMode(D6, OUTPUT);
  digitalWrite(D6, 0);
  
  // Connect to WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  // Start the server
  server.begin();
  Serial.println("Server started");

  // Print the IP address
  Serial.println(WiFi.localIP());
}

void loop() {
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  
  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }
  
  // Read the first line of the request
  String req = client.readStringUntil('\r');
  Serial.println(req);
  client.flush();
  
  // Match the request
  int val;
  if (req.indexOf("/gpio/0") != -1)
    val = 0;
  else if (req.indexOf("/gpio/1") != -1)
    val = 1;
  else {
    Serial.println("invalid request");
    client.stop();
    return;
  }

  // Set GPIO2 according to the request
  digitalWrite(D6, val);
  
  client.flush();

  // Prepare the response
  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>\r\nGPIO is now ";
  s += (val)?"high":"low";
  s += "</html>\n";

  // Send the response to the client
  client.print(s);
  delay(1);
  Serial.println("Client disonnected");

  // The client will actually be disconnected 
  // when the function returns and 'client' object is detroyed
}

hallard, 2016-02-10:
I don't think sample sketch are buggy, did you tried changing power supply ? Using one from standalone phone charger could be great. Also the 1000uF cap between 3V3 and GND ;)

how freq do you request the ESP web server of sample sketch from your browser/command line?

osos, 2016-02-10:
I am also experiencing random crashes with both esp-01 and esp-12e submitting temp and humidity from DHT22.

Have tried solving using 470uF, but not enough. voltage seems stable both on battery and on usb charger with step-down converter.

The esp-12e is driving a oled i2c display as well, the esp-01 only the dht22 submitting date to a php-script on the local network.

Setup is arduino 1.6.7 and esp8266 v2.2.0.

igrr, 2016-02-10:
DHT library (Adafruit DHT-sensor-library) disables interrupts for the
duration of read, longer than 20 microseconds. With the latest SDK such
long periods of interrupts are not supported.

hymby, 2016-02-10:
I do not speak English only translate.

I have tried <WiFiClient.h> The above example program.
I sent a simple request to the server and sent back to temperature.
After crashing ranging within 160 requests.

Now I use the example ESP8266Webserver AdvancedWebServer >>
  this sends an HTTP request to Arduino and it works perfectly.
(6640 Now I look for.) You can Use the library.
hymby

osos, 2016-02-10:
@igrr  are there other dht22 libraries available which does disable interrupts. I guess the time measure could be achived using exactly a change interrupt on the data pin.

chaeplin, 2016-02-10:
@osos check 
https://github.com/piettetech/PietteTech_DHT
ported to esp8266 : https://github.com/chaeplin/PietteTech_DHT-8266
I am testing timing now.

osos, 2016-02-11:
@chaeplin very interesting thanks. Will look into it, hopefully during the weekend.

osos, 2016-02-19:
@chaeplin my esp-12e with oled i2c display and dht22 has been running for 5 days now without any lock-up/crash using your interrupt-driven DHT22 driver. Thanks.

Will probably switch to si7021 sensors anyway.

duncan-a, 2016-02-19:
@osos <https://github.com/osos> Any chance you could post your functioning
code?

I spent ages going over the examples that came with the PietteTech_DHT-8266
library and tied myself into knots trying to work out how to add a DHT to
what I had (the Adafruit DHT library was easy but doesn't work reliably on
the ESP for me - Ivan has, I think, explained why).

Thanks

chaeplin, 2016-02-19:
@duncan-a Sorry for that. Simple one is here.
https://github.com/chaeplin/PietteTech_DHT-8266/blob/master/examples/DHT_esp8266/DHT_esp8266.ino

osos, 2016-02-20:
The DHT22 essential parts of my code:

```
#include <PietteTech_DHT.h>

void setReadyForDHTUpdate();
bool readyForDHTUpdate = true;

const int DHT_UPDATE_INTERVAL_SECS = 2 * 60; // DHT hvert 2 min.

// DHT Settings
#define DHTPIN 14     // what pin we're connected to
#define DHTTYPE DHT22   // DHT 22  (AM2302)
#define DHT_SAMPLE_INTERVAL 2000

#define REPORT_INTERVAL 5000 // in msec
//declaration
void dht_wrapper(); // must be declared before the lib initialization

// Lib instantiate
PietteTech_DHT dht(DHTPIN, DHTTYPE, dht_wrapper);
// Connect pin 1 (on the left) of the sensor to Vcc
// Connect pin 2 of the sensor to whatever your DHTPIN is
// Connect pin 4 (on the right) of the sensor to GROUND
// Connect a 10K resistor from pin 2 (data) to pin 1 (power) of the sensor

// This wrapper is in charge of calling
// must be defined like this for the lib work
void dht_wrapper() {
  dht.isrCallback();
}

float localhumidity;
float localtemperature;

void setup() {

  dht.acquire();

  ticker.attach(DHT_UPDATE_INTERVAL_SECS, setReadyForDHTUpdate);

}
void loop() {
  
  if (readyForDHTUpdate && !dht.acquiring()) {
    int acquireresult = dht.getStatus();
    if ( acquireresult == 0 ) {
      localtemperature = dht.getCelsius();
      localhumidity = dht.getHumidity();

    }
    readyForDHTUpdate = false;
  }
}

// Interrupt funktion
void setReadyForDHTUpdate() {
  Serial.println("Setting readyForDHTUpdate to true");
  dht.acquire();
  readyForDHTUpdate = true;
}
```

simonliu009, 2017-01-01:
I‘m having the same Exception(0) crash here. (ESP-12F with i2c LCD)
Running either the DHT21 code or WiFiManager code independently is fine , but it starts to crash when I put the codes together. So I guess it has nothing to do with the power supply.
And I tried both https://github.com/piettetech/PietteTech_DHT and https://github.com/olewolf/DHT_nonblocking and there is no luck.

todhunter, 2017-02-11:
Please use this to connect. I figured out that adding line WiFi.mode(WIFI_STA);
solves problem
My esp8266 hangs without this line every couple hours, after adding line it works without problem.
That line disables AP abilities of esp8266 
 ```
WiFi.disconnect();
WiFi.mode(WIFI_STA);
WiFi.config(ip, gateway, subnet);
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED) {}
```

hymby, 2017-02-11:
Kösz
  próbálom

devyte, 2017-10-18:
Before uploading a sketch that is different from a sketch already on the ESP, a full flash erase should be done.
I don't see anything obvious in this code, and the code base has changed significantly since it was opened. I am reading DHT22s on several ESP12-based boards without problems, so I don't think it's timing due to the reads.
Closing due to age and can't reproduce. If there is still a problem, please open a new issue and add the relevant info requested in the template, as well as updated info.

closed 2017-10-18 by devyte