esp8266/Arduino #1597: Trying to access ESP8266 web server by hostname from Windows

bmitov, 2016-02-07:
I have made a simple ESP8266 Web server and can access it from the browser in Winodows systems by IP.
Is there a way to access it by hostname? The Bonjour service is not available. Is there some DHCP or NetBIOS solution available ?
I tried also directly calling as example:

wifi_station_set_hostname( "test15" );

in the sketch but with no luck.

Adding:
espconn_mdns_set_hostname( "test15" ); 

results in :

'undefined reference to `espconn_mdns_set_hostname(char*)'

error when compiling :-( . Any other suggestions? I tried the mDNS but it requires installing Bonjour, and it is not an option.

WereCatf, 2016-02-07:
Windows can't magically know the IP-address of some random hostname, it's not omniscient. You either need a router that handles DHCP and DNS and remembers the hostnames DHCP-clients have supplied and then serves those hostnames via DNS, you need to manually set the IP-address for the hostname in c:\Windows\System32\Drivers\etc\hosts, or you need Bonjour.

Setting the hostname the ESP8266 uses only changes what hostname it reports to the DHCP-server.

igrr, 2016-02-07:
Another thing is that the call to wifi_station_set_hostname has to happen
before the ESP starts talking to the DHCP server. So make sure you call it
very early in your setup function.

WereCatf, 2016-02-07:
Aye, true, I forgot to mention that. Ie. you should set the hostname before Wifi.begin(), it doesn't help if you set it after the ESP has already gotten an IP-address.

bmitov, 2016-02-07:
I have a router (ASIS RTN66U) that handles domain names, and I have multiple named devices on the local network, that I can access by name. They discover each other by name. I can't get the ESP8266 to register itself in the router for some reason :-( .
I tried even calling:
wifi_station_set_hostname( "test11" );
first line in the setup()

and added:

extern "C" {
#include "user_interface.h"
}

but I still cant see the device.
I can ping it by IP:
ping 192.168.0.26

but I can't ping it by name:
ping test11

ot ping TEST11

don't work.
Now it is even more interesting... It seems to work up to some extend:
When I call:
nslookup 192.168.0.26

I get:
Server:  router.asus.com
Address:  192.168.0.1

Name:    TEST11
Address:  192.168.0.26

So it seems it at least is registering but the lookup is not working properly. Any ideas?
I am actually planning to write Instructable on solving this if I can find solution, as I think a lot of people need to be able to access device by name on DHCP networks.

bmitov, 2016-02-07:
Got it to work! :-)
The trick was adding "." at the end:
ping test11.

works!

closed 2016-02-07 by igrr

ahmad3ttallah, 2017-12-09:
Hello @bmitov did you manage to write an instructable for your solution?

bmitov, 2017-12-09:
Hello @ahmad3ttallah ,
Yes, it works on my router, however some people report that it does not work on some routers.
Here is the tutorial:
https://www.instructables.com/id/ESP8266-and-Visuino-GPS-Location-Web-Server-With-G/

ahmad3ttallah, 2017-12-10:
Thank you @bmitov ;)

saurabhreigns, 2017-12-18:
Well i am running a tcp server on 333 port of esp. while using a hotspot running on android device, a client can connect to this tcp server using ESPHOSTNAME:333 but on my router( netgear wsn 110 dual antenna), this doesn’t happen ANYMORE. Earlier , it was flawless. I have tested with some other routers too, almost with every router, this weird behaviour is persistent.
But if i create the wifi hotspot on an android device, connect esp and my other android device running tcp client on it, connection is always established and maintained as long as we keep sending data. 
Please help.
Thanks in advance .

Yeagob54, 2018-09-19:
> Got it to work! :-)
> The trick was adding "." at the end:
> ping test11.
> 
> works!

Thanks!!! 
It Works!!
You save me!!!