esp8266/Arduino #1598: Error compiling HTTPS Example

jorgegarciadev, 2016-02-07:
`./build.nodemcuv2/WiFiClientSecure.cpp.o:(.text+0x0): undefined reference to `ssl_write'
./build.nodemcuv2/WiFiClientSecure.cpp.o: In function `WiFiClientSecure::write(unsigned char const*, unsigned int)':
WiFiClientSecure.cpp:(.text+0x14): undefined reference to `ssl_write'
./build.nodemcuv2/WiFiClientSecure.cpp.o: In function `WiFiClientSecure::WiFiClientSecure()':
WiFiClientSecure.cpp:(.text+0x28c): undefined reference to `ssl_ctx_new'
./build.nodemcuv2/WiFiClientSecure.cpp.o: In function `WiFiClientSecure::WiFiClientSecure(WiFiClientSecure const&)':
WiFiClientSecure.cpp:(.text+0x290): undefined reference to `ssl_client_new'
WiFiClientSecure.cpp:(.text+0x294): undefined reference to `ssl_handshake_status'
./build.nodemcuv2/WiFiClientSecure.cpp.o: In function `WiFiClientSecure::operator=(WiFiClientSecure const&)':
WiFiClientSecure.cpp:(.text+0x2d0): undefined reference to `ssl_ctx_new'
WiFiClientSecure.cpp:(.text+0x30d): undefined reference to `ssl_client_new'
WiFiClientSecure.cpp:(.text+0x319): undefined reference to `ssl_handshake_status'
./build.nodemcuv2/WiFiClientSecure.cpp.o: In function `WiFiClientSecure::_connectSSL()':
WiFiClientSecure.cpp:(.text+0x340): undefined reference to `ssl_match_fingerprint'
WiFiClientSecure.cpp:(.text+0x3b7): undefined reference to `ssl_match_fingerprint'
./build.nodemcuv2/WiFiClientSecure.cpp.o:(.text._ZN10SSLContext5unrefEv[SSLContext::unref()]+0x0): undefined reference to `ssl_free'
./build.nodemcuv2/WiFiClientSecure.cpp.o:(.text._ZN10SSLContext5unrefEv[SSLContext::unref()]+0x4): undefined reference to `ssl_ctx_free'
./build.nodemcuv2/WiFiClientSecure.cpp.o: In function `SSLContext::unref()':
WiFiClientSecure.cpp:(.text._ZN10SSLContext5unrefEv[SSLContext::unref()]+0x20): undefined reference to `ssl_free'
WiFiClientSecure.cpp:(.text._ZN10SSLContext5unrefEv[SSLContext::unref()]+0x38): undefined reference to `ssl_ctx_free'
./build.nodemcuv2/WiFiClientSecure.cpp.o:(.text._ZN10SSLContext8_readAllEv[SSLContext::_readAll()]+0x0): undefined reference to `ssl_read'
./build.nodemcuv2/WiFiClientSecure.cpp.o: In function `SSLContext::_readAll()':
WiFiClientSecure.cpp:(.text._ZN10SSLContext8_readAllEv[SSLContext::_readAll()]+0x1f): undefined reference to `ssl_read'
WiFiClientSecure.cpp:(.text._ZN10SSLContext8_readAllEv[SSLContext::_readAll()]+0x3c): undefined reference to `ssl_free'
collect2: error: ld returned 1 exit status
make: *** [build.nodemcuv2/ds18b20.elf] Error 1`

Trying to compile HTTPSRequest.ino I get this error.

igrr, 2016-02-08:
Please tell us more about your setup (OS, Arduino IDE version, how you have installed the ESP8266 Arduino core, which version of the core do you use). This example sketch is part of our CI build, and usually we are notified when the build gets broken.
https://travis-ci.org/esp8266/Arduino/builds/107414537#L634

closed 2016-02-08 by jorgegarciadev

jorgegarciadev, 2016-02-08:
I forgot this flag `-laxtls`

kenpeter, 2016-08-12:
I was play around this example: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino

I need to compile like this.
`compiler.c.elf.libs=-lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -lsmartconfig -lmesh -lwpa2 {build.lwip_lib} -lstdc++ -axtls`

In another project, I need to remove `-axtls` and use `-lssl`
I was using mqtt with secure, https://github.com/i-n-g-o/esp-mqtt-arduino/tree/secure, it requires `-lssl`  

Is it a way for both of them to co exist? Right now it complain multiple definition.
`compiler.c.elf.libs=-lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -lsmartconfig -lmesh -lwpa2 {build.lwip_lib} -lstdc++ -lssl -axtls`