esp8266/Arduino #1609: Sketch uses too much space

tuxedo0801, 2016-02-08:
I just tried to evaluate how much space my code is using, and figured out, that even the simple "Blink" sketch from the examples uses 216.650 bytes --> 49% ... (Board is generic with 512K (64K SPIFFS))

Same example eats on an Arduino Leonadro just 4.454 bytes --> 15% ...

Tested with 2.1.0-rc2 ...

That's in my eyes very strange... Any ieas how to digg into this issue?

Links2004, 2016-02-08:
that is normal the 216.650 includes the full SDK with WiFi (including WPA, WPA2, WPS), OS, SSL and so one.
but you need to add many code to get the 512K full, and the most ESP8266 have now 1MB flash or more. 
a full VNC client for example needs only 253 KB.

jimibi, 2016-02-08:
The SDK takes a lot of space.

igrr, 2016-02-08:
Yep, there is a lot of code which is pulled in by the SDK even if you are not, in fact, utilizing WiFi and TCP stuff.

closed 2016-02-08 by igrr