esp8266/Arduino #1612: WiFi.softAP channel not honored in STA+AP

mytrain, 2016-02-09:
Hi

WiFi AP channel changes to the one used by the access point connected via the STA.

After initializing the AP on channel 6 everything until initialization of STA (STA+AP). Once connected to the STA access point the channel of the AP is switched to the one used by the STA access point and not the one provided as third parameter of WiFi.softAP(...)

I this a hardware limitation or a SDK bugg ?

igrr, 2016-02-09:
This is a hardware limitation, the chip can not maintain connection to AP on one channel while having SoftAP on another one.

mytrain, 2016-02-09:
@igrr 
OK fine.
Is there a section somewhere in the document where such 'limitation' are mentioned in order to raise a 'false' issue next time.

igrr, 2016-02-09:
There isn't, i think it was mentioned on the Espressif BBS somewhere, but can't find the link now.

mytrain, 2016-02-09:
OK Thanks anyway.
ISSUE CLOSED

closed 2016-02-09 by mytrain

Links2004, 2016-02-09:
docs:
https://github.com/esp8266/Arduino/issues/119#issuecomment-157418957
https://github.com/esp8266/Arduino/issues/119#issuecomment-157493799