esp8266/Arduino #1613: Could we get rid of the ArduinoOTA.handle() ?

mangelajo, 2016-02-09:
Looking at this,
https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/ArduinoOTA.cpp#L317

we could leave an empty .handle() for backwards compatibility,
and then handle updates also in the _Rx event.

Or allow/disallow such thing via a setting.

Any feedback on why this could be/not be a good idea?.

igrr, 2016-02-09:
`ArduinoOTA::_onRx` is called from the network task. Therefore we can't use blocking functions (i.e. WiFiClient::read and WiFiClient::write) in any task other than the main task, so calling `_runUpdate` from `_onRx` is not possible.

mangelajo, 2016-02-09:
Ahh, ok @igrr , understood.

Thanks for the feedback. I guess the only way to do such thing then, would be to keep a list of transparent loop callbacks to be called on every "loop" call, and having the ArduinoOTA.begin() register such callback.

But not sure how reasonable that would be.

igrr, 2016-02-09:
Yeah, that would work. We have an issue opened for that: https://github.com/esp8266/Arduino/issues/1064.

igrr, 2016-02-09:
Actually, this won't work if one does all the work inside `setup` method and never reaches `loop`.

mangelajo, 2016-02-09:
Oh, right, but, ok, we can still allow people to use .handle in setup if they need to do such thing.

closed 2016-02-09 by mangelajo