esp8266/Arduino #1614: Doesn't see any AP with Arduino IDE [Ai-Cloud inside ESP8266 - Black]

molakttj, 2016-02-09:
I've got the problem with my black ESP8266 - When I run the code which turns the led on and off everything works ok, but when I'm trying to scan available APs or when I'm trying to connect to one of them, then it doesn't work. However, I could list APs and connect to them with standard AT firmware, which my module came with. Has anybody have the same issue?

krzychb, 2016-02-09:
Hi molakttj,

I understand that:
* If you are able to flash ESP with Arduino IDE
* Blink.ino from Arduino examples works correctly
* You are able to scan for APs and connect using AT firmware 

If this is the case then you are likely stuck on some simple issue and well positioned to quickly move forward :smile:

I would load File > Examples > ESP8266WiFI > WiFiScan.ino sketch and post what is shown on your serial monitor. 

I am using Arduino IDE 1.6.7 together with core ver. 2.0.0.
AP scanning works correctly for me.

Krzysztof

molakttj, 2016-02-10:
That's what I see on my serial monitor after flashing ESP8266 with WiFiScan program:
```
scan start
scan done
no networks found
```
So that's the problem. I'm sure, that there are lots of wifi signals in the area.

krzychb, 2016-02-10:
Hi molakttj,

This may be the case if you are using a staging release like 2.0.0-rc1. Please check [this issue](https://github.com/esp8266/Arduino/issues/1355#issue-124615278).

I did my testing using stable 2.0.0.

Krzysztof

molakttj, 2016-02-10:
But my module also doesn't work for WiFiAccessPoint.ino.

krzychb, 2016-02-10:
molakttj,

I see two possible issues:

1. Basing on your statement "I'm powering it through FTDI",  depending on what FTDI module you have, it may not deliver enough current to power Wi-Fi circuity. Before saying "but it works with AT firmware", just verify if it is able to reliably deliver >= 250mA and has decoupling caps. :smile:
2. Corrupt esp8266 package installation. The simple solution is to delete Arduino15 folder and then to reinstall the package (starting with stable 2.0.0). Note - this folder contains other packages like Arduino DUE (sam)

Good luck and let me know results of troubleshooting :+1:

Krzysztof

closed 2016-06-23 by igrr

ramanc51, 2018-10-09:
I am also facing similar issue, ESP01 module(Ai Cloud inside) accept program but not showing it's AP name received 10 modules from Chinese company all not working, how to resolve, i thing this issue relevant to power supply circuit.