esp8266/Arduino #1616: Crash using AsyncWebServer with SPIFFS

mozgy, 2016-02-09:
Serving jpegs from SPIFFS -

```
Decoding 29 results
0x4000df64: ?? ??:0
0x4000df64: ?? ??:0
0x4020630d: spiffs_phys_rd at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\spiffs/spiffs_cache.c:145
0x40205ef4: spiffs_object_read at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\spiffs/spiffs_nucleus.c:1681
0x40203155: SPIFFS_read at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\spiffs/spiffs_hydrogen.c:872
0x4020b96a: AsyncClient::write(char const*, unsigned int) at C:\Users\Mozz\Documents\Arduino\libraries\ESPAsyncTCP\src/ESPAsyncTCP.cpp:506 (discriminator 4)
0x40227710: etharp_output at ??:?
0x402129ed: SPIFFSFileImpl::read(unsigned char*, unsigned int) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/spiffs_api.cpp:284
0x402139f0: fs::File::read(unsigned char*, unsigned int) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/FS.cpp:182
0x4020c31e: AsyncFileResponse::_fillBuffer(unsigned char*, unsigned int) at C:\Users\Mozz\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebServerResponseImpl.h:77
0x4020c927: AsyncAbstractResponse::_ack(AsyncWebServerRequest*, unsigned int, unsigned int) at C:\Users\Mozz\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebServerResponseImpl.h:77
0x4010020c: _umm_free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c:1281
0x4010020c: _umm_free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c:1281
0x40100aec: free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c:1727
0x4022c0c1: tcp_rexmit_rto at ??:?
0x402137f8: AsyncWebServerRequest::_onAck(unsigned int, unsigned int) at C:\Users\Mozz\Documents\Arduino\libraries\ESPAsyncWebServer\src/WebServerClient.cpp:145
0x4020ddbd: _M_invoke at C:\Users\Mozz\Documents\Arduino\libraries\ESPAsyncWebServer\src/WebServerClient.cpp:784
0x40229cbe: tcp_input at ??:?
0x4020b8ee: AsyncClient::_sent(tcp_pcb*, unsigned short) at C:\Users\Mozz\Documents\Arduino\libraries\ESPAsyncTCP\src/ESPAsyncTCP.cpp:506 (discriminator 4)
0x4020b90c: AsyncClient::_s_sent(void*, tcp_pcb*, unsigned short) at C:\Users\Mozz\Documents\Arduino\libraries\ESPAsyncTCP\src/ESPAsyncTCP.cpp:506 (discriminator 4)
0x40229779: tcp_input at ??:?
0x40229834: tcp_input at ??:?
0x402286d2: ip_input at ??:?
0x40227467: etharp_find_addr at ??:?
0x40227a12: ethernet_input at ??:?
0x402333b7: ets_snprintf at ??:?
0x402124d7: loop_task at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp:43
```

wireshark included ..
[arducam-crash-wshark.txt](https://github.com/esp8266/Arduino/files/123417/arducam-crash-wshark.txt)

andig, 2016-02-13:
See https://github.com/esp8266/Arduino/issues/1576#issuecomment-180286280 too. I'm also seeing crashes when loading the async server, spiffs or not. This seems according to @igrr related to lwip compiler issues.

Would need gdb'ing the relevant crash point.

closed 2016-06-02 by igrr

HugoSH, 2017-11-21:
Not sure if this has been resolved. 
I am using the AsyncWebServer and SPIFFS. 
Write to SPIFFS works ok. 
But when Read a file from SPIFFS, it crashes. 
Does this have been fixed?               Thanks !

Panic /Users/hugo/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp:98 __yield

ctx: sys 
sp: 3ffffb60 end: 3fffffb0 offset: 01b0

>>>stack>>>
3ffffd10:  3fff194c 00000001 3ffffd78 4020fc5d  
3ffffd20:  0000000a 3ffffd60 3ffffd78 4020e8ad  
3ffffd30:  3ffe0029 00000049 3ffffd60 4020dd11  
...
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <FS.h>  
#include <time.h>

AsyncWebServer server(80);

void indexLoad(AsyncWebServerRequest *request) ;
void indexSubmit(AsyncWebServerRequest *request) ;

const char* ssid = "ChinaNet-tz4i";
const char* password = "qfk9n6en";
const char * hostName = "ESP-async";

void setup() {
  Serial.begin(115200);
  Serial.println(" Async WebServer Test "); 
  Serial.setDebugOutput(true);
  WiFi.hostname(hostName);
  WiFi.mode(WIFI_AP_STA);
  WiFi.softAP(hostName);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    delay(500);
    WiFi.begin(ssid, password);
  }
  Serial.print("STA mode IP address:"); 
  Serial.println(WiFi.localIP());
    if(SPIFFS.begin()){
           Serial.println("SPIFFS mount success."); 
           server.serveStatic("/scripts", SPIFFS, "/scripts", "max-age=86400");
     }
    else  Serial.println("SPIFFS mount failure.");   
  
  MDNS.addService("http","tcp",80);

  server.on("/", HTTP_GET, indexLoad);
  server.on("/index", HTTP_GET, indexLoad);
  server.on("/index", HTTP_POST, indexSubmit);
  server.begin();  
}

void loop() {

}

void indexLoad(AsyncWebServerRequest *request) {
      Serial.println("\nShow index.html ");
      request->send(SPIFFS, "/index.html");
}

void indexSubmit(AsyncWebServerRequest *request) {           
      String inputValue, tempStr;

      Serial.println("index Page Submit()");
      inputValue = request->arg("inputTxt");
      File registerFile = SPIFFS.open("/scripts/registration.js", "r");   //crashes here
      if(!registerFile){
        Serial.println("Failed to Read /scripts/registration.js ");
      }
      else{
                while(registerFile.available()){
                    tempStr += registerFile.readStringUntil('\n');
                }
                registerFile.close();    
                Serial.println(tempStr);
      }       
      Serial.println("index page submit() End");     
}

me-no-dev, 2017-11-21:
I answered on the Async repo issue that you opened.