esp8266/Arduino #1619: SPIFFS bin file web flashing using the update sketch

nccchirag, 2016-02-10:
modified the flash update sketch to web update **SPIFFS** but I'm getting the following error. Although Update sketch works fine for firmware flashing
-------------
change 
-------------
if(!Update.begin(maxSketchSpace, **100**)){
          Update.printError(Serial);
        }
-------------------
Serial Log / ERROR
-----------------
Update: firmware.spiffs.bin
sleep disable
**ERROR[10]: Magic byte is wrong, not 0xE9**
bcn 0
del if1
usl

generating the SPIFFS file using the jar tool ( latest version )

Kindly guide!

Links2004, 2016-02-10:
what version of the core you use?
may its the same as #1499?

nccchirag, 2016-02-11:
True!
It's the same issue .. I'm using the latest rc1 core

nccchirag, 2016-02-11:
my bad! ... command sent while upload was always 0 instead of 100
its working fine with the latest core

closed 2016-02-11 by nccchirag