esp8266/Arduino #1622: How to do a software reset

odilonafonso, 2016-02-11:
Does anyone know how to reset the ESP8266 ESP-01 for software?
Constantly to connect the module, it starts with the blue LED lit and does not connect.
I can make it work through a switch sending to ESP-01 RST pin LOW for a short period of time and then leave it on the 3.3V or simply floating (not figured out the best way)
In an Arduino ProMini ATMega328 using the ESP-01 as WiFi via AT firmware, tried to simulate this operation through the code:
```
#define RST 4
softReset void () {
   pinMode (RST, OUTPUT);
   digitalWrite (RST, LOW);
   delay (500);
   digitalWrite (RST, HIGH);
}
```
In this case, the ESP-01 RST is connected to the switch via a resistor .5K from the Arduino pin 4 and a 1K resistor in the ground. This is intended to simulate the pressure switch (which connects to ground RST)

But not always work. I can not 100% positive to restart the module.

Any idea how I can do this?

Thank you.

hallard, 2016-02-11:
You can try `ESP.reset();` but it will not be a Hard Reset

odilonafonso, 2016-02-12:
Hi. What lib is ESP ?

hallard, 2016-02-12:
It's native and included to this dev environnement, you have nothing to do to use it

it's in core subfolder
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.cpp

odilonafonso, 2016-02-12:
My god .... other libraries more !!
Arduino.h? HardwareSerial ?? Stream ???
These libraries all comes with the Arduino IDE !!
What makes this repository - replaces the original Arduino libraries?
I already have a lot of things in duplicate in my environment.
I do not have that EspClass in my environment. Are even necessary that lot of libraries?
Thank you for your help, Hallard.

odilonafonso, 2016-02-12:
It's getting increasingly difficult to manage the libraries ESP8266 environment with Arduino.

odilonafonso, 2016-02-12:
Hi Hallard,
I think the solution you proposed to me will not work. 
I am using a ArduinoProMini ATMega328 board to make the IoT jobs, and a ESP8266 ESP-01 board only to make the WiFi connection. With the AT firmware.

 * AT version:0.50.0.0(Sep 18 2015 20:55:38)
 * SDK version:1.4.0 BOOT MODE
 * compile time:Sep 18 2015 21:46:52

I use the "ESP8266.h" library, supplied by the ITEADLIB ESP8266 library:

https://github.com/itead/ITEADLIB_Arduino_WeeESP8266. 

I would love to have a software function for resetting the ESP-01 module, since the functions for this purpose exist in ITEAD library do not always work.
esp8266.restart () or AT + RST not work very well, sometimes stoping, or slow to return.
It seems that the community that develops routines support ESP8266 module decided that the best way to work is without Arduino, only ESP8266.
I have not come to the same conclusion.

Links2004, 2016-02-12:
this is no library,
the repo here is to run code direct on the ESP8266 not an any AVR controller.
you not need any other chip/controller then the ESP module itself.

odilonafonso, 2016-02-14:
OK, I figured it out. But...
I find it easier to find 5V devices on the market. Especially here in the country where I live .. Brazil.
And I like the stability and security that the years of development and huge community around the Arduino offer us.
I know ESP8266 will soon become a mighty competitor Arduino, but for now I prefer to see them as partners.
The solution ArduinoProMini and a ESP8266 (together) provide seem well suited to the development of a number of very interesting products.
I have tested in parallel the ESP8266 working alone. I'm checking using the ESP12 and with NodeMCU firmware.
And I am very surprised (and grateful) to see how the ESP8266 community is active and working. This makes me very optimistic about the future of this component.
Thank you all for your help and comments!

closed 2016-02-29 by igrr

slavino, 2017-01-03:
I will also most probably redesign my home automation devices. Experiencing same situation using ESP-01 as WiFi UART for Arduino Pro Mini. But it seems to be wiser to handle connectivity and init communication from ESP-01 module and let Arduino be the Serial attached "slave" grabbing the sensor data. This could mean only SW refactor without need to touch the HW setup.

curlyz, 2017-03-10:
@odilonafonso  surely one day man , you will want to do everything with the ESP for IoT, only use Atmega328 as a bridge for some IC.

logiclayer, 2017-11-27:
I also tried many commands such as ESP.restart() and ESP.reset() after than connecting pins RST and D0. but it did not work for me. So very simple solution is working for me now;
`void setup() {
  pinMode(D0, OUTPUT);
  digitalWrite(D0, HIGH);
}`

and I take D0 pin to LOW whenever I need like;
`digitalWrite(D0, LOW);
 delay(100); //After than reset no need to delay but I put it for fun :)
`

brianjmurrell, 2018-03-30:
When I try to use `ESP.reset()` it prints this on the serial port:

```
 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

but then doesn't actually reset.  I need to press the reset button to get it reset.

simar0511, 2018-05-21:
Hi brianjmurrell, 
I am also getting same issue. can you help me out how you resolved this issue.

Regards
Simarjeet Singh

basavarajum2010, 2018-06-11:
Hi, 
Am new to using nodemcu esp8266. when I connect modemcu to usb port and execute simple Arduino sketch ( set baud rate and mode to station or access point), boot mode: (1,4) and boot mode: (1,6) error appears. 
 No code after this executes.
The code is not executing even after pressing reset button on the board and re-executing. I request any help available to understand and execute this .

brianjmurrell, 2018-06-11:
@basavarajum2010 Your question is off-topic for this issue.  It's rude and disrespectful to hijack other people's issues, so please don't do it.  If you have an issue of your own (and I don't suspect you actually do -- you need to find a help forum) please open a new issue (but as I said, your problem is not an issue for this project but is a more general help question, I think).

If the owner of this project would like to delete this and the previous comment, I would most certainly not object.

dandywing, 2019-02-01:
I found out.
I do not know why.
After uploading the code
You must press hardware reset button once unconditionally
That way,
The code succeeds normally.