esp8266/Arduino #1623: Problem with SPIFFS and ESP12F

acca84, 2016-02-11:
Hello,

I'm using ESP12F with Arduino IDE. Everything works well excepted SPIFFS, i use "ConfigFile" example.

If i choose the board Generic espmodule 4M (3Mspiffs), i obtain this error :

```
Mounting FS...

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

If i choose other memory parameters, i obtain :

```
Mounting FS...

Exception (4):
epc1=0x400005cb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffecdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffecf70:  40101acd 40101aca 00000000 00000000  
3ffecf80:  400005e1 00000000 00000000 00000000  
3ffecf90:  4020dd1c 00000033 00000015 00000000  
```

Could someaone help me to solve this problem ?

acca84, 2016-02-11:
I found the solution : i had to create FS using https://github.com/esp8266/arduino-esp8266fs-plugin

Regards

toyan, 2016-02-12:
You must format SPIFFS before mounting. This can be done right from sketch: SPIFFS.format()

closed 2016-02-29 by igrr