esp8266/Arduino #1624: WIFI_AP_STA can cause AP too slow and not working if it can't connect to STA

hallard, 2016-02-11:
Following this case #1615 I'm opening this dedicated one to avoid mixing issue.

To isolate and see the issue, I've done a sketch that switch Wifi autoConnect parameter back and forth (I'm using NodeMCU board with flash button, pushing button toggle autoConnect on and off). But as changing autoConnect in loop does nothing (SDK says) so on each change we need to reboot the board, that what's the sketch do.
What is clear is that if autoConnect is on and and Wifi STA SSID/PSK does not connect (wrong SSID/PSK or main router AP down), the problem occurs in AP mode (may be take too much time trying to auto connect to wrong router)
And, as soon as autoConnect is disabled, AP is working fine, so I suspect autoConnect trying to connect cause issue in AP code.

But there is a 2nd point, all I've just written is true only if you call ESP.eraseConfig() at the beginning of setup(). If you don't do that, AP is slow and broken whatever autoConnect is on or off. This means that SDK is may be storing something else somewhere used in conjunction with autoConnect/SSID/PSK.

Note also that `ESP.eraseConfig()` preserve autoConnect flag ;-)

Now the real problem is that when I'm at home, my NodeMCU Wifi connect to my main router and AP is up for 5 minutes for smartconfig in case we want to do so. Now I want to take my nodeMCU to  another place for a customer demo. Of course It can't connect to main Wifi because I'm not at home, but the issue if that I see AP but once connected it's too slow, I miss about 50% of ping packet and thus I can't do anything, my web page are not coming and I can't config the NodeMCU to connect to customer Wifi.

Here is the sketch used and step to reproduce issue
- burn it to a fresh ESP8266 (or use esptool.py with erase_flash option before)
- once launched use serial monitor
- after starting loop displayed, press flash button, this will disable autoConnect and reboot
- connect your computer to new AP `ESP-TESTAP`
- from your computer ping 192.168.4.1 => works fine
- now press flash button again , this will enable autoConnect and reboot
- connect your computer to new AP `ESP-TESTAP`
- from your computer ping 192.168.4.1 => does not work ping slow and missed packet

```arduino
#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>

#define SERIAL_DEBUG  Serial

bool autoConnect;
int buttonState;            
int lastButtonState = LOW;  
long lastDebounceTime = 0;  
long debounceDelay = 50;    // the debounce time

// NodeMCU led is GPIO16 and switch (Flash) on GPIO 0
uint8_t handle_key_led( uint8_t level)
{
  uint8_t temp;
  digitalWrite(16, 1);
  pinMode(16, OUTPUT);
  digitalWrite(16, !level);

  // Flash is on GPIO0
  pinMode(0, INPUT);
  temp = digitalRead(0);
  pinMode(0, OUTPUT); 
  return !temp;
}

// ==============
void setup() {
  int ret;
  // without delay sometime you got infinite reset loop 
  // (SDK config corrupted ???) which only a erase_flash can recover
  delay(500);
  ESP.eraseConfig();
  delay(500);
  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_AP_STA);
  SERIAL_DEBUG.begin(115200);
  SERIAL_DEBUG.println("\r\n========== SDK Saved parameters Start"); 
  WiFi.printDiag(SERIAL_DEBUG);
  SERIAL_DEBUG.println("========== SDK Saved parameters End"); 
  SERIAL_DEBUG.println("Connecting...");

  // Get current autoconnect state from SDK
  autoConnect = WiFi.getAutoConnect();
  WiFi.begin("DummySSID", "whateverkeyjustdontconnect");

  uint8_t timeout = 10; // 10 * 500 ms = 5 sec time out
  while ( ((ret = WiFi.status()) != WL_CONNECTED) && timeout-- ) {
    SERIAL_DEBUG.print(".");
    delay(500);
  }

  // connected ? disable AP, client mode only
  if ((ret = WiFi.status()) == WL_CONNECTED) {
    SERIAL_DEBUG.println("connected!");
    SERIAL_DEBUG.print("STA IP address "); 
    SERIAL_DEBUG.println(WiFi.localIP());
  // not connected ? start AP
  } else {
    //WiFi.mode(WIFI_AP);
    SERIAL_DEBUG.println("Configuring access point...");
    WiFi.softAP("ESP-TESTAP","allowmeistooshort");
    WiFi.softAPIP();
    SERIAL_DEBUG.print("AP IP address "); 
    SERIAL_DEBUG.println(WiFi.softAPIP());
  }

  SERIAL_DEBUG.println("Starting loop");
}

// ==============
void loop() {
  uint8_t btn = handle_key_led(autoConnect);

  // Debouncing
  if (btn != lastButtonState) {
    lastDebounceTime = millis();
  }

  if ((millis() - lastDebounceTime) > debounceDelay) {
    // if the button state has changed:
    if (btn != buttonState) {
      buttonState = btn;

      // Only on button pressed, not released
      if (buttonState == HIGH) {
        autoConnect = !autoConnect;
        WiFi.setAutoConnect(autoConnect);
        SERIAL_DEBUG.print("Autoconnect will be now ");
        SERIAL_DEBUG.println(autoConnect);
        SERIAL_DEBUG.println("Restarting...");
        delay(500);
        // Changing Auto Connect in loop does nothing
        // it will be taken into account on reboot only 
        // see SDK wiﬁ_station_set_auto_connect 
        ESP.reset();
      }
    }
  }

  lastButtonState = btn;
}
```

Hoping some SDK's guru will be able to help me to understand this strange behavior

far5893, 2016-02-11:
I notice the anomaly randomly

Links2004, 2016-02-11:
two short notes:
```ESP.eraseConfig()``` take only a real effect when you do a forced reboot after it ```ESP.reboot();```
or the SDK may store back some data from ram later, but not all this can lead to a undefined state (may the reason for your infinite reset loop).

for the AP problem have you tried to use a other WiFi channel for the AP.
may the scanning for your wifi (fast channel switching) is the reason for your package lost, have read something in the SDK long ago. 
have add it too a issues to this time need to search it.

Links2004, 2016-02-11:
here it is (the second half)
https://github.com/esp8266/Arduino/issues/119#issuecomment-157493799

hallard, 2016-02-11:
@Links2004 
whooo, what's an amazing post!!!
So I think you're right, since it can't connect to router, it tries on every channel and this cause AP issues due to channel changes, makes sense.

May be algorithm could be changed on SDK side, IE as SSID/PSK are stored in flash, and may be channel also, perhaps SDK could :
- Try to connect to saved channel 
- if can't, try other all channels in case router as changed
- if can't, ok we mainly have a problem (router off or ESP moved away), so let autoConnect activated but only on saved channel this will let AP mode working fine and also allow reconnect if router is back (on saved channel only)

Links2004, 2016-02-13:
you can do the scan your self then your have full control over it.
basicly:
 - ```WiFi.disconnect();``` /// SDK not longer try to connect any where
 - ```WiFi.scanNetworks(true);``` /// to trigger scan when you think it will not block system
 - if you see your router connect to it with  ```WiFi.begin```

its more or less the way my WiFi Multi code work, it scans for wifi on every ```run``` when its not connected.
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp

hallard, 2016-02-13:
Markus,
Nice idea, I like it ;-)

dreamer2, 2016-03-21:
i have same problem with AP instability when STA tries to connect to nonexistent network
(incorrect password not lead to the problem)

sdk based webserver https://github.com/pvvx/esp8266web/ have "reconnect" delay and seems this parameter solves this problem

pikolo84, 2017-05-04:
You could use WiFi.setAutoConnect(!WiFi.softAPgetStationNum()); and if some client connect to ESP_AP disable autoreconnect.

devyte, 2017-10-18:
Channel hopping is known, behavior is expected and due to STA and SoftAP sharing a single radio.
Closing.

closed 2017-10-18 by devyte

me21, 2017-10-24:
Does WiFi.scanNetworks also cause channel hopping and associated clients disconnect?