esp8266/Arduino #1628: interrupt only working when digital pin set to output

Michvw, 2016-02-12:
Hi,
I've got the weirdest thing, whenever is set a digital pin to input i can't get the interrupt to work but just a minute ago i found out (by accident) that when i set the digital pin to output the interrupts work just fine! I'm really not kidding, i'm using a nodeMCU v1.0 board with the esp8266 community version 2.0.

`#include <ESP8266WiFi.h>
#include <PubSubClient.h>

const char* ssid     = "XXXXXX";
const char* password = "XXXXX";
const char* mqtt_server = "XXXXXX";
const int pin1 = 4;
const int pin2 = 5;
const int pin3 = 12;
const int pin4 = 13;
const int pin5 = 14;

WiFiClient espClient;
PubSubClient client(espClient);

void statement_pin1() {
  Serial.println("reached statement pin 4!");
}
void statement_pin2() {
  Serial.println("reached statement pin 5!");
}
void statement_pin3() {
  Serial.println("reached statement pin 12!");
}
void statement_pin4() {
  Serial.println("reached statement pin 13!");
}
void statement_pin5() {
  Serial.println("reached statement pin 14!");
}

void setup() {
  pinMode(pin1, OUTPUT);
  pinMode(pin2, OUTPUT);
  pinMode(pin3, OUTPUT);
  pinMode(pin4, OUTPUT);
  pinMode(pin5, OUTPUT);
  attachInterrupt(pin1, statement_pin1, RISING);
  attachInterrupt(pin2, statement_pin2, RISING);
  attachInterrupt(pin3, statement_pin3, RISING);
  attachInterrupt(pin4, statement_pin4, RISING);
  attachInterrupt(pin5, statement_pin5, RISING);
  
  Serial.begin(115200);
  delay(10);    
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(200);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
}

void loop() {
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
  yield();
}` 

as you can see i'm using some simple wifi mqtt code to try some stuff out. I'm happy i got the interrupts to work (after 2 days!) but i can't wrap my head around it. Anyone knows what happening here?

WereCatf, 2016-02-12:
You're attaching interrupts on the pins and only trigger them on RISING -- you do know that when you set the pin as INPUT it is HIGH by default and you need to attach an interrupt on FALLING or CHANGE? Obviously, you won't get an interrupt when the pin is already HIGH and whatever you've attached on it tries to pull it HIGH since nothing changes.

Michvw, 2016-02-12:
i had no idea... but what is i want to have an interrupt when a signal becomes HIGH? like a PIR?

WereCatf, 2016-02-12:
All the Arduino PIR - modules I know of pull the pin LOW when they're triggered.

Michvw, 2016-02-12:
it was just an example, but nevermind. I suppose if i ever need a interrupt when a signal becomes HIGH this is a way to do it. Thanks!

closed 2016-02-12 by Michvw

WereCatf, 2016-02-12:
If you really need a pin that is pulled LOW by default you could use GPIO16 for that, e.g. enable PULLDOWN-resistor with _pinMode(16, INPUT_PULLDOWN_16);_

Only GPIO16 has a PULLDOWN-resistor, all the others have a PULLUP-resistor. Or you could just add a physical PULLDOWN-resistor instead, then you could get an interrupt when the pin is pulled HIGH.

Hope this helps.