esp8266/Arduino #1631: problem with compilation of example sketch "Blink"

koliplus, 2016-02-14:
Hello,

When I try to compile example "Blink" from File > Examples > ESP8266 > Blink I have an error: 
`In file included from c:\documents and settings\user\ustawienia lokalne\dane aplikacji\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:62:0,`

`                 from C:\Documents and Settings\user\Ustawienia lokalne\Dane aplikacji\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266\Esp.cpp:24:`

`c:\documents and settings\user\ustawienia lokalne\dane aplikacji\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:59:28: fatal error: bits/c++config.h: No such file or directory`

` #include <bits/c++config.h>`
`                            ^`

`compilation terminated.`

`exit status 1`

I checked it, It is true, in specified folder file "c++config.h" is actually missing. 
What is wrong ? Incomplete ESP library package ... ? 

Regards

koliplus

_OS: Windows XP, CPU Athlon XP 2600+_
_IDE: Arduino 1.6.7_
_ESP library: v.2.0.0_

ikbelkirasan, 2016-02-17:
Maybe you need to reinstall ESP8266 package

igrr, 2016-02-29:
Seems to be the same issue as #1698.
The file should actually be located in `c:\documents and settings\user\ustawienia lokalne\dane aplikacji\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits\c++config.h`

The issue may be due to how Arduino IDE or gcc deal with international characters on Windows.

BorgMcz, 2016-02-29:
Yes installations are new and the same procedure.
Yes, I tend to this reason, the various pathways.  [#1698](https://github.com/esp8266/Arduino/issues/1698)
Is there any solution?

igrr, 2016-02-29:
Could you please enable verbose compiler output (in Arduino Preferences) and copy&paste the full output?

BorgMcz, 2016-02-29:
I attach the full output of the compiler.
[debug.txt](https://github.com/esp8266/Arduino/files/151508/debug.txt)

` Arduino: 1.6.7 (Windows XP), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

"C:\Documents and Settings\uzivatel\Local Settings\Data aplikacĂ­\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\uzivatel\Local Settings\Data aplikacĂ­\Arduino15\packages\esp8266\hardware\esp8266\2.1.0/tools/sdk/include" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L   -DARDUINO=10607 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Documents and Settings\uzivatel\Local Settings\Data aplikacĂ­\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266" "-IC:\Documents and Settings\uzivatel\Local Settings\Data aplikacĂ­\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\variants\nodemcu" "C:\Documents and Settings\uzivatel\Local Settings\Data aplikacĂ­\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\Esp.cpp" -o "C:\DOCUME~1\uzivatel\LOCALS~1\Temp\build7c955e20053892e726efe23569c11769.tmp\core\Esp.cpp.o"
In file included from c:\documents and settings\uzivatel\local settings\data aplikací\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:63:0,

                 from C:\Documents and Settings\uzivatel\Local Settings\Data aplikací\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\Esp.cpp:24:

c:\documents and settings\uzivatel\local settings\data aplikací\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\allocator.h:46:75: fatal error: bits/c++allocator.h: No such file or directory

 #include <bits/c++allocator.h> // Define the base class to std::allocator.

                                                                           ^

compilation terminated.

exit status 1
Error compiling.`

koliplus, 2016-02-29:
For me, the compilation messages are slightly different - see atachment:
[compilation-messages.txt](https://github.com/esp8266/Arduino/files/151824/compilation-messages.txt)

For me error is generated from ```c:\documents and settings\user\ustawienia lokalne\dane aplikacji\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h```

I have a "Generic ESP8266 Module" selected -  maybe that is a difference reason ...

BorgMcz, 2016-03-02:
At least a partial solution to the PORTABLE installation Arduino IDE. By circumventing the problem of localized versions of Windows XP.

This solution is not out of my head. The same problem to solve here:
http://digistump.com/board/index.php?topic=2052

RobertClemenzi, 2016-03-04:
I have the same problem, but with `cpu_defines.h`. I checked, the file is there, but "not found". In the following, I replaced my username with 'xxx'.

```
Using previously compiled file: C:\DOCUME~1\xxx\LOCALS~1\Temp\build897163bd4f836f68a68b59a40965d35b.tmp\core\spiffs\spiffs_nucleus.c.o
"C:\Documents and Settings\xxx\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\xxx\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Documents and Settings\xxx\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Documents and Settings\xxx\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\generic" "C:\Documents and Settings\xxx\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266\Esp.cpp" -o "C:\DOCUME~1\xxx\LOCALS~1\Temp\build897163bd4f836f68a68b59a40965d35b.tmp\core\Esp.cpp.o"
In file included from c:\documents and settings\xxx\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:59:0,

                 from c:\documents and settings\xxx\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:62,

                 from C:\Documents and Settings\xxx\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266\Esp.cpp:24:

c:\documents and settings\xxx\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits\c++config.h:429:30: fatal error: bits/cpu_defines.h: No such file or directory

 #include <bits/cpu_defines.h>
                              ^

compilation terminated.

exit status 1
Error compiling.
```

This is the file on my system that I think it is looking for

> C:\Documents and Settings\xxx\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits\cpu_defines.h

Using Windows XP, English, Arduino 1.6.7 IDE, downloaded board manager 02-25-16. I have tried Generic and both NodeMCU selections.

RobertClemenzi, 2016-03-05:
I commented out the call for the missing file and recompiled, yet another file in the same directory was "missing". After several failed experiments, including updating to the current version, I copied the contents of 
```
esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits
```
to
```
esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits
```
and now blank sketches compile without error and I was able to compile, upload, and run **ESP8266 Blink**.

RobertClemenzi, 2016-03-06:
I installed the same 1.6.7 IDE and ESP8266 board manager on a Vista system, and a blank sketch compiled without a problem.

hr3, 2016-03-17:
I had a similar problem with XP and IDE 1.6.8.
After a portable Installation like http://www.instructables.com/id/Portable-installation-guide-of-Arduino-IDE-v165-fo/?ALLSTEPS it works.
In a Windows 10 environmet the standard installation works without problems.

NigelTwo, 2016-03-27:
@hr3. Thanks, the Portable installation fix worked for me too on XP and IDE 1.6.8.

richard78, 2016-04-12:
WINXP ,the length of path limited by 260 char

RobertClemenzi, 2016-04-13:
One of the failing files had a full path of 210 characters. So perhaps that is not the problem.

krzbor, 2016-04-18:
I am waiting for the module ESP-01 but I tested compilation and I had a similar error (Windows XP). The problem was solved by RobertClemenzi (for me).

rgwood1234567890, 2016-11-22:
This is still an issue. Fixed with RobertClemenzi 's suggestion. Win7, 64bit, Eclipse neon (latest updates), Arduino plugin 3.1.0.201608012005. Blink sketch. NodeMCU 2.3.0.

I strongly suggest using the shortest pathnames possible and installing close to "root" (on whichever OS) as filesystems are awful, awful naming systems.

OrdevSoftware, 2017-04-24:
It can be SOLVED.

This bug only happens on XP (but don't know why). The problem is ocurred when, in Windows XP OS, you installed 'boards esp8266' in Arduino IDE. They indicated that you have to go to preferences and write:
 'http://arduino.esp8266.com/stable/package_esp8266com_index.json'

Well, when you do this ... in your computer doesn't install correctly the packages (it ocurrs in Windows XP OS).

So ... you can go to the directory 'c:\documents and settings\...\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2' and here the solution:

if you copy the whole contents of the folder 4.8.2\xtensa-lx106-elf\bits to 4.8.2\bits you should not receive any overwriting messages.

rgwood1234567890, 2017-04-24:
This happens in win7 too.

OrdevSoftware, 2017-04-24:
I have just made ... and it worked with board WEMOS D1 R2. 

I have copied the libraries .h from the folder **4.8.2\xtensa-lx106-elf\bits** to **4.8.2\bits** and It solved.

By the way, ... in Win7 64 bits, I don't remember about this problem. I think that it worked correctly without problems.

camp0s, 2017-05-28:
Same problem on WinXP and Win7, just rienstalled the package 2.3.0 on both systems and getting the same problem compiling the Blink program. Fixed with RobertClemenzi 's solution :)

sochs, 2017-06-07:
Would it be possible that the file path is just too long? I wanted to use the portable version and had the same problem (Win10). After I put the complete arduino folder directly under C:\ the problem was gone.

hr3, 2017-07-07:
Could it be, there are different problems?
1. take version 2.3.0
2. select Board: NodeMCU xxx
3. on NodeMCU (1.0) the blue LED is on pin 2

SaeedMardani, 2017-07-07:
@hr3 
Hi

Thanks for your attention. It seems that it was the problem with my settings with Arduino IDE. Recently,I was working on another project combining Matlab and Arduino. As a result the preferences of the IDE was set to get some files from a Matlab folder (very annoying, why should this happen, it took me several hours to fix it). So, it was making an error because it could not find the files from the original source. I needed to reinstall the Arduino IDE and remove the preferences text file located in  
C:\Users\zeedo\AppData\Local\Arduino15\preferences.txt

Then , by reinstalling the arduino IDE the problem was solved.

KaneIbrahima, 2017-09-18:
No need to reinstall the arduino IDE. I had the same issue. I opened the preferences.txt from atom and then change the path ( settings.path) to the arduino docs path ( in users\~\Appdata\local\Arduino15). You need to close the IDE while you modify it. Save and install the board one more time. It should be fixed

jimsy3, 2017-10-02:
well, i have the same issue with this:
"#include <bits/c++allocator.h> // Define the base class to std::allocator."
so i deleted the 2 folders as said here:
https://github.com/esp8266/Arduino/issues/1698#issuecomment-190104137

but now i cant find the "esp8266 by Esp8266 Community" anymore, so how do i reinstall it?
and maybe the whole problem was caused by installing the wrong library? (by adafruit, not esp8266 community)
but how do i get it back visible in the library manager so i could install it again?

usamayou, 2017-10-16:
This instructable helped me to get rid of this problem.

[https://www.instructables.com/id/Portable-installation-guide-of-Arduino-IDE-v165-fo/](url)

Experienced the same problem. Uninstalled the arduino program and then downloaded Arduino.ide in zip format and followed the instructions given in the article. This will really help you. 

Thank me later.

brianwilson2, 2017-12-20:
Managed to get over the 'compile' problem using Robert Clemenzi's fix. Thanks for that. But still have a problem adn this time I don't know what the problem is. Here is the error info below.

Any ideas? Is it unable to connect? If so, how do I get round this? Seem to be able to connect to Arduino using the same cable.

C:\Documents and Settings\brian\Local Settings\Application Data\Arduino15\packages\esp8266\tools\esptool\0.4.9/esptool.exe -vv -cd nodemcu -cb 115200 -cp COM5 -ca 0x00000 -cf C:\DOCUME~1\brian\LOCALS~1\Temp\arduino_build_115432/BlinkHuzzah.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM5
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 226416 bytes from C:\DOCUME~1\brian\LOCALS~1\Temp\arduino_build_115432/BlinkHuzzah.ino.bin to flash at 0x00000000
	erasing flash
	size: 037470 address: 000000
	first_sector_index: 0
	total_sector_count: 56
	head_sector_count: 16
	adjusted_sector_count: 40
	erase_size: 028000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 15000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................ [ 36% ]
................................................................................ [ 72% ]
..............................................................                   [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete

WestfW, 2018-05-13:
So... I can definitely reproduce this as a path-length issue in recent Windows-10 (windows 10 Home N v1709, build 16299.125), current Arduino IDE (1.8.5), and current ESP8266 board definition (2.4.1)
I downloaded the Windows .zip file, and unpacked it into
   **c:\users\billw\downloads\arduino-1.8.5-win-portable-experiment\arduino-1.8.5\**
Builds of the minimal sketch fail with the "fatal error: bits/c++config.h: No such file or directory" error (as reported by people above."

If I shorten the directory name(s) to "c:\users\billw\downloads\exp\y\", then the same sketch compiles fine...

Now, none of the paths look like they are that close to the 260-character Windows limit, but those are the INPUTS to gcc, and what we need to look at is what actually ends up used in the file-search.    We can take the gcc command that the IDE produces, and add a "-Wp,-v" switch to operate the preprocessor in verbose mode, which will dump it's actual paths.  It turns out that the longest "path" reported is:
 **c:\users\billw\downloads\arduino-1.8.5-win-portable-experiment\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/include/c++/4.8.2/xtensa-lx106-elf**
That's particularly interesting, because it IS where we expect to find bits/c++config.h, AND it's getting really close to that 260-character limit (252 characters.)
If we append the target .h file ("/bits/c++config.h"), we go over the limit easily!

This appears to be one of the paths generated by the compiler configuration, rather than the -Ipaths added by the Arduino IDE, so I don't know whether they're fixable - it's "annoying" that the path is so redundant.

Theoretically, newer versions of Window10 allow the 260-character limit to be turned off.  I tried this and it didn't help; I suspect that gcc would need to be made aware of the removed restriction?
https://msdn.microsoft.com/en-us/library/windows/desktop/aa365247(v=vs.85).aspx

Jivansha, 2018-12-31:
I had the same issue of various header files missing due to change of settings.path when I configured my Matlab with Arduino. Resolved this by opening the preferences.txt file and editing the setting.path to default i.e. C:\Users\AppData\Local\Arduino15 for my system. The Arduino IDE should be closed and after saving the changes reinstall the board for NodeMCU. 
This solved the problem. Thanks

devyte, 2019-03-01:
#1698 is closed, and all toolchains and libs have been rebuilt and made consistent for all platforms.
Closing.
If the problem still exists, please open a new issue and provide the info required in the issue template.

closed 2019-03-01 by devyte

udayaprakashjayaraman, 2019-08-23:
I too had same issues and now its rectified after following below mentioned steps:

Step 1: Download this file in link  -  https://drive.google.com/open?id=1ymTPH6M70T7Y5JrLw8QkMRD_Et3u7LSh

Step 2: go to this file location in your computer   

C:\Users\(User name of your Login)\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\bin

Step 3: Paste extracted file (xtensa-lx106-elf-g++ ,   xtensa-lx106-elf-gcc ) and replace it with the files downloaded from step 1 link.

( If unable to replace and get error ( File is running in other application) then open task manager and check any program (starts with *xtensa* this name) and end process of that and then try to copy that file. 

Now that error is rectified......!