esp8266/Arduino #1632: Arduino's PORTx and PINx not declared

supersjimmie, 2016-02-15:
I am trying to use an Arduino library, but get:
```error: 'PORTD' was not declared in this scope```
Same for PORTB, PORTC, PINB, PINC, PIND.
The lib is found here: https://github.com/xs4free/Itho-library
Is there a way to get PORTx and PINx declared properly on ESP?

igrr, 2016-02-15:
ESP's GPIO registers do not have the same layout as AVRs, so the stuff in `digitalWriteFast.h` won't work.

In general, library writers are advised to use `portInputRegister`, `portOutputRegister`, `portModeRegister`, and `digitalPinToBitMask` functions/macros for fast access to GPIO registers. These functions/macros are normally defined in Arduino.h and provide direct access to GPIO. I think you can add in ifdef ESP8266 clause to `digitalWriteFast.h` and provide the correct definitions for ESP GPIO registers. 

Check out these for reference: https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.h#L122-L131

supersjimmie, 2016-02-15:
```digitalWriteFast2()``` and ```digitalReadFast2()``` both can fallback to normal digitalWrite/Read in certain cases (which I don't fully understand). For now I will just try to use the normal digitalWrite/Read and see if it will work.  At least the read functions seem to work already.
(quickfix: don't include digitalWriteFast.h, define digitalReadFast2 digitalRead etc)

closed 2016-02-15 by supersjimmie