esp8266/Arduino #1637: Nodemcu program does not start on first boot

IMMANUVELMS, 2016-02-16:
I am new to NodeMcu ,i just uploaded the Sketch to NodeMcu which i bought from Aliexpress.i uploaded the blink sketch below

/*
 ESP8266 Blink by Simon Peter
 Blink the blue LED on the ESP-01 module
 This example code is in the public domain
 
 The blue LED on the ESP-01 module is connected to GPIO1 
 (which is also the TXD pin; so we cannot use Serial.print() at the same time)
 
 Note that this sketch uses BUILTIN_LED to find the pin with the internal LED
*/
#define LED     D8  
void setup() {
  pinMode(LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}

Everything works fine after uploading but when remove and reconnect the Nodemcu from the pc it does not start.

then i need to press the reset button once 

after that it starts blinking

any help

Links2004, 2016-02-16:
is the same happening when you connect it to a USB charger? 
if yes may some tool on you PC opens the serial port and this end in booting to Bootloader and not you sketch.

IMMANUVELMS, 2016-02-17:
Thanks for the Reply,Yes,the same happening when i connect to the usb charger i have tried different sketches the  result is same

mtnbrit, 2016-02-17:
What version nodeMCU is it? I had a similar issue with the nodemcu v1.0 board (the blue one). 9 times out of 10 it does not restart after a wdt reset. Would take a press of the reset button or power-cycle to get it back running again. Admittedly I never had trouble getting it to run the sketch on power-up as you seem to. v0.9 nodemcu board worked fine always. 

Does it output anything at all on the serial console? Do you have access to a logic analyzer which you could use to check on the state of GPIO0 and 2, CH_PD and REST pins?

IMMANUVELMS, 2016-02-17:
Thanks to Mtnbrit and Links2004,Finally i have solved it by flashing the firmware with some tweaks and changing the Flash Mode: QIO to DIO in boards.txt located at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0
Those Some tweaks:
changing baudrate to 9600 and SPI Mode to DIO in NODEMCU FIRMWARE PROGRAMMER
now its working fine with usb adaptor and pc also.

closed 2016-02-17 by Links2004

Hein1966, 2020-03-29:
I had the same isue with a NodeMCU V3.  I found out that it can be solved by adding a delay(2000) as first command in the setup() function. After that I start to init the display and wifi. 

void setup() {
  delay(2000);
  display.init();
   .....

ThomasWoeg, 2021-06-22:
the delay is actually the first thing that worked for me, thanks!

I was also able to reduce the delay to 1100 as 2000 is a little long for me

Cospis, 2023-11-04:
I'm surprised it worked for me too, you deserve some coffee @Hein1966 .