esp8266/Arduino #1641: DNS and SPIFF can't work together?

Rickgg, 2016-02-17:
I'm trying ot make a DNSWebServer with the included library and a SPIFF backend, but it doesn't work. Debug says: `ets Jan  8 2013, rst cause:2, boot mode: (3,7)`

Code is:
`void setup() {
  SPIFFS.begin();
  USE_SERIAL.begin(9600);
 
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP(netName, netPass);

  dnsServer.start(DNS_PORT, "editShow.com", apIP);
 
  // handle index
  server.onNotFound([]() {
    String message = "Hello World!\n\n";
    message += "URI: ";
    message += server.uri();

    server.send(200, "text/plain", message);
  });

  server.begin();`

Any help or ideas for me to help would be welcome.

Rickgg, 2016-02-17:
Found the issue, didn't include a FS header in the makefile.

marvinroger, 2016-02-17:
What is your loop code?

Rickgg, 2016-02-18:
Found the problem: I didn't specify the Flash size (1 mB). Now all I need is to upload the files...

closed 2016-02-19 by Rickgg