esp8266/Arduino #1642: web server optimization

mkeyno, 2016-02-17:
I used the last staging version of the library and set up module for access point and run the SPIFF simple authentication sample code , the problem is when just open  one tab of browser , there is no problem but if you concurrently have couple of open tabs (for example my gmail, yahoo mail, ...) which they are constantly send the request to the server , the ESP will chocked soon and need the hardware reset, and firmware auto-reset is not working
what I'm thinking about is to find the way that close the connection  As Soon As find requested page or find is not located in the SPIFF or anywhere inside the chip
maybe following code would be helpful to my idea

`
/////////////// in set up
server.onNotFound(handleOther);
///////////////////////////
handleOther()
{
check if the requested file not exits sever the concretion before choking the ESP
}
`

mkeyno, 2016-02-19:
@igrr  @Links2004  guys any suggestion?

Links2004, 2016-02-19:
your requests overload the ESP socktes.
do you run CaptivePortal example or the DNS server with settings like this:
```cpp
 dnsServer.start(DNS_PORT, "*", apIP);
```
normally the browser do a DNS first before it connects to a IP.

some more infos will sure help:
 - your code or link to the examle you use (when not modified)
 - serial debug log of the ESP (https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md)
 - if you have a stack dump (https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md)

mkeyno, 2016-02-20:
@igrr @Links2004 
yes , I used `dnsServer.start(DNS_PORT, "*", apIP);  ` in AP mode  and MDSN in station mode, but my questions is how to optimize the respond speed and reliability when normally people open couple of tabs in browsers ? is there way to cut down the request when server realize , it is not from the true client, or make Q to not let the serve  be choken

Links2004, 2016-02-20:
to make this filtering at least the HTTP header from the client need to be revised.
you can try the async webserver form @me-no-dev it can handle multible requests way better.
https://github.com/me-no-dev/ESPAsyncWebServer

closed 2016-04-10 by mkeyno