esp8266/Arduino #1643: make shebang more specific python vs python2.7

Duality4Y, 2016-02-17:
Hi there!
I am running Arch and in Arch linux python == python3.x while on debian probably it is python2.7
It would be really nice if all the lines in the python tools start with:
```
#!/usr/bin/env python2.7
```
instead of:
```
#!/usr/bin/env python
```

that way it is more portable if your not going to write python3.x compatible code anyway.
and it save's me the trouble of editing those files every single time.

so:
more portable
reaches more linux distro's.
== Better right?

igrr, 2016-02-17:
Possibly relevant comments from another popular ESP tool:
https://github.com/themadinventor/esptool/pull/20#issuecomment-73304890

Duality4Y, 2016-02-17:
I see didn't know it went that far :)
the suggested change works under arch and ubuntu/debian.
I don't know how all the other distro's have thier python linked.

devyte, 2017-10-18:
Per history, there have been several commits for python3 compat. Closing, as this issue is old over a year old.

closed 2017-10-18 by devyte