esp8266/Arduino #1645: SPIFFS read exception!

EagleSmith, 2016-02-18:
**ESP 12F:**

I read the file at boot time, and sometimes cause an exception.
someone help me to see what is the reason? thank you

Exception (0):

epc1=0x40211168 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 

sp: 3ffffc10 end: 3fffffb0 offset: 01a0

>>>stack>>>

3ffffdb0:  40224b78 3fff7610 00000010 60000600  

3ffffdc0:  00000001 402204e8 3fffc250 4000050c  

3ffffdd0:  400043e6 00000030 00000016 ffffffff  

3ffffde0:  400044ab 3fffc718 3ffffed0 08000000  

3ffffdf0:  60000200 08000000 08000000 00000000  

3ffffe00:  0000ffff 00000001 04000002 003fd000  

3ffffe10:  3fff7848 000003fd 3fff2c64 00000030  

3ffffe20:  40101709 00000008 00000008 00000020  

3ffffe30:  6b19c57b 0c45aaa2 26d7399a 63bae515  

3ffffe40:  3fff7718 00000001 40239ac1 3fff76b0  

3ffffe50:  3fff7610 00000010 60000600 00000020  

3ffffe60:  40230f20 3fff7758 3fff76d4 402399f7  

3ffffe70:  402308e4 40231c36 3fff2155 3fff7600  

3ffffe80:  3ffffec8 00000010 40231dd9 3fff1690  

3ffffe90:  3fff6f48 3ffed7c0 60000600 3fff71e0  

3ffffea0:  3fff1690 3fff1690 3fff6f48 3fff7520  

3ffffeb0:  3fff2fe8 40101233 d634fe1a fffeffff  

3ffffec0:  00000001 00000000 4022e34e 3fff6f48  

3ffffed0:  00000002 4000410f 3fff2a94 3fff6f48  

3ffffee0:  3fffc718 40004a3c 000003fd 3fff7848  

3ffffef0:  3fffc718 40101510 00000378 3fff2155  

3fffff00:  000003fd 4021e05b 00000378 000003ff  

3fffff10:  00001000 4021e0f1 3fff2c64 000003ff  

3fffff20:  000003fd 60000600 003fd000 3fff2c64  

3fffff30:  ffffff00 55aa55aa 00000a90 0000001c  

3fffff40:  0000001c 000000fd 000000e4 000003ff  

3fffff50:  4021df98 3ffed690 00000000 3ffed7c0  

3fffff60:  00000001 4021d05d 00000003 60000600  

3fffff70:  4021cde5 3ffed684 3ffed630 0026b530  

3fffff80:  4021ce2a 3fffdab0 00000000 3fffdcb0  

3fffff90:  3ffed650 3ffeded8 00000005 4020f393  

3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  

<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 

tail 0

chksum 0x42

csum 0x42

~ld

Links2004, 2016-02-18:
use the Esp Exception Decoder to see where the problem is comming from.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md
what core version do you use?

EagleSmith, 2016-02-19:
thankyou ,my core version is 2.0.

EagleSmith, 2016-02-22:
Exception (0):

epc1=0x40211168 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 

sp: 3ffffc10 end: 3fffffb0 offset: 01a0

>>>stack>>>

3ffffdb0:  40224b78 3fff7610 00000010 60000600  

3ffffdc0:  00000001 402204e8 3fffc250 4000050c  

3ffffdd0:  400043e6 00000030 00000016 ffffffff  

3ffffde0:  400044ab 3fffc718 3ffffed0 08000000  

3ffffdf0:  60000200 08000000 08000000 00000000  

3ffffe00:  0000ffff 00000001 04000002 003fd000  

3ffffe10:  3fff7848 000003fd 3fff2c64 00000030  

3ffffe20:  40101709 00000008 00000008 00000020  

3ffffe30:  6b19c57b 0c45aaa2 26d7399a 63bae515  

3ffffe40:  3fff7718 00000001 40239ac1 3fff76b0  

3ffffe50:  3fff7610 00000010 60000600 00000020  

3ffffe60:  40230f20 3fff7758 3fff76d4 402399f7  

3ffffe70:  402308e4 40231c36 3fff2155 3fff7600  

3ffffe80:  3ffffec8 00000010 40231dd9 3fff1690  

3ffffe90:  3fff6f48 3ffed7c0 60000600 3fff71e0  

3ffffea0:  3fff1690 3fff1690 3fff6f48 3fff7520  

3ffffeb0:  3fff2fe8 40101233 d634fe1a fffeffff  

3ffffec0:  00000001 00000000 4022e34e 3fff6f48  

3ffffed0:  00000002 4000410f 3fff2a94 3fff6f48  

3ffffee0:  3fffc718 40004a3c 000003fd 3fff7848  

3ffffef0:  3fffc718 40101510 00000378 3fff2155  

3fffff00:  000003fd 4021e05b 00000378 000003ff  

3fffff10:  00001000 4021e0f1 3fff2c64 000003ff  

3fffff20:  000003fd 60000600 003fd000 3fff2c64  

3fffff30:  ffffff00 55aa55aa 00000a90 0000001c  

3fffff40:  0000001c 000000fd 000000e4 000003ff  

3fffff50:  4021df98 3ffed690 00000000 3ffed7c0  

3fffff60:  00000001 4021d05d 00000003 60000600  

3fffff70:  4021cde5 3ffed684 3ffed630 0026b530  

3fffff80:  4021ce2a 3fffdab0 00000000 3fffdcb0  

3fffff90:  3ffed650 3ffeded8 00000005 4020f393  

3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  

<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 

tail 0

chksum 0x42

csum 0x42

~ld

sticilface, 2016-02-22:
You've just pasted the stack dump again.  You need to copy paste that into the exception decoder in the arduino IDE for anyone to make any sense of it. 

https://github.com/me-no-dev/EspExceptionDecoder

DoctorWho8, 2016-02-22:
Hello!
But trying to visit that page via Chrome results in error messages.
Are you sure its even at home today?
-----
Gregg C Levine gregg.drwho8@gmail.com
"This signature fought the Time Wars, time and again."

igrr, 2016-02-27:
@EagleSmith Are you using I2S in your sketch?

EagleSmith, 2016-02-28:
yes,i used hx711

igrr, 2016-02-28:
Ok, it's clear then. Need to add I2S to the interrupts being masked.

igrr, 2016-02-29:
Closing this one in favour of an earlier issue: #1388

closed 2016-02-29 by igrr