esp8266/Arduino #1648: ESP12 Exception (29)

alinleonard, 2016-02-18:
Hi I got this error when I try to power esp.

```
Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff09e0 end: 3fff0e10 offset: 01a0

>>>stack>>>
3fff0b80:  3fff2050 00000182 40233fb0 4020bc0a  
3fff0b90:  40212ced 3fff2010 00000000 ffffff80  
3fff0ba0:  65736724 4020e1db 3fff2010 3fff1a90  
3fff0bb0:  40212f2e 3fff2010 3fff1a90 00000000  
3fff0bc0:  3ffefddc 3ffed6e0 3ffed65c 3fff1a90  
3fff0bd0:  00000000 00000000 00000001 00000000  
3fff0be0:  00000018 00000064 da34fe1a fffeffff  
3fff0bf0:  3ff20a00 0000ffff 3fff0c18 4020e882  
3fff0c00:  3ffed65c 3fff1a90 3fff1a90 00000000  
3fff0c10:  da34fe1a 3fff127f 00000000 00000000  
3fff0c20:  00000000 00000000 00000000 00000000  
3fff0c30:  00000000 00000000 00000000 00000000  
3fff0c40:  00000000 00000000 00000000 4020f9d4  
3fff0c50:  3fff1a90 00000001 00000001 00000000  
3fff0c60:  00000000 3ffed6e0 3ffeeaf8 00000000  
3fff0c70:  402277ad 00000003 00000003 0000007d  
3fff0c80:  402278f9 00000003 00000001 00000000  
3fff0c90:  3fff1240 4022796e 00000003 00000001  
3fff0ca0:  402025dd 3ffefba0 00000001 00000000  
3fff0cb0:  40202614 feefeffe 3ffefba0 40202636  
3fff0cc0:  feefeffe 3ffe84d8 3ffe84c4 4020279e  
3fff0cd0:  feefeffe feefeffe feefeffe feefeffe  
3fff0ce0:  feefeffe feefeffe feefeffe feefeffe  
3fff0cf0:  feefeffe feefeffe feefeffe feefeffe  
3fff0d00:  feefeffe feefeffe feefeffe feefeffe  
3fff0d10:  feefeffe feefeffe feefeffe feefeffe  
3fff0d20:  feefeffe feefeffe feefeffe feefeffe  
3fff0d30:  feefeffe feefeffe feefeffe feefeffe  
3fff0d40:  401052ce 00000000 3ffe8404 40205ab8  
3fff0d50:  00000020 40105427 3ffefdf0 3ffefc54  
3fff0d60:  40205a34 0001c200 4020674c 3ffe8477  
3fff0d70:  3ffefd1c 0000001a 3ffe8476 40206c90  
3fff0d80:  0000001c 3ffefd1c 3fff1118 3ffefddc  
3fff0d90:  3fffdad0 00000000 3ffefd1c 402060bd  
3fff0da0:  3ffefba0 00000000 00000001 feefeffe  
3fff0db0:  feefeffe feefeffe 3ffefd1c 3ffefddc  
3fff0dc0:  3fffdad0 3ffefba0 3ffefd1c 4020238c  
3fff0dd0:  feefeffe feefeffe feefeffe feefeffe  
3fff0de0:  feefeffe feefeffe feefeffe 3ffefddc  
3fff0df0:  3fffdad0 00000000 3ffefdd5 40206790  
3fff0e00:  feefeffe feefeffe 3ffefdf0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
system param error
0f 31 c8 ff c0 20 00 29 03 2d 0c 01 cb ff c0 00 00 08 31 c8 21 12 c1 10  €

```

Source code 

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>

const char* ssid = "xxxxxx";
//at least 8 characters long
const char* password = "xxxxxxxxx";

ESP8266WebServer server(80);
ESP8266HTTPUpdateServer httpUpdater;

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp12!");
  digitalWrite(led, 0);
}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  Serial.println();
  Serial.print("Configuring access point...");
  WiFi.begin(ssid, password);
    
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.on("/reset", [](){
    server.send(200, "test/plain", "ESP reset");
    delay(500);
    ESP.restart();
  });

  server.onNotFound(handleNotFound);

  httpUpdater.setup(&server);
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
}
```

**I use latest staging (2.1.0-rc2)**

Pins:
VCC, EN - **3.3V**
GPIO15,GND - **GND**

Links2004, 2016-02-18:
use the Esp Exception Decoder to see where the problem is coming from.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

at the first look its a null ptr usage or flash access problem.
the stack dump will show more.

alinleonard, 2016-02-18:
I decoded:
```
Decoding 24 results
0x40233fb0: lwip_init at ??:?
0x4020bc0a: esf_buf_alloc at ??:?
0x40212ced: ieee80211_freedom_output at ??:?
0x4020e1db: ieee80211_getmgtframe at ??:?
0x40212f2e: ieee80211_beacon_alloc at ??:?
0x4020e882: ieee80211_hostap_attach at ??:?
0x4020f9d4: wifi_softap_start at ??:?
0x402277ad: wifi_get_opmode_default at ??:?
0x402278f9: wifi_set_broadcast_if at ??:?
0x4022796e: wifi_set_opmode at ??:?
0x402025dd: ESP8266WiFiGenericClass::mode(WiFiMode) at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.cpp:80
0x40202614: ESP8266WiFiGenericClass::getMode() at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.cpp:80
 (inlined by) ESP8266WiFiGenericClass::enableSTA(bool) at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.cpp:228
0x40202636: ESP8266WiFiGenericClass::enableSTA(bool) at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.cpp:80
0x4020279e: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp:635
0x401052ce: ets_timer_setfn at ??:?
0x40205ab8: uart_start_init(int, int, unsigned char, unsigned char, unsigned char) at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/HardwareSerial.cpp:745
0x40105427: ets_timer_arm_new at ??:?
0x40205a34: uart_interrupt_enable(uart_*) at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/HardwareSerial.cpp:745
0x4020674c: esp_yield at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_main.cpp:43
0x40206c90: Print::write(unsigned char const*, unsigned int) at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/Print.cpp:42
0x402060bd: Print::write(char const*) at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/Print.h:60
0x4020238c: setup at C:\Users\John\Documents\GIT\esp8266-remote\firmware/firmware.ino:48
0x40206790: loop_wrapper at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_main.cpp:43
0x40100114: cont_norm at C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/cont.S:109

```

Links2004, 2016-02-18:
there is a problem during the WiFi init.
I see too possible reasons atm:
- instabile power (what power supply do you use?)
- WiFi config in flash corrupted
flash a sketch with: 
```cpp
void setup() {
    WiFi.softAPdisconnect(true);
    WiFi.disconnect(true);
}
void loop() {}
```
and then try yours again.
if this still not helps:
```cpp
void setup() {
    ESP.eraseConfig();
    ESP.reset();
}
void loop() {}
```

alinleonard, 2016-02-18:
I use a 12V 1000mA source then I use a step-down to 3.4V (LM2596S)

Gonna try those sketches.

igrr, 2016-02-27:
Also please check that your flash size setting in the IDE matches the size of the flash chip. There is a CheckFlashConfig example, please run it on your board.

DedeHai, 2016-05-02:
I have the same problem. I got a sketch to run that only uses  WiFi.mode(WIFI_AP) but not in station mode. Also I had a test sketch crash that did not even use any wifi configuration.
I tried ESP.eraseConfig() as well ass disconnecting AP and wifi as suggested by @Links2008. The flash maybe corrupted/broken perhaps?
Here is a the decoded stackdump:

`0x401004f4: calloc at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1682
0x401071d0: pvPortZalloc at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/heap.c line 33
0x40226f4d: wifi_param_save_protect_with_check at ?? line ?
0x40105b32: spi_flash_read at ?? line ?
0x40227035: system_param_save_with_protect at ?? line ?
0x40227424: wifi_station_ap_number_set at ?? line ?
0x4022757f: wifi_station_ap_number_set at ?? line ?
0x4022760c: wifi_station_ap_number_set at ?? line ?
0x40227648: wifi_station_set_config at ?? line ?
0x4020dbb5: Print::write(char const*) at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 188
0x40206cb0: ESP8266WiFiSTAClass::disconnect(bool) at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 432
0x4020e69c: String::changeBuffer(unsigned int) at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 681
0x4020ddc1: Print::printNumber(unsigned long, unsigned char) at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 188
0x4020e71d: String::copy(char const*, unsigned int) at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 681
0x4020f7a4: Print::write(unsigned char const*, unsigned int) at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 38
0x4020dbb5: Print::write(char const*) at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 188
0x4020dcb0: Print::println() at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 188
0x4020d5fc: EspClass::getFreeHeap() at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Esp.cpp line 467
0x402036e6: ConfigureWifi() at C:\Users\MAXPOW~1\AppData\Local\Temp\buildba9cf6797dc05bc50d0abcb46b453bb8.tmp\sketch/ackpayloads.h line 49
0x40203f31: std::_Function_base::_Base_manager ::_M_init_functor(std::_Any_data&, void (*&&)(), std::integral_constant ) at c:\users\max power\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1983
:  (inlined by) std::_Function_base::_Base_manager ::_M_init_functor(std::_Any_data&, void (*&&)()) at c:\users\max power\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1958
:  (inlined by) function  at c:\users\max power\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2451
:  (inlined by) setup at D:\GitHub\lambda_nodes\nodes\AlphaNode_Firmware/NRF24L01.ino line 114
0x4020eff0: loop_wrapper at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at C:\Users\Max Power\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/cont.S line 109
`

The sketch works fine on other modules, just not this particular one.

closed 2016-06-23 by igrr

rostlaur, 2017-05-23:
Never Surrender ;^)
Past 3 days investigating an error:
On COM Monitor with WiFiScan, after Build/Upload I was always getting an “Exception (29)” error.
On this page the second suggested solution (‘if this still not helps’) worked wonders:
---
Links2004 commented on 18 Feb 2016 
there is a problem during the WiFi init.
I see too possible reasons atm:
•	instabile power (what power supply do you use?)
•	WiFi config in flash corrupted
flash a sketch with:
void setup() {
    WiFi.softAPdisconnect(true);
    WiFi.disconnect(true);
}
void loop() {}
and then try yours again.
if this still not helps:
void setup() {
    ESP.eraseConfig();
    ESP.reset();
}
void loop() {}
---
Thank you so much Links2004 for your post

wfraga18, 2017-05-28:
i'm getting THE SAME

ratnick, 2017-06-01:
I'm getting the same error. 
It turned out that I had a variable allocating too much space. Reducing this variable by 10k resolved the problem.

sskcooldude, 2017-09-07:
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 27 results
0x40203df4: ax_port_calloc at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 592
0x40221c42: ssl_new at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 549
0x4022337b: ssl_client_new at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 53
0x401004d8: malloc at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x402036fc: WiFiClient::_s_connected(void*, void*, signed char) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 327
0x40203c7e: SSLContext::connect(ClientContext*, char const*, unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
:  (inlined by) WiFiClientSecure::_connectSSL(char const*) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 279
0x402031a2: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 327
0x40203ecd: WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 269
0x402020a0: Gsender::Send(String const&, String const&) at C:\Users\User\AppData\Local\Temp\arduino_build_749702\sketch/Gsender.cpp line 44
0x40206aac: String::changeBuffer(unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 589
0x40206aac: String::changeBuffer(unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 589
0x40206afb: String::reserve(unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 589
0x40206b2d: String::copy(char const*, unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 589
0x40206b7e: String::String(char const*) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 589
0x40202548: send_mail(char const*, char const*, String) at E:\Projects\LY Project KMPS\LYProjKMS/LYProjKMS.ino line 176
0x40206bf0: String::operator=(String const&) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 589
0x40206c18: String::String(String const&) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 589
0x4020279a: give_response(int, String, String) at E:\Projects\LY Project KMPS\LYProjKMS/LYProjKMS.ino line 269
0x40206c18: String::String(String const&) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 589
0x4020288a: handleNewMessages(int) at E:\Projects\LY Project KMPS\LYProjKMS/LYProjKMS.ino line 210
0x40206b2d: String::copy(char const*, unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 589
0x402029b4: bot_handler() at E:\Projects\LY Project KMPS\LYProjKMS/LYProjKMS.ino line 290
0x40202acc: loop at E:\Projects\LY Project KMPS\LYProjKMS/LYProjKMS.ino line 316
0x402070b8: loop_wrapper at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

MCGrumm, 2018-10-24:
try to inlcude #include <ArduinoJson.h> in library and install ArduinoJson in library

sadiqueimam, 2018-12-27:
I was getting the same issue then i tried the 2nd suggestion 
void setup() {
ESP.eraseConfig();
ESP.reset();
}
void loop() {}
and now ESP is not working properly. Getting unreadable data in Serial Monitor. Can anyone help me with this issue?

codingneko, 2019-05-25:
@sadiqueimam you've probably already done this, but check you've got your serial monitor listening to the right baud rate

dan27az, 2019-09-07:
Simply erase template field and save worked for me.

yourdeviceip./co?

[image]