esp8266/Arduino #1651: 0x40211508: HardwareSerial::isRxEnabled() at C:\Program Files ???

EagleSmith, 2016-02-19:
could someone help me for this exception?thank you .

Exception (0):
epc1=0x40211508 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc10 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdb0:  40224f18 3fff8060 00000010 60000600  
3ffffdc0:  00000001 40220888 3fffc250 4000050c  
3ffffdd0:  400043e6 00000030 00000016 ffffffff  
3ffffde0:  400044ab 3fffc718 3ffffed0 08000000  
3ffffdf0:  60000200 08000000 08000000 00000000  
3ffffe00:  0000ffff 00000001 04000002 003fd000  
3ffffe10:  3fff8298 000003fd 3fff2dc4 00000030  
3ffffe20:  40101709 00000008 00000008 00000020  
3ffffe30:  3c948f3a 5079df22 0fea5f32 5436c086  
3ffffe40:  3fff8168 00000001 40239e69 3fff8100  
3ffffe50:  3fff8060 00000010 60000600 00000020  
3ffffe60:  402312c0 3fff81a8 3fff8124 40239d9f  
3ffffe70:  40230c84 40231fd6 3fff22b5 3fff8000  
3ffffe80:  3ffffec8 00000010 40232179 3fff17f0  
3ffffe90:  3fff7998 3ffed920 60000600 3fff7c30  
3ffffea0:  3fff17f0 3fff17f0 3fff7998 3fff7f70  
3ffffeb0:  3fff3148 40101233 d634fe1a fffeffff  
3ffffec0:  00000001 00000000 4022e6ee 3fff7998  
3ffffed0:  00000002 4000410f 3fff2bf4 3fff7998  
3ffffee0:  3fffc718 40004a3c 000003fd 3fff8298  
3ffffef0:  3fffc718 40101510 00000378 3fff22b5  
3fffff00:  000003fd 4021e3fb 00000378 000003ff  
3fffff10:  00001000 4021e491 3fff2dc4 000003ff  
3fffff20:  000003fd 60000600 003fd000 3fff2dc4  
3fffff30:  ffffff00 55aa55aa 00000ac0 0000001c  
3fffff40:  0000001c 000000e1 000000a2 000003ff  
3fffff50:  4021e338 3ffed7f0 00000000 3ffed920  
3fffff60:  00000001 4021d3fd 00000003 60000600  
3fffff70:  4021d185 3ffed7e4 3ffed790 0025854a  
3fffff80:  4021d1ca 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffed7b0 3ffee000 00000005 4020f733  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

DECODER:

Decoding 29 results
0x40211508: HardwareSerial::isRxEnabled() at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/HardwareSerial.cpp:550
0x40224f18: wDev_Insert_KeyEntry at ??:?
0x40220888: ic_set_key at ??:?
0x4000050c: ?? ??:0
0x400043e6: ?? ??:0
0x400044ab: ?? ??:0
0x40101709: os_printf_plus at ??:?
0x40239e69: wpa_auth_sm_event at ??:?
0x402312c0: ieee80211_add_xrates at ??:?
0x40239d9f: wpa_auth_sm_event at ??:?
0x40230c84: ieee80211_send_setup at ??:?
0x40231fd6: ieee80211_freedom_output at ??:?
0x40232179: ieee80211_beacon_alloc at ??:?
0x40101233: ets_timer_arm_new at ??:?
0x4022e6ee: ieee80211_hostap_attach at ??:?
0x4000410f: ?? ??:0
0x40004a3c: ?? ??:0
0x40101510: spi_flash_erase_sector at ??:?
0x4021e3fb: wifi_param_save_protect_with_check at ??:?
0x4021e491: system_param_save_with_protect at ??:?
0x4021e338: wifi_set_broadcast_if at ??:?
0x4021d3fd: system_get_os_print at ??:?
0x4021d185: ets_timer_handler_isr at ??:?
0x4021d1ca: ets_timer_handler_isr at ??:?
0x4020f733: loop_task at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp:43
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x4010f000: ?? ??:0

igrr, 2016-02-27:
Could you please try with 2.1.0 stable? HardwareSerial has been reworked recently.

EagleSmith, 2016-02-28:
2.0.1 is work fine,thanks

closed 2016-02-28 by EagleSmith