esp8266/Arduino #1653: Espressif SDK 1.5.2 released

osos, 2016-02-19:
http://bbs.espressif.com/viewtopic.php?f=46&t=850

Might be worth to include in the upcoming 2.1.0 release of esp8266-arduino?

andig, 2016-02-20:
Would be great if we were able to recompile lwip, too? Imho higher priority than sdk version bump?

andig, 2016-04-03:
from gitter:

> to use (gcc lwip) instead of Espressif-provided one, you need to replace `-llwip` with `-llwip_gcc` in `platform.txt`, and restart the IDE.

kaeferfreund, 2016-04-03:
@andig 
thx for the advice. Is there any obvious advantage with that?

andig, 2016-04-03:
> thx for the advice. Is there any obvious advantage with that?

Not unless you're debugging and get stuck when the code touches the lwip internals.

andig, 2016-04-06:
Quick note here- async web server is running rock stable with sdk 1.5.2 according to my tests. The ugly crashes have been fixed thanks to @igrr and the insights from lwip.

kaeferfreund, 2016-04-06:
are you referring to your fork or me-no-devs version?

andig, 2016-04-06:
me-no-dev plus the open pr on the webserver for handling lowmem.

closed 2016-04-18 by igrr