esp8266/Arduino #1654: Cannot use SDK PWM

patrickjahns, 2016-02-19:
In relation to #836 I`ve been trying to use the PWM functions provided by the sdk

I included the header file with
`extern "C" {
#include "pwm.h"
}`

Trying to initialize it in setup() with
`
pwm_start();
`

Linker complains that it can`t find pwm_init() function ??

> build1c4a41795fe117f0cc67c7576913ee7e.tmp\sketch\CallSDKFunctions.ino.cpp.o: In function 'setup':
>CallSDKFunctions.ino:24: undefined reference to `pwm_start'

patrickjahns, 2016-02-19:
Managed to include it by changing the linker flags and adding lpwm.

But it won`t run smoothly - experiencing constant watchdog resets - did anyone else try this already?

kaeferfreund, 2016-03-06:
@patrickjahns how did you proceed to include the sdk pwm into the sketch?

patrickjahns, 2016-03-06:
@kaeferfreund 

I changed the build files to include lpwm library - but a stated - it won`t run. As soon as I initialized it - the esp would reset. I`ve been using the code from the espressif iot_sdk example. 

After trying everything that came to my mind I switched to a different framework (SMING) for the time being. The same code works there without problems

ge0rg, 2016-05-18:
The crash problem should be fixed as of https://github.com/esp8266/Arduino/commit/234a8743c9533d1b12d7962802869fe68e83179f - I tried to run the PWM code I had trouble with in #836, however so far all my channels remain dark. Will need to play around a little more.

patrickjahns, 2016-05-18:
Thanks for the information - I migrated my code to utilize a different framework (SMING) since it required hardware pwm. I`d be curious to know if you manage to get it working.

I utilized the esp8266 pwm output in this library of mine:
https://github.com/patrickjahns/RGBWWLed/blob/master/RGBWWLedOutput.cpp

You probably need to change the `EspDigitalPins` part to match the arduinowiring
Also i found the iot example from espressif also a good start for working with pwm

If you can confirm it working - i probably would polish the library to be SMING and Arduino SDK compatible

ge0rg, 2016-05-18:
Yeah, I had ripped a large part of the IoT light demo and frankensteined it into my arduino project out of sheer despair, just to end up with a wdt reset.

Now, my code is running without resets, and I can see my logs over serial, as well as reactions to MQTT.

However, the RGB lines are all off, no matter which vaules I write (and I'm pretty sure the pins are properly configured, it's copy&paste from `user_light.c`).

igrr, 2016-05-18:
What do you guys mean by "hardware pwm" in the context of ESP8266? There is only one hardware PWM source, also known as Sigma-Delta, which is shared between all pins. I.e. this is a single channel PWM which can be sent to multiple pins.
The thing that libpwm.a from SDK does is pretty much the same as we do: it generates PWM in software using FRC1 timer as an interrupt source. The only difference currently is that SDK PWM implementation configures the interrupt to fire at NMI level, while we still are using it as a level-1 interrupt.
Now that the reason why we weren't able to use NMI interrupt in Arduino framework has been identified and fixed, we can also switch PWM implementation to use NMI. 
There even is an open ticket for that if I recall correctly :)

igrr, 2016-05-18:
Oh, I'll keep this open because of the issue you have — no actual output on pins. Are you sure your code actually switches pins to output mode? We configure all pins as input as part of early initialization, that's probably the only difference with the SDK environment.

patrickjahns, 2016-05-18:
You are correct in the sense of there is only one PWM/Timer source. 
When I was writing about "hardware pwm" I was referring to the espressif SDK pwm implementation which has proven to be reliable (compared to the current Arduino ESP SDK implementation). 

I currently don`t know what the standpoint in terms of compatibility of utilizing espressif SDK functions within Arduino esp SDK is. But it would be great to document the working/not working parts

ge0rg, 2016-05-18:
@igrr thanks for the pointer. I had indeed the wrong pins set to OUTPUT. After fixing my #defines and adding `pwm_start()` after each `pwm_set_duty()`, I now have working SDK-PWM in an Arduino project.

With the 22222 PWM range at 1000Hz it also looks much smoother than the Arduino PWM code base. I'll let it run for some hours to see if there is any flickering.

igrr, 2016-05-18:
Okay, great to hear that!

kaeferfreund, 2016-05-22:
I also tried to move PWM to NMI timer, which indeed works. But if I fade the PWM in a shot period of time, I get a `Fatal exception:0 flag:4 (SOFT_RESTART) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000`

What I did for now is replace `ETS_FRC_TIMER1_INTR_ATTACH(timer1_isr_handler, NULL)` with `ETS_FRC_TIMER1_NMI_INTR_ATTACH(timer1_isr_handler)`

Is there any other parts that need to be moved?

igrr, 2016-05-22:
Let me see if i can reproduce this exception. When you say that you are fading PWM in a short period of time, do you mean something like this?
```c++
for (int val = PWMRANGE; val >= 0; val -= 10) {
  analogWrite(pin, val);
  delay(1);
}
```

kaeferfreund, 2016-05-22:
yap, I have:
```
for (uint16_t i = 1; i < brt; i++) {
        analogWrite(PIN1, i);
        analogWrite(PIN2, i);
        yield();
      }
```

But I also have some web sockets-stuff going on, which is a bit slower but still to fast...

Just switched to SDK-pwm, seems like this also is fading a bit slower than arduino-pwm

igrr, 2016-05-22:
@kaeferfreund also please let's move discussion related to Arduino implementation of PWM back to #836. This issue was about usage of SDK PWM library.

kaeferfreund, 2016-05-22:
Okay. I just moved to SDK-pwm. Which in deed seems smoother, but fading doesn't look too nice. It's a bit more stuttery for me, than the original arduino-pwm. Anyone else seeing that issue?

closed 2016-06-23 by igrr

JoaoLopesF, 2017-04-10:
Hi,

I made one sample to use the new pwm.c: https://github.com/StefanBruens/ESP8266_new_pwm in Arduino. This resolves the  any issues of original pwm.c of SDK 
See it in: https://github.com/JoaoLopesF/ESP8266-Arduino-PWM-SDK-Sample
Only this works fine to me.
Regards,
Joao