esp8266/Arduino #1655: WiFiClientSecure, SSL not used at all?

comino, 2016-02-19:
Hello, 

I realized, that my https only server is refusing the https connection from the ESP and I investigated the issue, but couldnt make it to work. 

Either I oversee something here or there is a bug in the implementation of WifiClientSecure::connect. The overloaded funtion with handled char url array doesnt call _connectSSL, but just a normal TCP connection (WiFiClient::connect()). 

The overloaded method with Ip adress does. 

Thus the "HTTPSRequest" example is using the char array. So to my mind its performing a simple http request (which is not blocke by the example url). 

[image]

Is this a bug and SSL is not used at all or am I missing something here?

igrr, 2016-02-19:
`WiFiClientSecure::connect(name, port)` will call `WiFiClient::connect(name, port)`, which will perform name resolution and will call `Client::connect(ip, port)`. `Client::connect(ip, port)` is a virtual function, so `WiFiClientSecure::connect(ip, port)` will be called, which in turn will call `_connectSSL()`.
So no, this isn't a bug.

comino, 2016-02-19:
ok thanks.. will investigate my issues with HTTPS and reopen if needed.

closed 2016-02-19 by comino