esp8266/Arduino #1656: WiFiClient.write(const *buf ... why the buf is const? Read only buffer?

gosewski, 2016-02-20:
What is the reason of having the buffer that is constant and can not be filled in with new data (read only buffer) ?
I wanted to use this function to speed up uploading imeges from SPIFFS to client but in this form I wasn't able to use it. Therefore I created overloaded function:

WiFiClient.h:
`virtual size_t write(uint8_t *buf, size_t size);`

WiFiClient.cpp:
```
size_t WiFiClient::write(uint8_t *buf, size_t size)
{
    if (!_client || !size)
    {
        return 0;
    }

    return _client->write(reinterpret_cast<char*>(buf), size);
}
```
Now I am able to us dynamically filed buffer what speeds up the transmission greatly.
Is there any risk of using such function?

Links2004, 2016-02-20:
no risk there, but simple cast by the write call,
there is no need to add a new function.
```cpp
uint8_t buffer[100];
tcp.write((const uint8_t *)buffer, 100);
```

gosewski, 2016-02-23:
thank you for the information, I have figured out that even cating to uint8_t* (without "const") works but what is the idea of creating a function that seems to be useless without the casting? Isn't it better to make one overloaded function than use castings each time the function is used?

Links2004, 2016-02-23:
the function is mandatory for the print class.
It just the way the Arduino API work.
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Print.h#L61

closed 2016-02-23 by Links2004

gosewski, 2016-02-23:
ok, everything is clear now. Thank you for clarification.