esp8266/Arduino #1657: Add SPIFFS.unmount() for use in OTA.onEnd()

andig, 2016-02-20:
Use case: allow to store configuration from ram back to freshly updated SPIFFS. Avoid loss on configuration due to OTA wiping the filesystem config files.

andig, 2016-06-02:
Great stuff, thank you! Please feel free to close.

closed 2016-06-23 by igrr

andig, 2016-07-04:
@igrr once OTA finishes I'm assuming that SPIFFS already is unmounted and I'd need to wrap further writes into SPIFFS begin/end. Is this correct?

igrr, 2016-07-04:
I think you need to unmount SPIFFS before you start OTA, i.e. in ArduinoOTA onBegin handler or similar.
Once OTA is finished, you may mount it again in onEnd handler.

andig, 2016-07-04:
> I think you need to unmount SPIFFS before you start OTA

If find that a bit confusing. Shouldn't- if SPIFFS is touched by OTA- the unmount be part of the default OTA code? Remounting is clear for the onEnd handler, but unmounting?

igrr, 2016-07-04:
I see this the other way around. If you happen to use SPIFFS in the sketch, and you call `SPIFFS.begin` somewhere, you have to make sure you return SPIFFS to uninitialized state before modifying its flash storage.
I would find it confusing if `.begin` calls were at one level of abstraction (sketch) and `.end` calls were at lower layer (hidden inside the library).

andig, 2016-07-04:
> I see this the other way around. If you happen to use SPIFFS in the sketch, and you call SPIFFS.begin somewhere, you have to make sure you return SPIFFS to uninitialized state before modifying its flash storage.

Right. That makes perfect sense! The only thing is- up to now this wasn't even possible ;) So just wondering if this is worth a note or even a comment line in one the OTA examples (may this would be a good place: https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino) to make sure people understand this?

Could do a PR for you!