esp8266/Arduino #1658: Gpio transitions on startup

penfold42, 2016-02-21:
Hi all,

I've noticed some transients signals on GPIO0 and GPIO2 on startup.

GPIO2 is the boot loader - is there any way to disable this ?

GPIO0 - this seems to pulse low after power on (not reset)  - I'm 99% sure this isn't my code.
Is this something we can fix in the core ?

If not, we should probably document this as it causes "bad stuff" to happen if you have w relay on it.

WereCatf, 2016-02-21:
GPIO0 apparently outputs some clock-signal at startup and shouldn't be counted on being stable for ~200ms after power-on. I, at least, can't find anything suggesting that anything could be done about it; I think it's simply how the hardware has been implemented.

Links2004, 2016-02-21:
GPIO2 is TX on boot its outputs the same then the TX pin, 
there is no way to disable this behavior.

penfold42, 2016-02-21:
Has anyone looked at the other gpio pins for similar junk ?

For now I've used a transistor and RC delay to disable the optio coupler. It's not pretty but it does suppress the power on transient.

closed 2016-06-23 by igrr