esp8266/Arduino #1659: Binary for gdb.exe in toolchain

hreintke, 2016-02-21:
The toolchain download for windows version 1.20.0-26-gb404fb9-2
http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz

Does not contain the xtensa-lx106-elf-gdb.exe

Is there a location where I can download the gdb.exe binary ?

Links2004, 2016-02-21:
check out @kireevco builds they include gdb.
https://github.com/kireevco/esp-alt-sdk
https://bintray.com/kireevco/generic/esp-alt-sdk/view

igrr, 2016-02-27:
As far as i can tell, there is an issue with gdb in esp-alt-sdk: https://github.com/kireevco/esp-alt-sdk/issues/23. Once it is resolved, we can switch to these toolchain builds.

devyte, 2017-10-18:
@igrr it has been over a year and a half, and the referenced issue is still open in the other repo.  Do we still want to use that repo? What alternatives are there?
BTW, I'd love to move to a newer toolchain, if at all possible.

igrr, 2017-10-18:
I think we can try moving at least to gcc 5.2.0. I have CI environment set up for it here, so we don't need esp-alt-sdk repo linked above to build the tools.

pandalion98, 2018-04-18:
Still waiting for a fix for this

VirtualGit, 2018-05-16:
In the meantime, you can download the toolchain from http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz and extract the xtensa-lx106-elf-gdb.exe file to C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin

hydra, 2018-05-17:
@VirtualGit that archive does not contain GDB.

VirtualGit, 2018-05-17:
@hydra My bad, I picked it from https://bintray.com/kireevco/generic/download_file?file_path=esp-alt-sdk-v1.5.0.258-windows-x86.zip (link from above comment)

earlephilhower, 2018-12-03:
GDB for Win32 and Win64 hosts is in via #5376 .  ACtually supporting GDB debugging will be a 2.6.0 thing.  Closing this part of the issues.

closed 2018-12-03 by earlephilhower