esp8266/Arduino #1660: ADC read while WIFI is disabled.

amrlsayed, 2016-02-21:
Hi , I need to declare an issue regarding using ADC when WIFI is disabled. 
I disable WIFI by calling 
`ESP.deepSleep(1, WAKE_RF_DISABLED); `
on wake up I get samples from ADC every minute , **however the reading is always 1024**,
but If I tried to read analog samples while WIFI is Enabled , I get it correct with no errors. 
the problem start to happen when I disable WIFI. 
**here's my code for Reading ADC**
`int reading  = analogRead(A0);`
 ` delay(3);`
 ` Serial.println("Light Sensor Reading ");`
 ` Serial.println(reading/4,DEC); // print light sensor level.`
I made a workaround that's by software I restart the ESP after the wake up from deep sleep 
by calling ESP.reset();
**By this way the ADC read problem is solved while disabling WIFI but this introduced another issue**
that is on calling ESP.reset(); , the board try to reset then hang and the 
wdt reset is introduced then start the program ,this make problems on the connected items  for my project , I don't want to disturb you by the details.
**Serial messages captured on the reset after calling ESP.reset() are**
> ets Jan  8 2013,rst cause:2, boot mode:(3,7)

>load 0x4010f000, len 1264, room 16 
>tail 0
>chksum 0x42
>csum 0x42
>~ld
>
> ets Jan  8 2013,rst cause:4, boot mode:(3,7)
>
>wdt reset
>load 0x4010f000, len 1264, room 16 
>tail 0
>chksum 0x42
>csum 0x42
>~ld
>
**My hardware connection is**
GPIO0 --- HIGH 
GPIO2 --- HIGH 
GPIO15 ---LOW

**My code for software reset is**
`wifimode = wifi.getEEPROMString(WIFI_STATE_ADDRESS,2);`
`if (wifimode == "d") { // disabled.`
`   debug_print("WiFi just Disabled");`
`    wifi.setEEPROMString(WIFI_STATE_ADDRESS, 2, "r");`
`    delay(200);`
`    delay(200);`
`    delay(200);`
`    delay(200);`
`    yield();`
`    ESP.reset();`
`    yield();`
`    delay(5000);`
`  }`
So I need help to resolve those Issues either a way to read ADC on WIFI disable (this is the main problem )
or a way to reset ESP correctly.

I also tried  to solve the ADC read problem by adding this macro after all the included libraries
`ADC_MODE(ADC_TOUT) `
I did so because I think the problem is that it read the vcc instead of the analog value,but it didn't resolve the problem .
also Tried calling
`ESP.wdtDisable();`on setup() and also on loop() but with no help.
My hardware is this board [http://tronixlabs.com.au/news/new-product-esp8266-esp12-full-evaluation-board/](url)
and I installed ESP8266 2.0.0 package from the board manger , I use Arduino IDE version 1.6.5. 
I don't know if I am doing something wrong , please Let me know If you need further information.
Thanks a lot.
Amr.

tommck, 2016-04-07:
I have the same exact problem!  Trying to read battery level and I always get 1024 without the WiFi enabled

lrmoreno007, 2016-04-07:
It's a ESP8266 core restriction:  ---> http://bbs.espressif.com/viewtopic.php?t=827
- TOUT must be floating
- The 107th byte in esp_init_data_default.bin(0～127byte) must be 0xFF
- RF must be enabled

tommck, 2016-04-07:
I'm not using ESP.getVcc().. I'm using analogRead(A0), which I guess is called by that underneath.

This seems like a REALLY weird restriction.

amrlsayed, 2016-04-07:
Hi,
As I said in the post ,the problem of the Analog read is solved by calling ESP.reset() after disabling WIFI , I disable WIFI by calling ESP.deepSleep(1, WAKE_RF_DISABLED); which already reset the board , so I reset it again( I save the WIFI status in the EEPROM to know when to reset the Board) , this solve the Analog read problem (I don't know why), but it introduces wdt reset problem ,the weird thing is if I am not using serial messages in my code , the wdt is not introduced anymore.
I really hope someone clarify this issue.

tommck, 2016-04-07:
So, you still use `ESP.deepSleep(1, WAKE_RF_DISABLED);`, but you then call `ESP.reset()` and the ADC works then?

amrlsayed, 2016-04-07:
Yes
1-I save the WIFI status  in the EEPROM to be "Just Disabled" then I call ESP.deepSleep(1, WAKE_RF_DISABLED); , the ESP reset itself.
2- and after the reset I check the EEPROM WIFI status if the WIFI is "Just Disabled" , I change the EEPROM WIFI status to another symbol for example "Already Disabled" and call ESP.reset() ,I use the EEPROM to save WIFI status so that it will not always reset itself.
then the ADC works fine, Again I don't know why.
but as I said ESP.reset() introduce wdt reset if I use Serial Debugger , but may this will not affect your application.

lrmoreno007, 2016-04-07:
Forgive my previous mistake.

I'm not sure, but it may be that you should set "vdd33_const" to real power voltage or at least 3.3v, because when you disable RF vdd33_const it may be set to 0 or other value. 

[image]

amrlsayed, 2016-04-07:
excuse me , how to change this byte on the run time ?
because resetting the ESP twice is not a good way.

closed 2016-06-23 by igrr

erikbakke, 2017-06-10:
Sorry to drag up an old thread, but it seems this is an issue in current git as well.  I can raise a new issue for it, if you prefer, but the symptoms are exactly as described above.

On my Adafruit Huzzah ESP8266 boards, with the 2.3.0 release, the ADC reads correctly even with the WiFi disabled.  However, in current git, the readings come back higher than expected.

I have 5.2V being run through a 18K/100K voltage divider.  With a multimeter I measure 0.77V at the ADC input, and the ESP measures 0.79V.  Close enough for my needs.

However, when I replace 2.3.0 with the git version, the ESP measures the voltage as 0.87V.

This only happens when the ESP wakes up from deep sleep with WAKE_RF_DISABLED.

YONG81K, 2017-10-16:
I experienced same effect when ESP12E wakes up from deep sleep (RF disabled) and try to read an analogue input from sensor. Could you let me know how I can do this please? Thank you very much.

amrlsayed, 2017-10-17:
it seems that this issue is solved on the followed releases, Please Open New issue describing your problem even if it's the same as this one , so that the Admins can follow up with you as this issue is already closed.

tjb36, 2022-11-23:
Sorry, I know this is old, but I am experiencing the same problem described above by @erikbakke , and this is the only place I have seen it described - did anyone find a solution? 

I am periodically waking the chip from deep sleep to perform an ADC read.

I measure a different conversion relationship between the voltage applied to the ADC pin (using an external bench PSU and measured with a precision floating multimeter) versus the counts returned by the analogRead() function, depending on whether I include the "WAKE_RF_DISABLED" flag in the ESP.deepsleep() command:
 
[image]

Strangely, when the flag is included, the ADC value returned on the very first boot after power on is actually the same as if no flag was passed. But then all subsequent wakes from deep sleep return the same ADC count as when no flag is passed.

Why does going into deep sleep with WAKE_RF_DISABLED change the values read by the ADC when the chip is woken up? 

(Using an ESP-12S board with 3.0.2 software version).