esp8266/Arduino #1661: esp8266 web server not accessible

geoffreyhhh, 2016-02-22:
with example code below, it acts as wifi client to connects to a wifi ap "testAP", and obtain a dhcp ip from the wifi ap, e.g. 192.168.43.52. at the same time, the esp8266 has it own ip, e.g. 192.168.4.1.

we can use web browser to access the 192.168.43.52 and to turn on/off led on esp8266 board.
at the same time, we use another pc to connect to esp8266_AP 192.168.4.1 and obtain a dhcp ip, e.g. 192.168.4.3, and use web browser to access 192.168.4.1 to turn on/off led on esp8266 board.

when the 192.168.43.* wifi connection is not established, accessing 192.168.4.1 is very difficult, ping to 192.168.4.1 shows many time-out and only rarely successful ping response. accessing 192.168.4.1 with web browser is failed

could anyone advise when is the cause that when 192.168.43.* wifi connection is not established, 192.168.4.* shows very poor connection performance and the 192.168.4.1 web server is hardly not accessible.

thank you.

=======================================================

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
 
const char* ssid = "testAP";
const char* password = "12345678";
MDNSResponder mdns;

ESP8266WebServer server(80);

const int led = 13;

void handleRoot() {
  server.send(200, "text/plain", 
    "hello from esp8266!) \n/on: to tuen LED ON \n/off: to tuen LED OFF \n");

}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}
 
void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  
  if (mdns.begin("esp8266", WiFi.localIP())) {
    Serial.println("MDNS responder started");
  }
  
  server.on("/", handleRoot);

  server.on("/on", [](){
    digitalWrite(BUILTIN_LED, LOW);
    server.send(200, "text/plain", "LED ON");
  });
  
  server.on("/off", [](){
    digitalWrite(BUILTIN_LED, HIGH);
    server.send(200, "text/plain", "LED OFF");
  });

  server.onNotFound(handleNotFound);
  
  server.begin();
  Serial.println("HTTP server started");

  //delay a moment, 
  //for terminal to receive inf, such as IP address
  delay(1000);
  Serial.end();
  pinMode(BUILTIN_LED, OUTPUT);
}
 
void loop(void){
  server.handleClient();
}

haidermirza, 2016-02-27:
I am having the same issue :( . Please update if you have found a workaround.

bjpirt, 2016-03-10:
I ran into this problem when using the SDK directly (no Arduino) - what I found was that if the STA interface was configured to connect but the access point it was trying to connect to was unavailable (or the password was wrong) the AP interface would become unstable. I fixed this by using a timer after enabling the STA interface that would check after 10 seconds and disable it if it hadn't connected.

I'm not sure if this is a known issue or not but this fixed the stability issues my users were seeing.

Here's a quick snippet to show what I now do in Arduino for this:

```
Ticker sta_tick;

void enableWifi(){
    // Set up the STA connection
    WiFi.begin(sta_ssid, sta_pass);
    // Check if it's connected after 10 seconds
    sta_tick.attach(10, staCheck);
}

void staCheck(){
  sta_tick.detach();
  if(!(uint32_t)WiFi.localIP()){
    WiFi.mode(WIFI_AP);
  }
}
```

igrr, 2016-03-10:
I think this is kind of expected behaviour. When the ESP is trying to connect to an AP, it scans all channels. Since SoftAP shares the same radio, it is disabled for the duration of the scan, as it should not hop between channels. I don't know if Espressif has this documented anywhere though.

bjpirt, 2016-03-10:
@igrr I think you're right, although it doesn't seem to get interrupted when doing a network scan, just when trying (and failing) to join an STA. I think this post is the closest we get to documentation:

https://github.com/esp8266/Arduino/issues/119#issuecomment-157493799

indexpert, 2016-03-14:
I am having the same issue ,what i did till now:
1) Set AP mode
2) Hosted webserver
3) Clients are server using soft AP IP
4) Before connecting to HTTP, scan N/w if home WIFI is available
5) if WIFI is available, 
      a) set STA IP= AP IP 
      b) AP IP=" some other ip"
      c) raise http request
      d) set AP IP back to original
      e) STA IP=<some other>

done...!

I am not sure , how ESP is able to connect HTTP, even if it is in AP mode, only by changing STA IP to AP IP....

dreamer2, 2016-05-26:
#1624

devyte, 2017-10-18:
This is known behavior, and as explained above, it is due to the ESP having a single radio shared between STA and SoftAP. If the Station part is to be used together with the SoftAP, the best practice is to force the SoftAP to the same channel as the AP that the Station is trying to connect to, and only then go and connect.
Closing.

closed 2017-10-18 by devyte

ukrsms, 2018-04-09:
it is written in esp_softap.h:
`@attention The ESP8266 is limited to only one channel, so when in the soft-AP+station mode, the soft-AP will adjust its channel automatically to be the same as the channel of the ESP8266 station`

MrteenBao, 2018-04-19:
Have any ideas for this problem ? Thanks