esp8266/Arduino #1662: Severe loss in stability when using WiFiUdp and ESP8266SSDP simultaneously

julianh2o, 2016-02-22:
I've traced down some severe stability issues in my current project to the concurrent use of ESP8266SSDP and a simple WiFiUdp listener.

I haven't yet had a chance to try to build a minimal sketch that reproduces the instability yet.. but I figured that there might be some known issue that someone may be able to comment on.

I'm also using a TCP server (using the WiFiServer) at the same time as both of these, so perhaps it's the three of them together that is causing issues.

The exception on the serial port is as follows:
    Exception (28):
    epc1=0x401030f7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000024 depc=0x00000000

    ctx: cont 
    sp: 3fff2760 end: 3fff2af0 offset: 01a0

    >>>stack>>>
    3fff2900:  3ffeb3d4 7fffffff 3ffeb3d4 00000001  
    3fff2910:  40104c35 00737574 00002200 40208632  
    3fff2920:  00040000 018562cb 53002200 3ffeb3ec  
    3fff2930:  4000050c 3fffc278 40104ab8 3fffc200  
    3fff2940:  00000022 00000096 00000992 ffdfffff  
    3fff2950:  4020dc9b 00000030 0000001b ffffffff  
    3fff2960:  4020dc98 0000002f 00000063 3fff3f72  
    3fff2970:  7e02c0fa 3fff3f72 00000000 60000308  
    3fff2980:  00000020 00000040 00000063 3fff159c  
    3fff2990:  00000031 3ffe8350 3fff0f80 00000030  
    3fff29a0:  00000000 00009200 3fff15b0 3fff10ca  
    3fff29b0:  0000009c 3fff15b0 00009200 3fff10ca  
    3fff29c0:  0000009c 0000009b 3fff15b0 4020ab16  
    3fff29d0:  00000000 3fff40a8 00000b14 3fff1118  
    3fff29e0:  00008600 00000174 0000004e 3fff10ca  
    3fff29f0:  00009200 00000126 0000009c 4020ab6c  
    3fff2a00:  3fff15b0 000001c2 00000100 ffffff00  
    3fff2a10:  3ffe8350 3fff1ab4 00000005 00000096  
    3fff2a20:  00000000 00000000 3fff1ad0 3ffe834c  
    3fff2a30:  3ffe8350 3fff0f80 4020decc 3fff1ad0  
    3fff2a40:  3fff0798 3fff0f80 00000031 3fff159c  
    3fff2a50:  3ffe8350 3ffe834c 3fff159c 40207a15  
    3fff2a60:  000063b5 3fff0f68 00000001 40208478  
    3fff2a70:  3fff0798 3fff0f68 3fff15bc 40208d1d  
    3fff2a80:  3ffe9350 00000000 000003e8 40210364  
    3fff2a90:  00000000 00000000 3ffe9350 00000000  
    3fff2aa0:  000003e8 3fff2b44 00000000 00000000  
    3fff2ab0:  00000000 3fff0fa0 3fff2b44 4020789c  
    3fff2ac0:  00000000 00000000 00000000 3fff1ac8  
    3fff2ad0:  3fffdc20 00000000 3fff1ac0 4020df15  
    3fff2ae0:  00000000 00000000 3fff1ad0 40100114  
    <<<stack<<<

     ets Jan  8 2013,rst cause:4, boot mode:(3,6)

    wdt reset
    load 0x4010f000, len 1264, room 16 
    tail 0
    chksum 0x42
    csum 0x42
    ~ld

I'll try to see if I can reproduce the same exception in a minimal sketch as I continue to work on a resolution.. but any ideas are welcome!

Links2004, 2016-02-22:
check the stack dump decode tool:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

and may enable the debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

give you the right information's.

julianh2o, 2016-02-22:
This is amazing.. I can't believe I didn't know about this stack dump decode tool.. I was part way through writing my own in Python last night while trying to debug this.

Also didn't know about the debugging tools, this is great, thanks a bunch!

julianh2o, 2016-02-22:
Hmm, at this point, I'm no longer convinced that this has anything to do with either of these uses of UDP.. I think it may be something else causing instability. Thanks again for the help!

closed 2016-02-22 by julianh2o