esp8266/Arduino #1663: Debugging ESP8266 via WiFi?!

tuxedo0801, 2016-02-22:
Hi there,

the Arduino Zeros comes with a dedicated debug port, which enables real debugging capabilities. Nice stuff.

The ESP8266 doesnt have this. Using serial port may conflict with other use cases/applications. 
As I personally don't need all the time the WiFi feature: 

Does anyone know of a additional library or so that enables kind of live debugging via WiFi?

The most simple solution would be: Sending debug-output not over serial, but over wifi ... But this needs maybe a lot of space for the debug-output strings... 

Any ideas/comments?

tuxedo0801, 2016-02-22:
Ah, found this:

https://github.com/esp8266/Arduino/tree/master/libraries/GDBStub
Works via serial. 
Any thing else than can handle WiFi to not use the ESP8266 serial's port?!

duncan-a, 2016-02-25:
I have no idea what that library is about, but I thought ESP had a second
UART output on a GPIO pin available for serial debugging when the main
RXDE/TXD pair were occupied? No idea how to use it though...

torntrousers, 2016-02-25:
See the comment at the bottom of #959, when/if it gets done.

tuxedo0801, 2016-02-25:
> but I thought ESP had a second UART output on a GPIO pin available for serial debugging when the  main RXDE/TXD pair were occupied? No idea how to use it though...

I read about it somewhere, but also no idea how to use it.

Pinout does not tell me where to find 2nd UART... (https://esp8266hints.files.wordpress.com/2015/05/top_layout_esp-12e1.jpg)

lrmoreno007, 2016-02-25:
2nd UART is TX only, and it's GPIO2.

How to configure?  I don't know.

tuxedo0801, 2016-02-25:
So, from code ist accessible via "Serial1", like with Arduino Leonardo ?!

tuxedo0801, 2016-02-25:
Found the solution for 2nd serial:

https://github.com/esp8266/Arduino/pull/31

So, this is for having serial debug output. But the thing mentioned in the first post:
Debugging capability with breakpoints and so on... not via serial (which will use the first serial port --> conflict if already used for other stuff like an UART connection to an external bus system) but via wifi ...

THAT would be great.

lrmoreno007, 2016-02-25:
Yes Serial1 work but always at 74880bps

igrr, 2016-02-25:
Keep in mind that ESP8266 has a single CPU, and WiFi is mostly implemented in software. You can not technically "halt" the CPU at a breakpoint at an arbitrary place in the program and maintain WiFi connection at the same time, because halting the CPU prevents WiFi stack from running.
If you need real debugging capabilities, don't waste your time and use JTAG for debugging (unless you also use SPI interface, which shares pins with JTAG).

tuxedo0801, 2016-02-25:
I feared that... Hmm. Any hints/links regarding JTAG with ESP8266 (especiallay ESP12 module). I'm totally new to JTAG. In principle I know what it is, but I have absolutely no clue how to use or even how to start with JTAG+ESP8266.

igrr, 2016-02-25:
You could start by checking out this tutorial: http://visualgdb.com/tutorials/esp8266/
Closing this issue as the original question has been answered.

closed 2016-02-25 by igrr