esp8266/Arduino #1664: esp8266 to esp8266

carlosjml4, 2016-02-22:
Regards Hello . Am new I am the subject and in this process of learning , I am trying to communicate two esp8266 , one I have configured as a server and it works fine, I responds to requests if I do from any other device , pc , tablet or phone. but I can not respond to requests me from another esp configured as a client.
Greatly I appreciate any help.

`server:
#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>

/* Set these to your desired credentials. */
const char *ssid = "my-ssid";
const char *password = "pwd";

ESP8266WebServer server(80);

void handle_led1(){
  digitalWrite(2, 1);
  server.send(200, "text/plain", String("LED ENCENDICO ")); 
}

void handle_led0(){
  digitalWrite(2, 0);
  server.send(200, "text/plain", String("LED APAGADO ")); 
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  // prepare GPIO2
  pinMode(2, OUTPUT);
  digitalWrite(2, 0);
  
  Serial.println();
  Serial.print("Configuring access point...");
  /* You can remove the password parameter if you want the AP to be open. */
  WiFi.softAP(ssid, password);

  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.begin();
  Serial.println("HTTP server started");
  server.on("/led1",handle_led1);
  server.on("/led0",handle_led0);
}

void loop() {
  server.handleClient();
}
`

`client:
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;
const char* ssid = "my-ssid";
const char* password = "pwd";

void setup() {
  USE_SERIAL.begin(115200);
  //USE_SERIAL.setDebugOutput(true);
  USE_SERIAL.println();
  USE_SERIAL.println();
  USE_SERIAL.println();

  for(uint8_t t = 4; t > 0; t--) {
    USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
    USE_SERIAL.flush();
    delay(1000);
 }

   WiFiMulti.addAP(ssid, password);
 
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        HTTPClient http;

        USE_SERIAL.print("[HTTP] begin...\n");
        // configure traged server and url
        http.begin("192.168.41.1", 80, "/led1"); //HTTP

        USE_SERIAL.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();
        USE_SERIAL.print("httpCode");
        USE_SERIAL.print(httpCode);
        /*if(httpCode) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == 200) {
                String payload = http.getString();
                USE_SERIAL.println(payload);
            }
        } else {
            USE_SERIAL.print("[HTTP] GET... failed, no connection or no HTTP server\n");
        }*/
    }

    delay(1000);
}

`

Links2004, 2016-02-22:
"-1" means ```CONNECTION_REFUSED```.
can have many reasons, but at first check if your WiFi router allow communication between devices.
some routers have disabled it by default.

and they server ESP ip: ```192.168.41.1``` and the client ip: ```192.168.4.2``` look like in different networks.
how do your network setup look like?

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h#L41

carlosjml4, 2016-02-22:
Hi thanks for answering . Let me explain again, the error -1 appearing understand. There are two ( 2) esp8266 one server and another as a customer, I can connect to the server from any device and allows on and off the led from the other esp8266 I connect , this is as a customer , but can not carry out the Connection http get.

carlosjml4, 2016-02-22:
I have settled some code and added the line  WiFi.mode ( WIFI_STA ) ; client side . now the code is running and devices are connected without problem.

`#include` `<ESP8266WiFi.h>`
`#include` `<ESP8266HTTPClient.h>`

```
const char* ssid     = "my-ssid";
const char* password = "pwd";
const char* host = "http://192.168.4.1";
HTTPClient http;

void rptaSrv(int httpCode){
  if(httpCode == 200) {
           String payload = http.getString();
           Serial.println(payload);
        } else {
            Serial.print("[HTTP] GET... failed, no connection or no HTTP server\n");
        }
}

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.println("proyecto inv client-server");
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);  //sin esta no funciona
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop(){
if(WiFi.status() != WL_CONNECTED){
  WiFi.begin(ssid, password);
  delay(500);
}else{
  //HTTPClient http;
        Serial.print("[HTTP] begin...\n");
        // configure traged server and url
        http.begin("192.168.4.1", 80, "/led0"); //HTTp
        int httpCode1 = http.GET();
        rptaSrv(httpCode1);
        delay(1000);
        http.begin("192.168.4.1", 80, "/led1"); //HTTP
        int httpCode2 = http.GET();
        rptaSrv(httpCode2);
        delay(1000);
    }
}
```

``

lrmoreno007, 2016-02-23:
Now you are connecting to 192.168.4.1. In the first one you was connecting to 192.168.41.1

It's not a issue, close it.

Saludos!!

closed 2016-02-23 by Links2004

mbostic, 2016-07-27:
I had exatcly the same issue and WiFi.mode(WIFI_STA) fixed it for me, so I don't think 41 was the problem @lrmoreno007

sgtwilko, 2016-11-16:
I have exactly this issue.

I'm using the examples provided in the Arduino IDE.  One ESP8266 is running a webserver with WiFi in AP mode (WifiAccessPoint example).
I can connect to this from my phone, it responds correctly.

I have another ESP8266 running the BasicHttpClient example (code almost identical to the code provided by @carlosjml4 above) and this will not GET the page from the ESP8266.

I've tried pointing it at my RPi webserver connecting via my home WiFi and this works.  It is only when one ESP connects to an AP made by another ESP that it fails to work.

blackworker87, 2016-12-10:
I tried also to connect two ESP8266 with the Arduino IDE. 
One ESP is running the example application "WiFiAccessPoint" which is performing quite well. (For instance notebook).

Then i started the second ESP8266 with the example "WiFiClient" changed the hostname to an IPAddress configured the Wifi to the AP of the first ESP8266 

The Client can connect to the Wifi but during
LN 54:  if (!client.connect(server, httpPort)) {  // server = IPAddress(192.168.4.1)
the function returns an error

ahmedriazku, 2017-03-29:
Carlos, i implemented your code with one server and two clients and it works fine. I need two clients to send data to the server. I am new to programming, what changes shall I make. I will be grateful if anyone else can also help me.