esp8266/Arduino #1667: OTA timeout caused by delay() function

Jyotishko, 2016-02-23:
Hello there, I have a slightly peculiar issue, where I think that the OTA is failing due to a delay() statement in my code.

I am reading data from an I2C temp+humidity sensor and am posting this data back to the local TCP server on my PC for now. I am currently using the delay() function to pause between taking readings and posting to the server.

When I have set my delay() to 600 miliseconds OTA works just fine. However, if I increase the delay to 1 minute, then on the IDE I get a response saying "[ERROR] No Response from device" or "[ERROR] No answer".

On the Serial output from the device I can see that it doesn't see the OTA request until the next temperature reading. But by the time it does see it the request has timed out and it fails with the message saying unable to connect.

I used the code modules in the BasicOTA example to build my program.

Has anyone seen a similar issue or has any ideas around how I can get the device to snap out of the delay loop as soon as the OTA interrupt is received?

Thanks

Details of my basic info below.

### Basic Infos

#### Hardware
Hardware:		ESP8266-12E
Core Version:      	2.1.0-rc2

### Description

Problem description

### Settings in IDE

Module:  		 ESP8266 Module - MCU
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?

Upload Using:		OTA

sticilface, 2016-02-23:
Use a non-blocking method, using mills() and a delta to decide to read or not.  
eg

```
void DoSmtg()
{
  static uint32_t tick = 0;
  if ( millis() - tick < 1000) { return; }
/*  Do stuff here every 1 seconds
  tick = millis();
}
```

Jyotishko, 2016-02-23:
Thank you very much. I will try this out.

With regards to millis(), I think I read somewhere that this refreshes
itself every 50 days or so, as it runs out of the buffer? Is that just in
the case of an Arduino board or does it apply to the 8266 as well?

Thanks in advance.

igrr, 2016-02-27:
You are right — millis will wrap around in about 50 days. 
However due to the way unsigned arithmetic works, doing `millis() - tick < 1000` is safe. I.e. if `tick == 0xfffffffe` and `millis` returned 20, then `millis() - tick` will evaluate to 22.

Demonbots, 2016-02-29:
Hi Ivan, thanks for your input on this. I added a check in my code to see if tick > millis() then tick=millis() to ensure that this does not fall over. I am about to start a test cycle over 50 days to see if this works.

However on a slightly different issue, I am also using the TimeLib.h library to get date and time from a NTP server during setup() and then keep track. I think that the library uses millis(). I think that this will have the same problem as well after around 50 days. Therefore, I am working on the fix to synchronise once millis() wraps around. I'm trying to think what is the value of millis() I should check for when it does wrap around. It needs to be under one or two seconds. What are your thoughts on this.

Thanks

P.S. On behalf of the community, thank you for all the help and effort that you guys put into this.

Bluebie, 2016-06-03:
@Demonbots I've read through some of the TimeLib codebase and it wont have any issue with millis rolling over. It stores time using a `time_t` variable (a 32-bit unsigned integer counting seconds since 1970 unix epoch) and does the same sort of tick-based millis() relative measurements to add to that `time_t` each time it's now() function runs. If you query it for the current time more frequently than once per second on average it will definitely stay in good sync. The time_t will roll over in the year 2038 causing the date to suddenly be in 1970, so whatever your building should be good for about 22 years. After that it'll need a firmware update to keep accurate time, with the way that library is currently written.

devyte, 2017-10-18:
Old issue, no feedback in over a year => closing.

closed 2017-10-18 by devyte