esp8266/Arduino #1669: ESP8266 & using Tabs gives (was not declared in this scope)

questuk, 2016-02-23:
Using
V1.6.7 IDE
ESP8266 V2.1.0 -rc2
NodeMCU 1.0

I tried the following

[BasicOTA.zip](https://github.com/esp8266/Arduino/files/142448/BasicOTA.zip)

If you have code on another TAB, it gives this error:

 "Arduino: 1.6.7 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

 In function 'void setup()':

BasicOTA:8: error: 'test' was not declared in this scope

   test();
        ^
exit status 1
'test' was not declared in this scope

Regards

Gaz

krzychb, 2016-02-23:
Hi Gaz,

This issue is tracked under https://github.com/esp8266/Arduino/issues/1066

A workaround is to declare function prototype before first use
```
void test();

void setup() 
{
  Serial.begin(115200);
  
  test();
  
}
```

Krzysztof

lrmoreno007, 2016-02-23:
Hi!! 

Rename IO.ino to IO.h and:

```
#include <ESP8266WiFi.h>
#include "io.h"

void setup() 
{
  Serial.begin(115200);
  
  test();
  
}

  
void loop() 
{
  delay(1000);
}
```

igrr, 2016-02-27:
This issue is fixed in the nightly builds of Arduino IDE, so i'm closing this ticket.

closed 2016-02-27 by igrr