esp8266/Arduino #1670: ESP8266WebServer Handle JSON post via request body.

mikehole, 2016-02-23:
### Basic Infos

#### Hardware
Hardware:			All versions
Core Version:      	       Latest

### Description

Apologies for using an issue to post this question. 

I am looking to have the web server handle JSON that is posted to the server as the request contents. How would it be possible to achieve this?

ajvpot, 2016-02-23:
Here's my code.
```
#include <ArduinoJson.h>
...

  server.on("/config/set", HTTP_POST, [](){
    StaticJsonBuffer<200> newBuffer;
    JsonObject& newjson = newBuffer.parseObject(server.arg("plain"));

...

    server.send ( 200, "text/json", "{success:true}" );
  });

mikehole, 2016-02-23:
Aha got it was looking through the code and didn't spot that the whole body was being placed in the plain argument!

closed 2016-02-23 by mikehole

mikehole, 2016-02-24:
Works a treat thanks for your help by the way :+1:

enzosagretti, 2016-06-11:
I am traing this Serial.println("plain: " + server.arg(val)); but i cant get to work. Return Empty

bkomac, 2016-12-16:
Look out for the size of the buffer. If your post payload is bigger then 200 it will be empty at the end.
Try with something bigger (StaticJsonBuffer<500> newBuffer;) or with  'DynamicBuffer'.

michaelvar, 2016-12-19:
What part of the request is the "plain"?