esp8266/Arduino #1671: ESP8266 stuck on reset loop

ranjit84, 2016-02-23:
Hi, 

I am trying to upload new sketch using ESP8266 board manager, I use blink example and code works,i can see inbuilt blue led of esp blinking without reset on serial monitor.

But when i add Serial.begin or try to upload WifiServer sketch, program uploads and get stuck on reset 
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Can anyone please help me on this.

### Basic Infos

Hardware:			ESP-01
Core Version:      	2

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			dio

ikbelkirasan, 2016-02-23:
First, make sure GPIO0 and GPIO2 are pulled high.

ranjit84, 2016-02-25:
I have 

VCC + RST +CHPD - 3.3V ( all connected to 3.3V)
GPIO0 + GPIO2 - 3.3 V
Rx - Tx ( FTDI )
Tx - Rx (FTDI)
GND - GND 

I have used separate power supply with 10 and 100 uF capacitors on Power and Gnd 
To upload I put Gnd to GPIO0

This is my connection till now, I have tried replace power supply
Nothing seems to work

I have Arduino 1.6.5 
with ESP8266 board - vers - 2
I am trying different setting to upload and upload is succesful all the time
I even tried blinking with GPIO2 pin and it worked with an led 
WifiWebServer example is still stuck on reset.

ikbelkirasan, 2016-02-26:
Your setup looks good, maybe you need to reset the module after the uploading is successful? Don't forget to pull GPIO0 and GPIO2 high after the uploading is finished.

igrr, 2016-02-27:
Also please run CheckFlashConfig example and verify that your flash chip is indeed 1Mb.

ranjit84, 2016-02-29:
I have checked my connections and still same error.
ets Jan 8 2013,rst cause:4, boot mode:(3,7)

I have observed few things here - 
1. ESP is very sensitive to power. 
I tried to upload sketches using nodemcu and esplorer, they have worked. While uploading the nodemcu firmware, power from FTDI chip was sufficient, but when I was trying to upload the sketch using esplorer with FTDI, it was not working. After I gave power to ESP with separate power supply and made common ground from FTDI and other power supply, then I only I could upload the sketch. 
And I tried this separate power supply with Arduino IDE and ESP Board, it is still not working. When I work with separate power supply and FTDI serial comm,I have to connect Ground from FTDI with the ground of separate power supply.

2. Blinking sketch is working with Arduino IDE and Serial.Begin() is not, I have uploaded blinking sketch and it works without resets, but as soon as I add Serial.begin(115200) it is stuck on reset loop. 
3. I can not make CheckFlashConfig this example work because it has serial printing that is not working. I have tried with configurations. Is there anyother way to check flash chip configuration may be I need to read chip number written on eeprom chip on esp.

ranjit84, 2016-02-29:
After changing Flash size to 512K (64K SPIFF),upload is success but after reset this is error on serial port - 

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)
wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
system param error
56 45 52 31 00 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff....

And now I am trying with different flash memory, others config are not giving this parse error but they get stuck 
I have removed all serial print code from CheckFlashConfg sketch, it is still stuck 

I will try with 9600 baud rate too now.

igrr, 2016-02-29:
Looks like you need to erase flash completely and then try CheckFlashConfig with 512K option selected.

To erase the flash, you can use esptool (the same one which is used by Arduino to upload the sketch).
Run it like this: `esptool -cp <port name> -cd ck -ce`. Make sure the ESP is in bootloader mode before running this command (if you are triggering it manually).

ranjit84, 2016-02-29:
thanks,I will try this and about bootloading mode - ESP8266 - 01
                                                      
UART Download Mode (Programming) 	GPIO0- 0   	GPIO2  - 1 	
Flash Startup (Normal) 		GPIO0- 1  		GPIO2- 1 	

Do I have to try it on Programming mode? 
I have checkd my circuit it behaves same no matter GPIO2 is 0 or 1.

tablatronix, 2016-03-06:
What do those switches do, i have been using
 esptool.py -p port erase_flash

igrr, 2016-03-16:
I meant to use esptool, not esptool.py. However you may use esptool.py to do the same if you have it installed.

tablatronix, 2016-03-16:
Found them
https://github.com/igrr/esptool-ck

heroseller17, 2016-06-11:
Hello i have problem with esp
when i want to switch relay esp resets!and its randomly but most of the time and i have this in serial port:
ets Jan 8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

i use arduino IDE for esp8266 programing .and i have esp-07
i have clear power supply with 1000uf || 100nf near the esp.
why it happens?

2-how many times esp could reset? 
it means that it resets many times....could any problem happend  because of this resets ?
for example in real time application for about 5 years esp resets about 100000 times...is it dangerous?

closed 2016-06-23 by igrr

polgaro, 2017-04-02:
I also had this problem but it was completely unrelated to every comment I read here.

I wanted to share my experience. I may help some of you.

My problem was simple: I didn't know that some boards (like mine) are labeled terribly. I thought that the D7 was GPIO7 but it was GPIO13. Actually, there's no GPIO 7, and by me trying to use it, it was causing the board to reset!!!

I found this page that maps the horrible labels to something that makes sense:
http://www.kloppenborg.net/blog/microcontrollers/2016/08/02/getting-started-with-the-esp8266

BTW, before you say anything: YES, I'm completely new to this. I worked with an Arduino for a week and as everything worked perfectly, I wanted to jump to the ESP8266. The Arduino UNO (even the fake ones) is properly labeled ;)

Anyways, I'm not saying that this is THE cause of your problems, but that was MINE. After changing the code that I was uploading, it worked perfectly

tablatronix, 2017-04-02:
They are not badly labeled those are the nodemcu pinouts, not arduino pinouts. You need to convert them or use the actual definitions D7

Daniel-Huber, 2017-04-14:
God knows how much time I have spent trying to find a solution to this issue, but @tablatronix and @polgaro posts fixed it! Thanks guys :smiley:

WheresWally, 2017-05-14:
Has anyone found a method for recovery a 12E module once it gets into an infinite reset boot loop?
I have 5 x 12E's on a small PCB with a MCP1825 3.3V linear reg and caps. All the normal pins needing to be tried high/low are handled. 
Now 3 of them work are working 24x7 without an issue in various circuits and applications.
BUT 2 of them just do the error text as listed at the bottom (at 74880 baud).

I tried this  [http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/](url)
And I do get the **ets_main.c%**  error when there is blank flash.

I tried programming many sketch via Arduino causes the same reset loop.

The LUA also causes the same reset loop.

I tried the AT firmware to and that also causes the reset loop
This seems to be a problem that many others have, but no one has solved, maybe easier to just buy a new one is the easiest way.

Fresh Out Of Ideas now....
Any help will be appreciated
```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
no GPIO select!
jump to run user1 @ 1000

Fatal exception (0): 
epc1=0x40201cdc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
```

Daniel-Huber, 2017-05-14:
@WheresWally need more information regarding the boards and setup. What's differentiating the code that's running on the 3 of them that work contrary to the code on the 2 that don't? Could you paste the code? Which IDE you using? Version of IDE?

tablatronix, 2017-05-14:
That link is not working, did you erase flash using esptool? Can you provide more detail about what you have done or are you having problems with a basic wifi example sketch?

WheresWally, 2017-05-14:
heres the link again
http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/

It shows how to use a 1Meg blank file to erase the entire flash using the ESP flash tool v.2.4

I didnt provide any code as it doesnt matter any code will not work at all. The most basic code wont work, the same code as the 3 workings devices will not work.
loop()
{
Serial.print("hello world");
delay(1000);
}
This doesnt work either. It reboots many times per second. the Blue LED is on almost constantly.

Flashing the AT SDK will not work.
Flashing the NodeMCU Lua will not work.
Using 3.3V /5A from the benchtop power supply will not make it work so I figure it isnt a power supply issue.

The code both were running was using SIPFSS (3M), SoftwareSerial to read data at 115200 and OTA was used a few hours earlier before it crashed in the wee hours of the morning.

I figure it has to be something to do with flash memory.

tablatronix, 2017-05-15:
You should use esptool to erase flash properly. You can find instructions around, i do not have then handy atm.

It comes with your esp install just gotta find it and run it manually or download it from its repo, also you may or may not have success with the exe and have to use the python version

WheresWally, 2017-05-15:
I did all the install esptool.py and erased the flash every which way, still no joy.
I did try only loading the boot.bin into flash. It didnt boot-loop so that was good.

`
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
no GPIO select!
jump to run user1 @ 1000

error magic!
first boot failed, reboot to try backup bin

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
no GPIO select!
jump to run user2 @ 81000

error magic!
backup boot failed.

user code done
`

rinelb, 2017-07-04:
I had similar issue **wdt reset**, how i fixed it was the changing the output "pinMode(4,OUTPUT);" I was using the ESP8266 relay board R1.5. 

When i changed the pin number to another "13" it stopped rebooting

Trospector, 2017-12-18:
One cause is the sketch's Global variables exceeding about 45% of dynamic memory, at least on a 1MB chip, like so:

Sketch uses 276611 bytes (30%) of program storage space. Maximum is 892912 bytes.
Global variables use 37640 bytes (45%) of dynamic memory, leaving 44280 bytes for local variables. Maximum is 81920 bytes.

When I make sure the global variables stay at or below 45%, the hardware resets disappear. One way to do this is to remove unnecessary #includes in the sketch, the other way of course is to reduce the number of global variables declared in the sketch.

DoiChris, 2018-02-20:
sorry, I do not have any idea how to run the esptool. could you please describe. Or can I erase the flash also by arduino IDE?

dynek, 2018-02-26:
@DoiChris this looks like esptool.py, then you don't need to pass those cp, cd, etc. arguments, but simply "erase_flash"

tablatronix, 2018-02-26:
Yeah its incredibly annoying that they use entirely different syntax, i use the python version exclusivly cause the exe is annoying.

Esptool.py -p ... erase_flash

sarmientosaurus, 2020-03-30:
Check also your SSID Password ist at least 8 characters long.