esp8266/Arduino #1672: Exception (29) in node_remove_from_list at .../app/lwip/app/dhcpserver.c:73

ajvpot, 2016-02-23:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.0.0

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU 0.9
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Debug Messages

See comments

Possibly related to #1162

I am running an HTTP server and a UDP server. I seem to get this exception randomly after an extended period of time. I'll comment with several stack traces.

ajvpot, 2016-02-23:
Exception (28):
epc1=0x402167ca epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: sys 
sp: 3ffffcf0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe90:  f2b75930 00000002 3ffeb1a4 00000016  
3ffffea0:  00000001 00000000 4022f054 3fff2110  
3ffffeb0:  3fff4d54 00000131 6704a8c0 3fff1ce0  
3ffffec0:  3fff3ed8 00000000 00000011 00000001  
3ffffed0:  3ffed768 4021e8a8 00000000 3ffeb814  
3ffffee0:  00000001 3fff38a0 3fff2a18 40214e84  
3ffffef0:  00000044 3fff2868 3ffeb800 3ffeb1d9  
3fffff00:  3fff0000 40211b76 3ffed768 3fff1df8  
3fffff10:  3ffeb800 3fff1dfc 3fff38a0 40211010  
3fffff20:  00000001 3fff2868 3fff2868 3ffeef20  
3fffff30:  00000000 3fff38a0 0000001c 3fff2868  
3fffff40:  3ffeb7f2 00000000 3fff38a0 40217729  
3fffff50:  00000000 00000181 00000000 00000016  
3fffff60:  00000002 00000018 4021ebbd 3ffed768  
3fffff70:  3ffeb7cc 000000f5 3ffe9c80 3ffe9c80  
3fffff80:  4021eb59 3ffed768 00000000 3fff28b8  
3fffff90:  3fffdc80 00000000 3fff38a0 4021817b  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Decoding 13 results
0x402167ca: node_remove_from_list at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/app/dhcpserver.c:73
0x4022f054: ieee80211_parse_wmeparams at ??:?
0x4021e8a8: ppRecycleRxPkt at ??:?
0x40214e84: udp_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/udp.c:339
0x40211b76: pbuf_alloc at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/pbuf.c:345
0x40211010: ip_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/ipv4/ip.c:553
0x40217729: ethernet_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/netif/etharp.c:1375
0x4021ebbd: pp_tx_idle_timeout at ??:?
0x4021eb59: pp_tx_idle_timeout at ??:?
0x4021817b: bit_popcount at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x4010f000: ?? ??:0

ajvpot, 2016-02-23:
```
Exception (28):
epc1=0x402167ca epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: sys 
sp: 3ffffcf0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe90:  b1626000 00000002 3ffeb1a4 00000019  
3ffffea0:  00000001 00000000 4022f054 3fff2110  
3ffffeb0:  3fff4534 0000012c 6704a8c0 3fff1ce0  
3ffffec0:  3fff4190 00000000 00000011 00000001  
3ffffed0:  ffffffff 00000000 3ffe9d81 3ffeb814  
3ffffee0:  00000001 3fff3818 3fff2a18 40214e84  
3ffffef0:  00000044 3fff2868 3ffeb800 3ffed470  
3fffff00:  3fff0000 40211b76 3ffed740 3fff1df8  
3fffff10:  3ffeb800 3fff1dfc 3fff3818 40211010  
3fffff20:  00000001 3fff2868 3fff2868 3ffeef20  
3fffff30:  00000000 3fff3818 0000001c 3fff2868  
3fffff40:  3ffeb7f2 00000000 3fff3818 40217729  
3fffff50:  00000000 0000017c 00000000 0000001a  
3fffff60:  00000002 00000018 4021ebbd 3ffed740  
3fffff70:  3ffeb7cc 000000f5 3ffe9d28 3ffe9d28  
3fffff80:  4021eb59 3ffed740 3fffdab0 3fff28d0  
3fffff90:  3fffdc80 00000000 3fff3818 4021817b  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Decoding 12 results
0x402167ca: node_remove_from_list at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/app/dhcpserver.c:73
0x4022f054: ieee80211_parse_wmeparams at ??:?
0x40214e84: udp_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/udp.c:339
0x40211b76: pbuf_alloc at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/pbuf.c:345
0x40211010: ip_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/ipv4/ip.c:553
0x40217729: ethernet_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/netif/etharp.c:1375
0x4021ebbd: pp_tx_idle_timeout at ??:?
0x4021eb59: pp_tx_idle_timeout at ??:?
0x4021817b: bit_popcount at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x4010f000: ?? ??:0
```

ajvpot, 2016-02-23:
```
Exception (28):
epc1=0x402167ca epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: sys 
sp: 3ffffcf0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe90:  50b75930 00000002 3ffeab60 00000022  
3ffffea0:  3ffed740 4021e8a8 00000000 00000000  
3ffffeb0:  3fff4024 0000013d 6704a8c0 3fff1ce0  
3ffffec0:  3fff3ea0 00000000 00000011 00000001  
3ffffed0:  3ffed740 4021e8a8 00000000 3ffeb1d0  
3ffffee0:  00000001 3fff3818 3fff2a18 40214e84  
3ffffef0:  00000044 3fff2868 3ffeb1bc 3ffeab95  
3fffff00:  3fff0000 40211b76 3ffed740 3fff1df8  
3fffff10:  3ffeb1bc 3fff1dfc 3fff3818 40211010  
3fffff20:  00000001 3fff2868 3fff2868 3ffeef20  
3fffff30:  00000000 3fff3818 0000001c 3fff2868  
3fffff40:  3ffeb1ae 00000000 3fff3818 40217729  
3fffff50:  00000000 0000018d 00000000 00000024  
3fffff60:  00000002 00000018 4021ebbd 3ffed740  
3fffff70:  3ffeb188 000000f5 3ffe9c78 3ffe9c78  
3fffff80:  4021eb59 3ffed740 00000000 3fff2900  
3fffff90:  3fffdc80 00000000 3fff3818 4021817b  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Decoding 13 results
0x402167ca: node_remove_from_list at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/app/dhcpserver.c:73
0x4021e8a8: ppRecycleRxPkt at ??:?
0x4021e8a8: ppRecycleRxPkt at ??:?
0x40214e84: udp_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/udp.c:339
0x40211b76: pbuf_alloc at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/pbuf.c:345
0x40211010: ip_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/ipv4/ip.c:553
0x40217729: ethernet_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/netif/etharp.c:1375
0x4021ebbd: pp_tx_idle_timeout at ??:?
0x4021eb59: pp_tx_idle_timeout at ??:?
0x4021817b: bit_popcount at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x4010f000: ?? ??:0
```

ajvpot, 2016-02-23:
```
Exception (28):
epc1=0x402167ca epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: sys 
sp: 3ffffcf0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe90:  f2b75930 00000002 3ffeb1a4 00000020  
3ffffea0:  00000001 00000000 4022f054 3fff2110  
3ffffeb0:  3fff3f14 00000131 6704a8c0 3fff1ce0  
3ffffec0:  3fff37b0 00000000 00000011 00000001  
3ffffed0:  00000000 00310b0b 00640102 3ffeb814  
3ffffee0:  00000001 3fff3768 3fff2a18 40214e84  
3ffffef0:  00000044 3fff2868 3ffeb800 3ffeb1d9  
3fffff00:  3fff0000 40211b76 3ffed718 3fff1df8  
3fffff10:  3ffeb800 3fff1dfc 3fff3768 40211010  
3fffff20:  00000001 3fff2868 3fff2868 3ffeef20  
3fffff30:  00000000 3fff3768 0000001c 3fff2868  
3fffff40:  3ffeb7f2 00000000 3fff3768 40217729  
3fffff50:  00000000 00000181 00000000 00000020  
3fffff60:  00000002 00000018 4021ebbd 3ffed718  
3fffff70:  3ffeb7cc 000000f5 3ffe9ca0 3ffe9ca0  
3fffff80:  4021eb59 3ffed718 00000000 3fff28c0  
3fffff90:  3fffdc80 00000000 3fff3768 4021817b  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Decoding 12 results
0x402167ca: node_remove_from_list at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/app/dhcpserver.c:73
0x4022f054: ieee80211_parse_wmeparams at ??:?
0x40214e84: udp_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/udp.c:339
0x40211b76: pbuf_alloc at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/pbuf.c:345
0x40211010: ip_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/ipv4/ip.c:553
0x40217729: ethernet_input at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/netif/etharp.c:1375
0x4021ebbd: pp_tx_idle_timeout at ??:?
0x4021eb59: pp_tx_idle_timeout at ??:?
0x4021817b: bit_popcount at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x4010f000: ?? ??:0
```

Links2004, 2016-02-23:
can you try 2.1.0-rc2 it has a newer SDK and lwip inside.
looks like a nullptr usage.
in raw cases it can be a flash access problem or powers supply problem.

ajvpot, 2016-02-23:
I recompiled the code with v2.1.0-rc2. I'll leave it running for a few hours and get back to you. Thanks!

devyte, 2017-10-18:
Old issue, no feedback in over a year => closing.

closed 2017-10-18 by devyte