esp8266/Arduino #1673: new Staging library(2.1.0-rc2) wont reset correctly and need hardware reset

mkeyno, 2016-02-23:
sometime when my web server is chocked and suppose to be reset however the watchdog reset freeze by  itself and I should do hardware reset
is there safe way to be sure for restarting the module 
 this is sample debug 
`
Exception (9):
epc1=0x401055b5 epc2=0x00000000 epc3=0x00000000 excvaddr=0xd9c429a3 depc=0x00000000

ctx: cont 
sp: 3fff1b30 end: 3fff1fd0 offset: 01a0

>>>stack>>>
3fff1cd0:  3fff5358 00000000 3ffe9320 401070e8  
3fff1ce0:  3fff4728 401070ff 4023f3c8 402261ea  
3fff1cf0:  00000000 00000000 000005b4 40211f86  
3fff1d00:  0000005a 00000000 40228c60 3fff4898  
3fff1d10:  00000000 00000000 3ffe9320 401070e8  
3fff1d20:  00000081 3fff1d40 40228e81 40211f86  
3fff1d30:  00000001 00000001 3fff1de0 40211f62  
3fff1d40:  3ffe0000 00000000 00000000 40211fba  
3fff1d50:  00303133 170400f8 3fff53e8 00000000  
3fff1d60:  00000001 4020b426 3fff4570 00000081  
3fff1d70:  3fff0270 000005b4 00000081 00000081  
3fff1d80:  00000000 3fff0250 00000000 00000001  
3fff1d90:  3fff0270 3fff53e8 00000081 4020b468  
3fff1da0:  00000000 00000000 3fff1de0 4021102c  
3fff1db0:  00000000 3fff0a50 3fff1de0 4020c0d3  
3fff1dc0:  3ffe9320 05c30000 3fff2050 000000c8  
3fff1dd0:  3fff4200 3fff1e20 3fff0250 4020c98c  
3fff1de0:  3fff53e8 0000008f 00000081 40211fea  
3fff1df0:  3fff1e94 3fff1e70 3fff1e20 00000000  
3fff1e00:  3fff1e94 3fff1e70 3fff0250 4020ca10  
3fff1e10:  00000008 00000000 3fff0250 4020a0a6  
3fff1e20:  3fff3f48 0000000f 00000000 00000000  
3fff1e30:  00000000 00000000 3fff1e70 4020f698  
3fff1e40:  3fff44a0 3fff44e8 00000001 00000000  
3fff1e50:  3fff2044 00000001 3fff1e70 3ffe84fc  
3fff1e60:  3fff2044 00000001 3fff1ec0 4020a1eb  
3fff1e70:  3ffe9bb0 00000000 000003e8 40211fba  
3fff1e80:  3fff44a0 3fff44e8 3fff3d20 0000001f  
3fff1e90:  00000010 3fff4200 0000001f 00000016  
3fff1ea0:  3fff46f8 401070ff 3fff028c 3fff0fa8  
3fff1eb0:  00000000 3fff028c 3fff028c 4020a38c  
3fff1ec0:  3fff30b0 0000000f 0000000d 4020e09d  
3fff1ed0:  00000000 00000000 00000000 00000000  
3fff1ee0:  00000000 00000000 00000000 00000000  
3fff1ef0:  00000000 3fffc6fc 3fff1f30 3fff0fa8  
3fff1f00:  00000000 3fff028c 3fff0250 40210e36  
3fff1f10:  00000000 00000000 00000000 4020cb16  
3fff1f20:  00000000 00000000 00000000 4020cbfa  
3fff1f30:  00000000 00000000 00000000 4020b3bc  
3fff1f40:  3fff028c 4020f234 3fff4540 00000000  
3fff1f50:  00000000 0000000d 000003e8 4020b346  
3fff1f60:  00000000 00000000 3fff0250 3fff0fa8  
3fff1f70:  00000001 3fff0270 3fff0250 4020ce83  
3fff1f80:  3ffe9378 00000000 000003e8 4020f115  
3fff1f90:  00000000 3fff4570 feefeffe feefeffe  
3fff1fa0:  3fffdad0 00000000 3fff0fa0 40206f06  
3fff1fb0:  3fffdad0 00000000 3fff0fa0 4020f34c  
3fff1fc0:  feefeffe feefeffe 3fff0fb0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
`

mtnbrit, 2016-02-24:
What is your hardware? Is it NodeMCU v1.0?

Links2004, 2016-02-24:
the no "reboot" problem is known see: #1017
more imported is why you get Exception (9).
use Esp Exception Decoder tool to decode the stack dump
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

igrr, 2016-02-27:
2.1.0-rc2 had a heap corruption issue, which is now fixed in 2.1.0. It could result in a similar stack dump.
Please try the latest stable version.

mkeyno, 2016-02-29:
tnx dear @Igrr , you are my savior , however I suffer from low speed downloading from SPIFF, it is actually reliable to put your website on the flash and  make webserver, I hope you and your friends @Links2004  have solution for that
BR

igrr, 2016-03-01:
@mkeyno what kind of download speeds are you seeing? What speed would you like to get? 
Out of the box, there are a few options available:
- gzip-compress the files which you plan to serve from SPIFFS. Browser will decompress them upon receiving.
- Use https://github.com/me-no-dev/ESPAsyncWebServer library for the web server. It handles multiple downloads in parallel and achieves much higher performance than ESP8266WebServer library.

mkeyno, 2016-03-01:
I could have download in range between 6 to 7 Kbyte/s but what bother me is sometimes the downloading process get delayed long before start sending the file to the browser client
any help really appreciated

mkeyno, 2016-03-01:
thanks @igrr  but it is needed to library which is work only with ESP31B  core and freeze when we use it for ESp8266  core , it seems the wifi library definition not  compatible for the  ESp8266

igrr, 2016-03-01:
It should work with ESP8266, that's what @me-no-dev originally based this library on. Please do create an issue in https://github.com/me-no-dev/ESPAsyncWebServer repository and attach an example which fails to work for you.

mkeyno, 2016-03-01:
dear @igrr  this is most common error

\arduino-1.6.4\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp: In member function 'size_t AsyncPrinter::_sendBuffer()':

\arduino-1.6.4\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:116:34: error: 'class cbuf' has no member named 'available'   size_t available = _tx_buffer->available();                                  ^
Error compiling.

me-no-dev, 2016-03-01:
that missing cbuf method should be in the core. The method name change was synchronized in both repos by me and @Links2004

mkeyno, 2016-03-01:
hi @me-no-dev 
would you tell me on which version and library you test your library  and should it @igrr  & @Links2004  put cbuf method in the next version

me-no-dev, 2016-03-02:
to me it seems to already be in the core. https://github.com/esp8266/Arduino/blob/master/cores/esp8266/cbuf.h#L35
maybe you have older version?

closed 2016-04-10 by mkeyno