esp8266/Arduino #1674: Serial upload works with TXD1 (debug) and TXD, safe assumption ?

hallard, 2016-02-24:
Hi guys,

I noticed something strange, I've designed a board with 2 positions jumper to be able to upload firmware and also have debug on Serial1 (because Serial0 used for other thing in my sketch). The jumber does the following, it connect TXD1 or TXD to the RXD pin of FTDI adapter.
- When FTDI RX Pin is connected to ESP12 TXD I'm able to upload program but I don't see any debug info because my debug code refers to TXD1 (Serial1), until there all is fine.
- When FTDI RX Pin is connected to ESP12 TXD1 (Serial1) I'm able to see debug info, sounds good too!

But what I've noticed, is that when FTDI RX Pin is connected to TXD1 (Serial1) I'm also able upload ?! Does this mean upload can use TXD or TXD1 as TX ? If so It's fine and avoid this jumper on my boards. 

Any confirmation would help.

Here the schematic I'm talking about
[image]

Thanks

Links2004, 2016-02-24:
yes that is normal, I use it this way all the time ;)
the bootloader always output the same data to TX and GPIO2 (TX1).

hallard, 2016-02-24:
Excellent, thx ;-)

hallard, 2016-02-24:
Just to be sure, you don't put any pullup on GPIO2 ?

closed 2016-02-24 by hallard

Links2004, 2016-02-24:
yes have never done it. since it TX on boot its High automatically, 
a 10k pullup will may improve stability on boot, but i never had any problems without it on all my setups.