esp8266/Arduino #1677: Soft-Power switch using deep sleep.

yukini3, 2016-02-24:
Hi,
I am looking for the solution of implementing soft-power switch using deep sleep. When I push the power button while the device is on, it goes to infinite deep sleep so that power consumption is minimum. When I push the power button while the device is off, it goes on. The button switch should be momentary on because sometimes device needs to go off by itself. It is like TV in your living room.

Can someone please help me to draw such kind of h/w?

Thanks!

ibaranov-cp, 2016-02-25:
Check here:

https://www.hackster.io/fablabeu/esp8266-thing-by-sparkfun-982bc6

yukini3, 2016-02-26:
> https://www.hackster.io/fablabeu/esp8266-thing-by-sparkfun-982bc6

I checked that web site but I couldn't find what I am looking for. Could you tell me which part is describing soft-power?
Thanks!

ibaranov-cp, 2016-02-26:
From the site"

"If you do want to wake up the ESP8266 THING while it is in deep sleep from the outside - like an external tigger: DTR to GND should makes this trick - works great!"

closed 2016-06-23 by igrr