esp8266/Arduino #1678: WiFi Webserver example uploads but then crashes.

waseber, 2016-02-25:
### Basic Infos

#### Hardware
Hardware:			ESP-01
Core Version:      	1.6.4(Arduino?)

### Description

I am able to upload the example sketch for WiFiWebServer with my wifi details included. Uploads successfully (seeing dots along the bottom ..... ).
However, when I disconnect the GPIO cable there is no serial indication at all, AND somehow blocks all wireless/wifi signals near it. I have spent WEEKS looking around the web for help. I have no idea what's wrong. I have also tried spare ESP01 modules.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

Using Exaple WiFi WebServer sketch

### Debug Messages

```
messages here
```
No error message given.. the leds may flicker a bit and then turn off.

Links2004, 2016-02-25:
how do your connection to the ESP01 look like have you added all needed pullups?
https://github.com/esp8266/Arduino/blob/master/doc/boards.md#boot-messages-and-modes
Note: GPIO15 is on GND on the board

try to connect with 77400 in the terminal and then power the ESP then you get the boot-loader messages.

waseber, 2016-02-25:
Thank you for your response!

I am trying to use a setup similar to the attached image (only I am using an FT232RL). I also have the jumper set to 3.3v instead of 5v.

Not sure I understand GPIO15--I don't think the ESP01 has this

By 77400 do you mean the upload speed/baud rate?

Thanks again!

[image]

Links2004, 2016-02-25:
open a serial terminal with 77400 baud and then power the ESP01.
GPIO15 is not exposed on the ESP01 but its on GND so dont worry there.
you need to pull up GPIO0 or your sketch will not boot.
| PIN           | Resistor | Serial Adapter |
| ------------- | -------- | -------------- |
| VCC           |          | VCC (3.3V)     |
| GND           |          | GND            |
| TX or GPIO2*  |          | RX             |
| RX            |          | TX             |
| GPIO0         | PullUp   | DTR            |
| Reset*        | PullUp   | RTS  (if your adaper have it)          |
| CH_PD         | PullUp   |                |

lrmoreno007, 2016-02-25:
FT232RL have not RST (must be inverted), have RTS (Ready to send). You can't do automatic flash without this circuit:
https://cloud.githubusercontent.com/assets/2480569/8785093/a39f0b80-2f25-11e5-9cc5-2137767ee491.png

With my old CP2102 it work fine directly, without the previous scheme, because it have (RST Inverted and DTR not inverted).

Here are a good scheme for manual upload, only add other pullup in RST line.

http://www.esp8266.com/wiki/doku.php?id=getting-started-with-the-esp8266

Always close the "Monitor Serie" window in Arduino.

Push PROG switch and hold it, push RESET switch and release, release PROG switch and then your ESP are ready to flash.

Regards.

waseber, 2016-02-26:
Thank you both for your help! I am trying out your suggestions. Will keep you posted on any developments.

igrr, 2016-02-26:
Make sure you don't power the ESP from serial adapter as shown in the picture above. It doesn't supply enough current for the ESP to operate reliably.

waseber, 2016-02-29:
igrr, so far I have always been able to upload the sketch successfully while being powered directly from the FT232RL, do I need a different power source when not uploading?

Irmoreno007, thanks for the tips, I have been trying to get that setup to work. I think it's close to working. One important question I have is: When uploading the sketch to ESP, should the GPIO0 be connected to ground but when looking at the serial monitor, should GPIO0 be connected to nothing?

Also, how can I set the Arduino serial monitor to 74880bps? Do I need to use another serial monitor like Lua?

Thanks again so much. Sorry I'm so slow at understanding!

igrr, 2016-02-29:
Absolutely. I can guarantee that you will get weird crashes and exceptions if you supply ESP from FTDI. While uploading it needs less current, but when WiFi is connected, there will be supply current spikes which FTDI internal regulator can not handle.

waseber, 2016-02-29:
igrr, I have an Arduino nano. Could I use its 3.3v to power the ESP?

waseber, 2016-02-29:
All right. With the power coming from the Arduino nano I am now getting this message (I was hoping to get the WiFi details, but this is certainly a step forward!)

Exception (9):
epc1=0x402166df epc2=0x00000000 epc3=0x00000000 excvaddr=0x4021a1f3 depc=0x00000000

ctx: sys 
sp: 3ffff980 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffb20:  4021b34d 3ffeda10 00000004 3ffeda14  
3ffffb30:  3ffe3a55 4021a1dd 3ffeda00 3ffedb74  
3ffffb40:  3ffeda10 3ffedd52 3ffedb74 3ffeda10  
3ffffb50:  3ffeda00 60000600 3ffeda10 118a57ea  
3ffffb60:  40215fe3 00000000 00000000 3ffedb74  
3ffffb70:  401059c2 60000e00 4021a8be 60000e00  
3ffffb80:  4021ae2a 3ffffb90 00000008 4020f292  
3ffffb90:  00000010 3ffe8364 00000000 000000d5  
3ffffba0:  00000000 00000000 00000000 3ffe8c70  
3ffffbb0:  000066d0 40201ed9 3ffe8c92 3fff0588  
3ffffbc0:  3ffe8c92 3fff04f8 0000007d 00000000  
3ffffbd0:  00000000 00000000 00000080 00000000  
3ffffbe0:  40101f31 4010131c 60000e00 4021028b  
3ffffbf0:  402106f7 60000e00 3fff0588 402106dc  
3ffffc00:  40101d70 0000007d 000000de 00000000  
3ffffc10:  3fffff10 3fff0608 3ffe8c92 1fc8576f  
3ffffc20:  ffffffff ffffffff ffff0001 00000006  
3ffffc30:  6e6e6542 00007465 00000000 00000000  
3ffffc40:  00000000 00000000 00000000 00000000  
3ffffc50:  ffffffff 3401ffff 38303242 45413737  
3ffffc60:  3ffe0033 00000001 3ffe878d 40203920  
3ffffc70:  3ffefbbc 00000005 3ffe856d 3ffeeb40  
3ffffc80:  3ffe8340 00000006 3ffefbbc 402036b5  
3ffffc90:  3ffe878c 00000000 ffffffff ffffffff  
3ffffca0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcc0:  ffffffff ffffffff 000000ff 00000000  
3ffffcd0:  0000000a 5f505345 35443131 00003531  
3ffffce0:  00000000 00000000 00000000 00000000  
3ffffcf0:  00000000 00000000 00000000 00000000  
3ffffd00:  00000000 00000000 00000000 00000000  
3ffffd10:  00000000 00000000 00000000 00000000  
3ffffd20:  00000000 00000000 00000000 00000000  
3ffffd30:  00000000 ffffffff ffffffff ffffffff  
3ffffd40:  ffffffff ffffffff ffffffff ffffffff  
3ffffd50:  ffffffff 000001ff ffffff04 ffff0001  
3ffffd60:  00000006 6e6e6542 00007465 00000000  
3ffffd70:  00000000 00000000 00000000 00000000  
3ffffd80:  00000000 30324234 41373738 fe003345  
3ffffd90:  0000013f fe878d00 2039203f fefbbc40  
3ffffda0:  0000053f fe856d00 feeb403f fe83403f  
3ffffdb0:  0000063f fefbbc00 2036b53f fe878c40  
3ffffdc0:  0000003f ffffffff ffffffff ffffffff  
3ffffdd0:  ffffffff ffffffff ffffffff ffffffff  
3ffffde0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdf0:  ffffffff ffffffff ffffffff ffffffff  
3ffffe00:  ffffffff ffffffff ffffffff ffffffff  
3ffffe10:  ffffffff ffffffff ffffffff ffffffff  
3ffffe20:  ffffffff ffffffff ffffffff ffffffff  
3ffffe30:  ffffffff ffffffff ffffffff ffffffff  
3ffffe40:  ffffffff ffffffff ffffffff ffffffff  
3ffffe50:  ffffffff ffffffff ffffffff ffffffff  
3ffffe60:  ffffffff ffffffff ffffffff ffffffff  
3ffffe70:  ffffffff ffffffff ffffffff ffffffff  
3ffffe80:  ffffffff ffffffff ffffffff ffffffff  
3ffffe90:  ffffffff ffffffff ffffffff ffffffff  
3ffffea0:  ffffffff ffffffff ffffffff ffffffff  
3ffffeb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffec0:  ffffffff ffffffff ffffffff ffffffff  
3ffffed0:  ffffffff ffffffff ffffffff ffffffff  
3ffffee0:  ffffffff ffffffff ffffffff ffffffff  
3ffffef0:  ffffffff ffffffff ffffffff ffffffff  
3fffff00:  ffffffff ffffffff ffffffff ffffffff  
3fffff10:  ffffffff ffffffff ffffffff ffffffff  
3fffff20:  ffffffff ffffffff ffffffff ffffffff  
3fffff30:  ffffffff ffffffff ffffffff ffffffff  
3fffff40:  ffffffff ffffffff ffffffff ffffffff  
3fffff50:  ffffffff 04080000 001c110c 01010000  
3fffff60:  ffffffff 00000003 00000000 ff000000  
3fffff70:  ffffffff ffffffff ffffffff ffffffff  
3fffff80:  ffffffff ffffffff ffffffff ffff0064  
3fffff90:  00000000 00000000 50d78f48 ff161094  
3fffffa0:  000201e9 6d0dc962 ffffff01 55aa55aa  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,5)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

igittigitt, 2016-03-01:
I have this issue either. Used another example sketch but some error. Upload is going fine, module initialize and then crashes all over with above output in serial interface. Im stucked now...

waseber, 2016-03-02:
Is the Arduino 3.3v also unreliable?

igittigitt, how are you powering your esp?

igittigitt, 2016-03-02:
Powered at 3.3V by a lab-powersupply. Kommunikation is done by a FDTI-Clone running with 3.3V jumper set.

waseber, 2016-03-07:
Just ordered a 3.3v breadboard power supply to be safe. Does anyone know what the above message means and how one fixes it?

waseber, 2016-03-10:
That fixed it. The 3.3v breadboard power supply made it work instantly. Thanks to all for all your help!

closed 2016-03-10 by waseber