esp8266/Arduino #1679: time() does not work without active NTP server

av1024, 2016-02-25:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.1.0-rc2

### Description

"Internal" time.c functions always return 0 /1970-01-01/ if not synced via NTP.

I want to initialize internal clock via constant or from RTC for log timestamping (so I don't need precise time on  startup, but valid measured intervals)

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	80Mhz
Upload Using:		 SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <Arduino.h>
#include <time.h>

void setup() {
 // WiFi.begin(); // wifi disabled for example
 // configTime(3*3600, 0, "not.avail.ntp"); 
 Serial.begin(115200);
}

void loop() {
  time_t now = time(nullptr);
 Serial.println(ctime(&now));
 delay(1000);
}
```

### Debug Messages

```
Thu Jan 01 00:00:00 1970

Thu Jan 01 00:00:00 1970

...
```

everslick, 2016-02-28:
I use this workaround for now:

```
static uint32_t starttime = millis() / 1000;
static uint32_t localtime = starttime;

void time_set(uint32_t time) {
  starttime = millis() / 1000;
  localtime = time;
}

uint32_t time_get(void) {
  return (millis() / 1000 - starttime + localtime);
} 
```

far5893, 2016-02-28:
I m looking for full software rtc in core libraries too.

av1024, 2016-02-28:
Thanks, @everslick!
I have my implementations RTC and/or NTP sync from my "raw C" projects on AVR, but I want to use "embedded" code because it already exists.

@far5893 There is widely known Time library from arduino.cc, but it is not compiled with 2.1.0-rc2 for me. And of course, this is not core lib

@igrr Please take attention to string handling in sntp_xxx code. I have got exception 28 (or sometimes 9) with sntp_asctime[_r] and sntp_sync_xxx approx after 22-24 hours uptime. there is only `ctime(&now)` calls in my code. One of saved traces:
```
Decoding 17 resultsException code: 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
0x402265f9: sntp_asctime_r at ??:?
0x402265f9: sntp_asctime_r at ??:?
0x4020d6cc: TwoWire::requestFrom(unsigned char, unsigned int, bool) at /home/av/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/libraries/Wire/Wire.cpp:240
0x40226647: sntp_asctime at ??:?
0x4020d6f8: TwoWire::requestFrom(unsigned char, unsigned char) at /home/av/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/libraries/Wire/Wire.cpp:240
0x40201b44: asctime at /home/av/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/cores/esp8266/time.c:110
0x4020df5e: Adafruit_BMP280::read24(unsigned char) at /home/av/Arduino/libraries/Adafruit_BMP280_Library/Adafruit_BMP280.cpp:210
0x40209742: check_float(float, float, float&) at /home/av/bin/arduino-1.6.5/WebConfig.ino:311
0x4020986e: readSensors() at /home/av/bin/arduino-1.6.5/WebConfig.ino:311
0x40105c2e: ets_timer_setfn at ??:?
0x4020a534: ESP8266WiFiSTAClass::status() at /home/av/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp:635
0x4020ad5c: WiFiClient::operator=(WiFiClient const&) at /home/av/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp:362
0x40209e08: every_s() at /home/av/bin/arduino-1.6.5/WebConfig.ino:311
0x40209eb8: handle_ms() at /home/av/bin/arduino-1.6.5/WebConfig.ino:323
0x40209ef3: loop at /home/av/bin/arduino-1.6.5/WebConfig.ino:337
0x4020f6f8: loop_wrapper at /home/av/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/cores/esp8266/core_esp8266_main.cpp:43
0x40100114: cont_norm at /home/av/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/cores/esp8266/cont.S:109
```

igrr, 2016-02-28:
sntp_xxx functions come from espressif SDK, as far as I can tell.

av1024, 2016-02-28:
@igrr Huh :( May be some try/catch over sntp_xxx in esp8266/time.c ? But I'm not sure this may help

Juppit, 2016-02-28:
Maybe you will have a look at this library:
https://github.com/Juppit/esp8266-SNTPClock

It was developed from a version with ticker library without network 
access. Therefore uncommenting the network stuff it should still give 
you time functions.

Juppit, 2016-02-28:
Wrong thread, I guess.
There was someone looking for time finctions without network.

av1024, 2016-02-28:
@Juppit Your version have same issue as this - system snmp_xxx functions does not provide correct time if no ntp sync was made prior. Another bug is in SNTPClock::begin - `while(!m_secsSince1900)` loop never ends if no network/success ntp sync made

igrr, 2016-02-28:
These are not c++ exceptions, these are CPU exceptions, so try/catch won't help here. Do you have a test case which is failing? I'd rather try to reproduce it and report to Espressif.

av1024, 2016-02-28:
@igrr I don't know how to reproduce it a bit stable. I'll try write minimal code only for timesync/print just now, but I have no idea how to speed up testing -  20+ hours before exception.. may be wifi connection issue while sync, may be some string formatting...

igrr, 2016-02-28:
20 hours is not that bad, I can probably hook it up to gdb and let it run
until it crashes.

kaeferfreund, 2016-02-28:
hey, if I got this right you are looking for some RTC capabilities. I have used the "TimeLib.h" for that. To initialize the time, I send a head request to a google server. I sync it every night and it's running since 2 weeks now...

av1024, 2016-02-28:
@kaeferfreund Nope. I am looking for system capability to work w/o internet/NTP server. I know about TimeLib (distribueted now as arduino.cc's Time.h). My use case is sensor, that may be sometimes offline and may go online w/o internet/NTP access. I plan to use DS3231 as backup RTC in my v3.1 PCB design but I just don't want to use third party lib where an system implementation exists.

@igrr, 5hrs. Still working. Neither blocking port 123 on router, nor disabling wifi access does not reproduce fail. But for now I know time intervals of internal time sync )) - if NTP accessible, then sunc performed every hour, if not - every approx 33seconds. PS: I am not familiar with gdb, but try tomorrow at work (not sure it will work on cubietruck, sot first try on PC)

av1024, 2016-03-02:
@igrr Three day test - all ok, no crash. _May_ be an issue with printing `asctime(nullptr)` - this call immediately throws exception 28. But I have no code with this call, only commented out from old exercises.

I'll try with more complex code and open separate issue if found something.

everslick, 2016-03-02:
FYI: I have the feeling, that stack growth has to do with Exception(28). I
could not find out yet, how to measure stack usage. I guess the stack
collides with heap occasionally. I could prevent Exception(28 by moving
stack variables to the heap.

igrr, 2016-03-02:
You can measure stack usage for the code which runs on Arduino task (i.e. called from `setup` and `loop`). See https://github.com/esp8266/Arduino/blob/master/cores/esp8266/cont.h#L65.
For code which runs on other tasks, there is no way to check this at the moment.

av1024, 2016-03-02:
In my case Exception 28 usually indicate incorrect usage of pointers - `asctime(NULL)` as above and mostly `print(PSTR/FPSTR/F(...))`. In most cases I use something like `print(PSTR("debug"))` instead of `print(F("debug"))` and got error as should.

everslick, 2016-03-03:
@Ivan: thanx for the hint. but i'm not sure, if i do it right:

i create a cont_t object:

static cont_t stack;

in setup() i call:
 cont_init(&stack);

in loop() i do:
  LogPrint("   free stack: %i bytes\n", cont_get_free_stack(&stack));
  LogPrint("  stack guard: %s\n",       cont_check(&stack) ? "CORRUPTED" :
"OK");

every two seconds.

but the log always reads:
  free stack: 4100 bytes

do i have to spawn a separate main loop with my cont_t object using
cont_run() ?

igrr, 2016-03-03:
Not exactly, you need to use the existing object.
```c++
extern "C" cont_t g_cont;

loop() {
  size_t free = cont_get_free_stack(&g_cont);
  // etc
}
```

everslick, 2016-03-03:
On Thu, Mar 3, 2016 at 4:32 PM, Ivan Grokhotkov <notifications@github.com>
wrote:

> extern "C" cont_t g_cont;

ahh, thx. without calling cont_init() on it i guess ?

hariprasad-d, 2017-01-30:
HI, Is this issue still open or closed?
Having the same problem here time api return "Thu Jan 01 00:00:00 1970".
looks like sntp.c never gets compiled from my IDE. put asserts under sntp_request and enabled LWIP_DEBUGF debug logs. But not core or logs seen.

joaohenriquef, 2017-10-24:
Same problem here. Any workarounds?

5chufti, 2017-11-06:
it looks like
a) s_bootTime only gets set on first use of via _gettimeofday_r() what seems to be some helper function for the newlib libc. 
b) s_bootTime may only be regarded via libc functions.
 time() definitely does not care about s_bootTime.

So no changes when using bare metal sntp functions ...

d-a-v, 2017-11-06:
With current master, the above sketch works
```
extern "C" void sntp_set_system_time(uint32_t);
```
and in ```setup()```:
```
sntp_set_system_time(0);
```

Similar *could* be done with lwip1.4 but not without patching source and recompiling, just because all functions and variables are statically linked in sntp.c (which has public ```sntp_get_current_timestamp()``` used by ```ctime()```).

The main goal here, in both case, is to install the ```os_timer_setfn()``` function.

5chufti, 2017-11-06:
somehow the whole time() thing looks pretty uncared of and broken ...
best to just use bare metall functions and 3rd party time(zone)lib.

P.S. with sdk sntp functions I can't connect to timeservice on M$servers, so I have to use "hand coded" timeservice after all.

d-a-v, 2017-11-06:
With example sketches, we can try and help.

PS: How do you use your ntp servers ?

PPS: The OP has to be modified with the two lines I showed.
We could propose a PR to make this automatic (without these two lines).
sntp timer would be initialized at boot and reinitialized when dhcp or user gives a ntp server.
Note that lwipv2 automatically uses dhcp-advertised ntp servers. Initially for me the OP sketch was working unmodified with current NTP time, and I had to ```ESP.eraseConfig()``` to go back 50 years ago.

5chufti, 2017-11-06:
ok, I can confirm that sntp_set_system_time(0); helps a lot.
I modified time.c / configTime()and inserted this after sntp_set_timezone() and now the call to configTime() starts the "fake rtc" and - if WiFi is working and the server responds - eventually the time will be correct even if WiFi drops later on.
Still an optional parameter to configTime() to set time to a value taken from  rtc chip would be nice to have. After this simple single initialisation all time operations could be done via standard functions.

now someone would have to set up a fake ntp server to check what dalightsaving rules espressif have implemented ..

P.S.: I use the M$ server exactly the same: just use its name instead of pool.ntp.org.
With hand coded udp requests I get an answer, sdk functions don't ...
Unfortunately I can't use wireshark, my IT-dept ...

```
#include <ESP8266WiFi.h>
#include <time.h>

#define ssid "your_ssid"
#define wpwd "your_secret"

// extern "C" void sntp_set_system_time(uint32_t);

void setup() {
  Serial.begin(74880);
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, wpwd);

  configTime(3600, 3600, "pool.ntp.org");
  
//  sntp_set_system_time(0);
}

void loop() {
  time_t tnow = time(nullptr);
  Serial.println(ctime(&tnow));
  delay(1000);

  if (millis()==30000) {
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    Serial.println("WiFi off");
  }
}
```

d-a-v, 2017-11-07:
You can fake using an external RTC with ```sntp_set_system_time(secs-since-epoch)```.
I fixed lwip2 so: 
- ```sntp_set_system_time()``` is exposed.
- ```sntp_set_system_time(0)``` is called at lwip init.
- ntp servers provided through ```configTime()``` can be used. Currently they are overridden by config to ntp.pool.org, it explains why you could not reach your preferred/authorized ntp server.

a PR is soon on its way

5chufti, 2017-11-07:
ok, so I see that the issue will be fixed with bells-and-whistles.
To spare you of further issues like this I suggest to do at least some rudimentary documentation on how to use the core time.c and the necessary preconditions.

5chufti, 2017-11-12:
@d-a-v 
Hi, had finally time to check your changes (used your fork) and time now starts w/o prior contact to ntp server, BUT
a) I was not able to use sntp_set_system_time(0); without prior declaration
b) TZ is somehow wrong.

if you test my small sketch with master, you will get following output:
with correct offset for MEZ
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
Thu Jan  1 01:00:00 1970
Thu Jan  1 01:00:01 1970
Thu Jan  1 01:00:02 1970
Thu Jan  1 01:00:03 1970
Thu Jan  1 01:00:04 1970
Thu Jan  1 01:00:05 1970
Thu Jan  1 01:00:06 1970
Sun Nov 12 12:09:06 2017
Sun Nov 12 12:09:07 2017
Sun Nov 12 12:09:08 2017
```

with your git it gives (with / without sntp_set_system_time(0))
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
Thu Jan  1 02:00:00 1970
Thu Jan  1 02:00:01 1970
Thu Jan  1 02:00:02 1970
Thu Jan  1 02:00:03 1970
Thu Jan  1 02:00:04 1970
Thu Jan  1 02:00:05 1970
Thu Jan  1 02:00:06 1970
Sun Nov 12 13:13:14 2017
Sun Nov 12 13:13:15 2017
******************************************************

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
Thu Jan  1 08:00:00 1970
Thu Jan  1 08:00:01 1970
Thu Jan  1 08:00:02 1970
Thu Jan  1 08:00:03 1970
Thu Jan  1 08:00:04 1970
Thu Jan  1 08:00:05 1970
Thu Jan  1 08:00:06 1970
Sun Nov 12 13:14:18 2017
Sun Nov 12 13:14:19 2017
```
as you can see in first case ds offset is wrongly applied 
in second case TZ seems to remain standard sdk value, still giving wrong time after sync (ds offset again wrongly applied)

d-a-v, 2017-11-13:
```sntp_set_system_time()``` is exposed in lwip include files.
You must include ```<arch/cc.h>``` or anything from lwip2 to have it.
I don't know if I should make it exposed for lwip1.4 too...

About DST, I just tried it again:
It is now ```Mon Nov 13 01:39:01 CET 2017``` on my laptop and my wall-clock (though the latter does not tell me TZ, and DST was sadly set about 2 weeks ago by hand... but agrees with my computers).
With winter time ```configTime(3600,0,"ntp.pool.org")``` which is MEZ (and CET), it esp-sketchly is ```Mon Nov 13 01:42:02 2017.```
With summer time configuration ```configTime(3600,3600,"ntp.pool.org")``` which is MESZ (and CEST too) esp says ```Mon Nov 13 02:42:48 2017```.

In your two last examples, what where ```configTime()``` parameters ?
How could you verify the validity of the patch with ```1499893466``` ? I guess you have the matching local time. It was in July'17, did you use it with (TZ,DST) = (3600,0)=MESZ or (3600,3600)=MEZ ?

5chufti, 2017-11-13:
HI,
ok there is not much sense in just adding dst-offset if it is present and not according to a dst rule.
If I have to decide if dst applies or not prior to call of configTime() I can just add it to the TZ value...

d-a-v, 2017-11-13:
DST is a difficult problem which seems not to be properly solved though I am not a specialist. There are two main cases, when you can access to internet, or when you can't. Check this [arduino lib](https://github.com/JChristensen/Timezone) and also [this](https://stackoverflow.com/questions/16086962/how-to-get-a-time-zone-from-a-location-using-latitude-and-longitude-coordinates).
If someone comes with a useful library, it might be worth a PR.

Juppit, 2017-11-13:
Take a look at [this](https://www.timeanddate.com/time/dst/2017.html).

So what about timestamps in periods with other rules?
I have just learned from it that e.g. in China was no DST after 1991.

5chufti, 2017-11-13:
I would have been "too good to be true" to have correct dst rules in sdk time functions.
Without major changes (forget about TZ as integer, string needed) there is no decent implementation.
With only integer TZ=XY you still could be in an area with different offset (0, 1800, 3600) or with different start/end dates, therefor textual TZ is needed.
So the achievable optimum is reliably start time() on startup with TZoffset from configTime() to get UTC, have sntp_set_system_time() properly exposed (including cc.h is even worse than have declaration).
Either place it in time.c or have ESP.time(), ESP.configTime() and ESP.sntp_set_system_time() 
Then it is up to the user to have correct TZdst handling via some TZ lib.

@Juppit unix-like "timestamps" are allways in UTC (ok, not really, no leapseconds) and have to be "translated" to local time. "DateTime" timestamps should never be used w/o postfix giving actual offset to UTC.

d-a-v, 2017-11-13:
@igrr (@devyte ?)
What do you think of exposing ```sntp_set_system_time()``` globally (in #3808) along with ```configTime()``` in ```Arduino.h``` ?
or with a better arduinishCapsed name like ```setTime()```/```setEpoch()``` ?

d-a-v, 2017-11-13:
@5chufti I put a library example ```esp8266/NTP-TZ-DST```.
@igrr I did not try to fix ```gettimeofday``` @BrandonLWhite is working on [it](https://github.com/esp8266/Arduino/pull/3818#issue-272961681)

BrandonLWhite, 2017-11-13:
I think implementing settimeofday as @igrr suggests sounds like the right approach.  I try to prefer use of existing CRT or POSIX APIs vs. introducing new stuff like low-level lwip functions.

I completed the work for getting gettimeofday functional just now, here in a new squashed PR: #3830.  

So what I'd recommend doing for a settimeofday implementation would be to extract my code 

```
s_bootTime_us = now_s * 1000000ULL - micros64();
s_bootTimeSet = true;
```

into a new method, then call it for both ensureBootTimeIsSet as well as settimeofday.

d-a-v, 2017-11-13:
I guess @igrr should take your PR first, then I'll adapt mine to yours.

BrandonLWhite, 2017-11-13:
@5chufti 
> I would have been "too good to be true" to have correct dst rules in sdk time functions.

I agree.  I like to think of timezones, DST, localtime as a user interface display concern.  Logic and event/data logging should stick to UTC.  You can certainly convey what the timezone is for the device, so that other systems can be aware and format accordingly.  As has been mentioned, you should use a textual ID such as an Olson ID.  A simple UTC offset is generally not good enough.

So, I agree that timezone and DST support is within the scope of the core lib.  If someone really wants or needs to build localized times in their firmware that should be accomplished with libraries.  That would have to pull in all of the tz database and logic and that's quite some baggage!  That's why I'd avoid it in an MCU.  Even if you are serving a web app straight from the ESP8266, you should use a JS client lib like moment.js to format time displays in a human format.  That can even be done with regard to future changes to DST rules without having to update your device firmware!

Thanks, just my 2 cents on the matter.

igrr, 2017-11-14:
Merged #3818, thanks!

igrr, 2017-11-14:
One data point the topic of timezone/dst support. Newlib (C library which we use) knows how to handle timezones and DST. Application needs to supply the timezone description (which may include DST rule), and newlib will take it into account when e.g. localtime_r is called.
So an application can use it like this:
https://github.com/espressif/esp-idf/blob/master/examples/protocols/sntp/main/sntp_example_main.c#L76-L87

Obviously the application needs to store the list of timezone descriptions somewhere (possibly mapped to user-friendly names of regions or major cities in each timezone). That, indeed, should be application's responsibility. But there is a project which can help generating TZ strings from the timezone database: https://github.com/nayarsystems/posix_tz_db.

5chufti, 2017-11-14:
@BrandonLWhite 
Hi, maybe I am wrong but isn't the whole "save bootTime" thing the wrong way around?
With your patch (in progress) we allways will have the "bootTime" for at least as long as int64 µs is worth (AFAIK more than a lifespan).

how about for time() just:
```
°) get real timestamp via sdk
°) if valid (t != 0) store diff to micros64() for later and return 
      tv_sec=timestamp
      tv_usec=micros64() % 1000000
°)  if not successful just return 
      tv_sec=(micros64() + diff)/1000000
      tv_usec=micros64() % 1000000 
```
this should return time from boot until ntp sync and +/- UTC after first sync?

reminder: we still need the correction of hardwired ntp servers from d-a-v patch.

BrandonLWhite, 2017-11-14:
@igrr 

> One data point the topic of timezone/dst support. Newlib (C library which we use) knows how to handle timezones and DST. Application needs to supply the timezone description (which may include DST rule), and newlib will take it into account when e.g. localtime_r is called.

Ah, yes of course, I completely forgot about that!  So this is still not a fully capable local time implementation, but it is good enough for getting local time pretty much "right now" and light duty UTC-localtime conversions.  The reason is that the TZ string that is used only supports a single DST forward/back rule.  So therefore all calculations must assume that is the rule for all of time.  While that is not correct for a great number of locales, it is acceptable for most applications.

You are right. This is capability exists in the runtime and people can and should use it if it meets their needs.

BrandonLWhite, 2017-11-14:
@5chufti 
2^64 us is about 584,554 years, so there is no danger there.

The time() CRT function only returns a 32-bit seconds-since-epoch value.  So there is no sub-second component. 

Regarding your example, I would be hesitant to mix values from different unrelated timing sources on each call.  There is no observable relationship between micros64 (monotonic) and sntp_get_current_timestamp().

> isn't the whole "save bootTime" thing the wrong way around?
It isn't ideal, but I think it's the best that can be achieved given what is available from the SDK.  Since the SDK can only yield 32-bit seconds, it is limited in terms of precision and Y2038.  It is good enough for use with the CRT time() function, but is not suitable for gettimeofday.  So for gettimeofday we can use the SDK's 32-bit value as a reference point, and then achieve precision and longevity by combining it with a 64-bit monotonic clock.
This would be unnecessary if the SDK provided a 64-bit way to get the current real time, with either us or ms precision.

5chufti, 2017-11-14:
@BrandonLWhite 
I took your micros64() implementation and the attached time.c 
(unfortunately system didn't let me attach any file)
Ofcourse it would be also possible to do sync only once and subsequently only if configTime() was called again if you have doubts on monotonicity. But the rest should be as monotonic as your micros64().
I teste it for general function and it works on old and new lwip.
```
/*
 * time.c - ESP8266-specific functions for SNTP
 * Copyright (c) 2015 Peter Dobler. All rights reserved.
 * This file is part of the esp8266 core for Arduino environment.
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 * See the GNU Lesser General Public License for more details.
 * You should have received a copy of the GNU Lesser General Public
 * License along with this library; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA 02110-1301 USA
 *
 */

#include <time.h>
#include <sys/reent.h>
#include "sntp.h"

#ifndef _TIMEVAL_DEFINED
#define _TIMEVAL_DEFINED
struct timeval {
  time_t      tv_sec;
  suseconds_t tv_usec;
};
#endif

static uint64_t s_UTCdiff_us = 0;

static void setServer(int id, const char* name_or_ip)
{
    if (name_or_ip)
    {
        //TODO: check whether server is given by name or IP
        sntp_setservername(id, (char*) name_or_ip);
    }
}

void configTime(int timezone, int daylightOffset_sec, const char* server1, const char* server2, const char* server3)
{
    sntp_stop();

    setServer(0, server1);
    setServer(1, server2);
    setServer(2, server3);

    sntp_set_timezone(timezone/3600);
    sntp_init();
}

int clock_gettime(clockid_t unused, struct timespec *tp)
{
    (void) unused;
    if (tp)
    {
        uint64_t currentTime_us = micros64() + s_UTCdiff_us;
        tp->tv_sec  = currentTime_us / 1000000;
        tp->tv_nsec = currentTime_us * 1000;
    }
    return 0;
}

time_t time(time_t * t)
{
    time_t now = sntp_get_current_timestamp();

    if (! now)
    {  now = (millis() + s_UTCdiff_us / 1000) / 1000;
    }
    else
    {  s_UTCdiff_us = now * 1000000ULL - micros64();
    }
	
    if (t)
    {
        *t = now;
    }
    return now;
}

int _gettimeofday_r(struct _reent* unused, struct timeval *tp, void *tzp)
{
    (void) unused;
    (void) tzp;
    if (tp)
    {
        uint64_t currentTime_us = micros64() + s_UTCdiff_us;
        tp->tv_sec = currentTime_us / 1000000;
        tp->tv_usec = currentTime_us % 1000000;
    }
    return 0;
}
```

d-a-v, 2017-11-15:
In #3835, there is still an issue with ```gettimeofday()``` if it is called for the first time before sntp-date is set by ntp. It works well after, or with ```settimeofday()```.
A solution would be to reset ```s_bootTimeSet``` when ntp is set (or adjusted?)
Can it be reset anytime with no harm ?

(note that this PR is now on my ntp branch if you try that way)

d-a-v, 2017-11-15:
With the latest #3835 commit (not yet merged, or my ntp branch), the main bug is ```gettimeofday()``` which is always wrong if it is called first time before a ntp date is set.
A good way to fix this would be to be able to re-set ```s_bootTimeSet``` and al. everytime ntp receive a new timestamp. It is done though ```sntp_set_system_time()``` called from both ```settimeofday()``` and sntp from inside lwip through an lwip-api-#define.

```sntp_set_system_time()``` is defined in lwip2, because it was defined inside lwip1.4. @igrr I think this code, which was introduced by espressif inside lwip - where it does not belong to - should be moved into ```cores/esp8266```, but that would introduce a change in lwip1.4 too (we could move it anyway from lwip2 with ```attribute((weak))``` and no change in lwip1.4)

In either case, ```sntp_set_system_time()``` can take one (epoch) or two (epoch, usec) arguments (it's a choice, it currently is one). 

Do you think ```time.c``` can be changed to be able to be updated/reset/restarted through a call from ```sntp_set_system_time(epoch,usec)``` (ie. from external RTC or SNTP) ?

igrr, 2017-11-15:
I very much agree about removing the time keeping functions from LwIP SNTP module.
In my mind, SNTP should just call `settimeofday` when new timestamp is received.
All timekeeping should be done in the core. All timezone management should be done in newlib. I think that having LwIP SNTP responsible for timekeeping is crazy, but apparently many of our customers already use these extra module SNTP functions, so they are not going away in the ESP8266 SDK.
But since we are moving towards using upstream LwIP in this project, i think we should nuke these additions and do things the right way.

d-a-v, 2017-11-15:
already in progress, commit is coming :)

d-a-v, 2017-11-15:
Complete commit log:
> moved time functions from lwip2 to core
> * sntp_set_system_time() is not anymore used by lwip2 nor callable, settimeofday() is the entry point
> * ntp timestamps or settimeofday() now reset time functions, and give (sec,usec) instead of (sec)
> * update esp8266/NTP-TZ-DST example accordingly
> * new core files: coredecls.h and sntp-lwip2.c
> * update lwip2 commit ref and lib accordingly
>
> further work has to be done in settimeofday() to properly take tv_usec into account

I will let you fix ```tv->tv_usec``` in ```settimeofday()```, I don't feel I'd be good at it :)

d-a-v, 2017-11-21:
master's core has been updated
 * including active time without NTP
 * with ```settimeofday()``` receiving but *not* honouring ```tv_usec```  (sntp or user calling) 
 * without ```adjtime()```

@igrr thanks !

5chufti, 2017-11-23:
Hi,
I think a little more info is needed:

what is the result of "time();"?
a) an internal timer based value or b) a real ntp-server answer?
if a) how often/when is this internal timer synced with a ntp-server (if configured and active)?
can this be configured/forced (how)?

bg:  I only need a longterm accuracy of appr. timestamp resolution (s), and I don't want to flood the ntp-server, so internal timer would suffice for a certain periode but eventual resync is necessary.

thanks for your patience

d-a-v, 2017-11-23:
Without NTP, it starts from 1st january 1970 8am (it's EPOCH in Espressif's HQ). It can be forced with [settimeofday()](http://www.manpages.info/linux/settimeofday.2.html).

Time is re-set / adjusted by sntp driver:
 * every time the ESP is receiving a new DHCP update (which happens regularly) containing a NTP server address (depends on your server)
 * every hour (3600000us == SNTP_UPDATE_DELAY) by default

5chufti, 2017-11-23:
thank you, 
I'm working my way through the "time&TZ" functions of newlib at the moment; looks good so far. If I'm using static IP and disable WiFi for energy efficiency there is no way to force "sync" during a WiFi enabled periode? 

I did see the example but missed something more enduser oriented like the following:

```
#include <ESP8266WiFi.h>
#include <sys/time.h>                   // struct timeval
#include <time.h>                       // time() ctime()

#define ssid "yourSSID"
#define wpwd "yourPASSPHRASE"

#define RTC_TEST 1510592825 // = Monday 13 November 2017 17:07:05 UTC
// #define RTC_TEST 1499893466 // = Wed 12 July 2017 21:04:26 UTC

void setup() {
  Serial.begin(74880);
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, wpwd);

  timeval tv = { RTC_TEST, 0 };
  timezone tz = { 0 , 0 };
  settimeofday(&tv, &tz);
  configTime(0, 0, "pool.ntp.org");
  setenv("TZ", "CET-1CEST,M3.5.0,M10.5.0/3", 3);
  tzset();
  // don't wait, observe time changing when ntp timestamp is received
}

void loop() {
  time_t tnow = time(nullptr);
  Serial.println(ctime(&tnow));

  if (millis() > 30000 and millis() < 31000) {
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    Serial.println("WiFi off");
  }
  delay(1000);
}
```

d-a-v, 2017-11-23:
Do you want to avoid using an external RTC ?
Do you want to get a new NTP time stamp, and put Wifi off once received until next time you decide to resynchronize time ?
In you sketch above, use configTime to set TZ, DST and default NTP server.

```sntp_request(NULL)``` is what is internally achieved every hour.
It can be called by the sketch.
Once NTP time stamp is received, ```settimeofday()``` is called by SNTP, then you can put Wifi off again.

There is curently no way to know when ```settimeofday()``` is called. It can be discussed later if it is needed, in the meantime you can wait a few seconds before going to sleep again and report if it fits your needs.

5chufti, 2017-11-23:
If I set the offset in configTime to 0,0 I know what I get: UTC.
If I set anything else the only thing I know is: its wrong 50% of the year (you cant get correct TZ rules from two plain offsets)
So I prefer to get plain UTC and have newlib take care of the TZ thing ...

And if it takes "several seconds" as I observe at the moment it is around 20!, this is not energy efficient. I think t is due to the stupid 15s default of SNTP_UPDATE_DELAY that also applies to the first sync... so no game for me, back to "manually" fetching sntp_timestamps.

d-a-v, 2017-11-23:
```SNTP_UPDATE_DELAY``` is the internal interval between to NTP request in the sntp driver in a world where network is always on and available.
In your case, if you do your request by hand with ```sntp_request(NULL)``` once the network becomes available, you should get your answer right away. Did you try that ?

5chufti, 2017-11-23:
unfortunately:
"sntp_request" was not declared in this scope

everslick, 2017-11-23:
extern "C" ?

[19 more replies]

PR #4001 (2017-12-24): handle tv.tv_usec in settimeofday()
optional settimeofday()'s callback
fix #1679
commit 9913e521 (2017-12-24): handle tv.tv_usec in settimeofday() (#4001)

optional settimeofday()'s callback
fix #1679
closed 2017-12-24 by devyte