esp8266/Arduino #1681: Sending data in CSV format

oKieran, 2016-02-25:
Hi,

I have just picked up the ESP8266 and have managed to connect it to a network using serial AT commands. I need to be able to send a reasonable amount of data to a location on the network drive (I.E a shared area) preferably in a CSV format. The data is collected using an Arduino Uno using various methods, I will use the same Arduino to control the Wifi.
Is this possible? I have seen alot of tutorials on how to send/recieve data from/to websites, but not to a network drive. Sorry if this seems like a noob quesion..

The only way I can think of is to send the data to an SD card mounted to the Arduino and have an SSH client (PuTTY?) on a PC to listen to the ESP8266 and create the CSV to a mapped drive on the PC... if this is even possible.. 

Thanks alot!

Links2004, 2016-02-25:
first this git has noting to do wit AT commands.
its a full replacement for the AT firmware, we run outer code direct on the ESP module no need for any other Arduino (the Arduino code runs on the ESP8266 chip).

SSH and samba are some big protocolls no one has implement them yet for a embedded system.
the ESP module as a SPI flash chip with up too 4MB and you can use up the 3MB for store data so you man not need a sd card.
to get the data form ESP to PC you can start a webserver on the ESP and simple access the data via browser.

oKieran, 2016-02-25:
Apologies I've obviously got the 'wrong end of the stick'! 

I'll try the webserver suggestion, though I need things to be automated without any user input..

closed 2016-06-23 by igrr