esp8266/Arduino #1683: UART RX BUFFER SIZE

snowbizz, 2016-02-25:
Hi, 
In 2.0.0 version there were two definitions of UART's buffer size in HardwareSerial.h file:
#define SERIAL_TX_BUFFER_SIZE 256
#define SERIAL_RX_BUFFER_SIZE 256

After 2.1.0 it is gone and buffer size can't be changed.
I can't receive more than 128 chars which is default value.

I found also in uart.h file:
#define UART_TX_FIFO_SIZE 0x80
But it is only for TX and I am not sure is this works.

Cheers.

igrr, 2016-02-25:
Software managed UART buffer has been removed in favor of hardware RX buffer, which is 128 characters long.

snowbizz, 2016-02-25:
So there is no chance to receive more than 128 chars by Hardware Serial?
Because in version 2.0.0 I was able to receive more than 128 chars by Hardware Serial.

penfold42, 2016-02-25:
You I'll have to grab the characters more often and maintain your own bigger software buffer.

The main purpose of the uart buffer is to catch characters while your software is busy doing other things.

snowbizz, 2016-02-26:
It is working this way, but in my opinion - having a possibilty to change a buffer size will be a nice feature :)

far5893, 2016-02-26:
You only need is to listen hw buffer overrun interrupt .

closed 2016-02-27 by igrr